raspiblitz/raspiblitz #3731: [RESEARCH] Update to 2TB SSD

rootzoll, 2023-04-03:
At one point we might recommend a 2TB SSD instead of 1TB. So this issue is starting to collect details on this.

Most important what SSDs run the most power stable & reliable.

Also see maybe related issues:
- https://github.com/rootzoll/raspiblitz/issues/2583#issuecomment-1483960945

openoms, 2023-04-05:
Report of the Samsung Evo 870 2tb working well
 https://t.me/raspiblitz/126170

allyourbankarebelongtous, 2023-04-05:
Rpi 4 model B 8gb RAM

Ran stable for over 1 yeah with:
SanDisk SSD PLUS 2TB Internal Ssd in a UGREEN External Hard Drive Enclosure USB 3.0 to SATA III Adapter for 2.5" SATA SSD HDD 9.5mm 7mm Hard Drive Case Housing with UASP

Ran stable for 4 months with:
SAMSUNG Electronics 870 EVO 2TB 2.5 Inch SATA III Internal Ssd in a UGREEN 2.5" Hard Drive Enclosure USB C 3.1 Gen 2 to SATA III 6Gbps for SSD HDD 9.5 7mm External Hard Drive Disk Case with UASP

joleka, 2023-04-09:
My Raspiblitz has been in action for 15 months. I equipped it with a brand new Samsung 870 QVO 2 TB version and it has been running flawlessly ever since. I read out the SMART values for the first time today and more than 158 terrabytes have been written so far. According to Samsung, the Total Bytes Written (TBW) is 720 terrabytes for this drive. So after this SMART readout today, if I was going to assemble a new Raspiblitz, I would probably buy an 870 EVO, which should achieve a TBW of 1.2 petabytes.