raspiblitz/raspiblitz #3733: Use Core Lightning JSON RPC interface by default

fusion44, 2023-04-03:
The API now supports the UNIX socket JSON RPC interface or Core Lightning. This should be the preferred way of connecting to Core Lightning for now. To activate, the `.env` file must contain the following settings:

```.env
ln_node=lnd_grpc

# cln json rpc - path to the socket file
cln_jrpc_path="/mnt/hdd/app-data/.lightning/bitcoin/lightning-rpc"
```
Note that `cln_jrpc_path` variable in the `.env_sample` file already points to the correct path on the RaspiBlitz.

rootzoll, 2023-04-08:
@fusion44  `ln_node=lnd_jrpc` should be `ln_node=cln_jrpc, correct?

fusion44, 2023-04-08:
@rootzoll Oh dang, yes correct. I've edited the original post!

rootzoll, 2023-04-08:
@fusion44 When running the JRPC I get the following error when trying to use the WebUI on the API:

```
Apr 08 17:29:17 raspberrypi python[742]: 2023-04-08 17:29:17 | ❌ | cln_jrpc.py:197 | get_ln_info() returned None
Apr 08 17:29:17 raspberrypi python[742]: 2023-04-08 17:29:17 | ❌ | service.py:84 | An error has been caught in function 'list_all_tx', process 'MainProcess' (742), thread 'MainThread' (548168671296):
Apr 08 17:29:17 raspberrypi python[742]: Traceback (most recent call last):
Apr 08 17:29:17 raspberrypi python[742]:   File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
Apr 08 17:29:17 raspberrypi python[742]:     return _run_code(code, main_globals, None,
Apr 08 17:29:17 raspberrypi python[742]:            │         │     └ {'__name__': '__main__', '__doc__': None, '__package__': 'uvicorn', '__loader__': <_frozen_importlib_external.SourceFileLoade...
Apr 08 17:29:17 raspberrypi python[742]:            │         └ <code object <module> at 0x7fa0d93450, file "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/__main__.py", ...
Apr 08 17:29:17 raspberrypi python[742]:            └ <function _run_code at 0x7fa0d8fc10>
Apr 08 17:29:17 raspberrypi python[742]:   File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
Apr 08 17:29:17 raspberrypi python[742]:     exec(code, run_globals)
Apr 08 17:29:17 raspberrypi python[742]:          │     └ {'__name__': '__main__', '__doc__': None, '__package__': 'uvicorn', '__loader__': <_frozen_importlib_external.SourceFileLoade...
Apr 08 17:29:17 raspberrypi python[742]:          └ <code object <module> at 0x7fa0d93450, file "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/__main__.py", ...
Apr 08 17:29:17 raspberrypi python[742]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/__main__.py", line 4, in <module>
Apr 08 17:29:17 raspberrypi python[742]:     uvicorn.main()
Apr 08 17:29:17 raspberrypi python[742]:     │       └ <Command main>
Apr 08 17:29:17 raspberrypi python[742]:     └ <module 'uvicorn' from '/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/__init__.py'>
Apr 08 17:29:17 raspberrypi python[742]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
Apr 08 17:29:17 raspberrypi python[742]:     return self.main(*args, **kwargs)
Apr 08 17:29:17 raspberrypi python[742]:            │    │     │       └ {}
Apr 08 17:29:17 raspberrypi python[742]:            │    │     └ ()
Apr 08 17:29:17 raspberrypi python[742]:            │    └ <function BaseCommand.main at 0x7fa05403a0>
Apr 08 17:29:17 raspberrypi python[742]:            └ <Command main>
Apr 08 17:29:17 raspberrypi python[742]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/click/core.py", line 1055, in main
Apr 08 17:29:17 raspberrypi python[742]:     rv = self.invoke(ctx)
Apr 08 17:29:17 raspberrypi python[742]:          │    │      └ <click.core.Context object at 0x7fa0e39040>
Apr 08 17:29:17 raspberrypi python[742]:          │    └ <function Command.invoke at 0x7fa053e8b0>
Apr 08 17:29:17 raspberrypi python[742]:          └ <Command main>
Apr 08 17:29:17 raspberrypi python[742]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
Apr 08 17:29:17 raspberrypi python[742]:     return ctx.invoke(self.callback, **ctx.params)
Apr 08 17:29:17 raspberrypi python[742]:            │   │      │    │           │   └ {'port': 11111, 'host': '0.0.0.0', 'root_path': '/api', 'app': 'app.main:app', 'uds': None, 'fd': None, 'reload': False, 'rel...
Apr 08 17:29:17 raspberrypi python[742]:            │   │      │    │           └ <click.core.Context object at 0x7fa0e39040>
Apr 08 17:29:17 raspberrypi python[742]:            │   │      │    └ <function main at 0x7fa02a4f70>
Apr 08 17:29:17 raspberrypi python[742]:            │   │      └ <Command main>
Apr 08 17:29:17 raspberrypi python[742]:            │   └ <function Context.invoke at 0x7fa0540af0>
Apr 08 17:29:17 raspberrypi python[742]:            └ <click.core.Context object at 0x7fa0e39040>
Apr 08 17:29:17 raspberrypi python[742]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke
Apr 08 17:29:17 raspberrypi python[742]:     return __callback(*args, **kwargs)
Apr 08 17:29:17 raspberrypi python[742]:                        │       └ {'port': 11111, 'host': '0.0.0.0', 'root_path': '/api', 'app': 'app.main:app', 'uds': None, 'fd': None, 'reload': False, 'rel...
Apr 08 17:29:17 raspberrypi python[742]:                        └ ()
Apr 08 17:29:17 raspberrypi python[742]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/main.py", line 404, in main
Apr 08 17:29:17 raspberrypi python[742]:     run(
Apr 08 17:29:17 raspberrypi python[742]:     └ <function run at 0x7fa029adc0>
Apr 08 17:29:17 raspberrypi python[742]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/main.py", line 569, in run
Apr 08 17:29:17 raspberrypi python[742]:     server.run()
Apr 08 17:29:17 raspberrypi python[742]:     │      └ <function Server.run at 0x7fa03c58b0>
Apr 08 17:29:17 raspberrypi python[742]:     └ <uvicorn.server.Server object at 0x7fa0dab4f0>
Apr 08 17:29:17 raspberrypi python[742]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/uvicorn/server.py", line 60, in run
[... cut ...]
Apr 08 17:29:17 raspberrypi python[742]:   File "./app/lightning/impl/cln_jrpc.py", line 200, in list_all_tx
Apr 08 17:29:17 raspberrypi python[742]:     for invoice in res[0]:
Apr 08 17:29:17 raspberrypi python[742]:                    └ [None, None, None, None]
Apr 08 17:29:17 raspberrypi python[742]: TypeError: 'NoneType' object is not iterable
Apr 08 17:29:17 raspberrypi python[742]:
Apr 08 17:29:17 raspberrypi python[742]: 2023-04-08 17:29:17 | ℹ️ | logging.py:87 | 127.0.0.1:48778 - "GET /lightning/list-all-tx?reversed=true HTTP/1.0" 500
```

closed 2023-05-10 by rootzoll