raspiblitz/raspiblitz #3735: lnd.tlscert.sh

Smiril, 2023-04-04:
please where do you gernerate the cert???
there is NO generation but checking ....

'''

### REFRESH

if [ "$1" = "refresh" ]; then 

  echo "# checking if LND is running"
  lndEnabled=$(sudo systemctl status lnd | grep -c 'enabled')
  if [ ${lndEnabled} -eq 0 ]; then
    echo "# FAIL: lnd.tlscert.sh refresh"
    echo "error='LND systemd service not enabled'"
    exit 1
  fi

  echo "# making sure services are not running"
  sudo systemctl stop lnd 2>/dev/null

  echo "# keep old tls data as backup"
  sudo rm ${TLSPATH}/tls.cert.old 2>/dev/null
  sudo rm ${TLSPATH}/tls.key.old 2>/dev/null
  sudo mv ${TLSPATH}/tls.cert ${TLSPATH}/tls.cert.old 
  sudo mv ${TLSPATH}/tls.key ${TLSPATH}/tls.key.old 

  echo "# start to create new generate new TLSCert"
  sudo systemctl start lnd
  echo "# wait until generated"
  newCertExists=0
  count=0
  while [ ${newCertExists} -eq 0 ]
  do
    count=$(($count + 1))
    echo "# (${count}/60) check for cert"
    if [ ${count} -gt 60 ]; then
      sudo systemctl stop lnd
      echo "error='failed to generate new LND cert'"
      exit 1
    fi
    newCertExists=$(sudo ls /mnt/hdd/lnd/tls.cert 2>/dev/null | grep -c '.cert')
    sleep 2
  done

  # stop lnd and let outside decide to restart or not
  sudo systemctl stop lnd
  sudo chmod 664 ${TLSPATH}/tls.cert
  sudo chown bitcoin:bitcoin "/mnt/hdd/lnd/tls.cert"

  echo "# symlink new cert to lnd app-data directory"
  if ! [[ -L "/mnt/hdd/app-data/lnd/tls.cert" ]]; then
    sudo rm -rf "/mnt/hdd/app-data/lnd/tls.cert"               # not a symlink.. delete it silently
    sudo ln -s ${TLSPATH}/tls.cert /home/admin/.lnd/tls.cert   # and create symlink
  fi
  echo "# OK TLS certs are fresh - start of LND service needed"
  exit
fi
''''

openoms, 2023-04-05:
you can reset the cert in the CONNECT menu

Smiril, 2023-04-08:
### lnd.credentials.sh reset mainnet
## Resetting Macaroons
# all your macaroons get deleted and recreated
find: ‘/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/’: No such file or directory
rm: cannot remove '/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/macaroons.db': No such file or directory
## Resetting TLS
# tls cert gets deleted and recreated
# checking if LND is running
# making sure services are not running
# keep old tls data as backup
mv: cannot stat '/mnt/hdd/lnd/tls.cert': No such file or directory
mv: cannot stat '/mnt/hdd/lnd/tls.key': No such file or directory
# start to create new generate new TLSCert
A dependency job for lnd.service failed. See 'journalctl -xe' for details.
# wait until generated
# (1/60) check for cert
# (2/60) check for cert
# (3/60) check for cert
# (4/60) check for cert
# (5/60) check for cert
# (6/60) check for cert
# (7/60) check for cert
# (8/60) check for cert
# (9/60) check for cert
# (10/60) check for cert
# (11/60) check for cert
# (12/60) check for cert
# (13/60) check for cert
# (14/60) check for cert
# (15/60) check for cert
# (16/60) check for cert
# (17/60) check for cert
# (18/60) check for cert
# (19/60) check for cert
# (20/60) check for cert
# (21/60) check for cert
# (22/60) check for cert
# (23/60) check for cert
# (24/60) check for cert
# (25/60) check for cert
# (26/60) check for cert
# (27/60) check for cert
# (28/60) check for cert
# (29/60) check for cert
# (30/60) check for cert
# (31/60) check for cert
# (32/60) check for cert
# (33/60) check for cert
# (34/60) check for cert
# (35/60) check for cert
# (36/60) check for cert
# (37/60) check for cert
# (38/60) check for cert
# (39/60) check for cert
# (40/60) check for cert
# (41/60) check for cert
# (42/60) check for cert
# (43/60) check for cert
# (44/60) check for cert
# (45/60) check for cert
# (46/60) check for cert
# (47/60) check for cert
# (48/60) check for cert
# (49/60) check for cert
# (50/60) check for cert
# (51/60) check for cert
# (52/60) check for cert
# (53/60) check for cert
# (54/60) check for cert
# (55/60) check for cert
# (56/60) check for cert
# (57/60) check for cert
# (58/60) check for cert
# (59/60) check for cert
# (60/60) check for cert
# (61/60) check for cert
error='failed to generate new LND cert'
# restarting LND ... wait 10 secs
A dependency job for lnd.service failed. See 'journalctl -xe' for details.

#is the result

openoms, 2023-04-08:
your lnd is not running See the output of:
```
sudo journalctl -fu lnd - n50
```
also can always check the lnd logs in the SYSTEM menu

Smiril, 2023-04-09:
thanks for support

closed 2023-04-09 by Smiril