raspiblitz/raspiblitz #3753: Bitcoin Install Fails during build_sdcard.sh due to PGP key verification

flyerbear, 2023-04-13:
System configuration:
HP ProDesk 600 G3 with i5-7500, 40GB RAM, 256GB SSD for OS, ZFS pool of 2TB NVMe and 1.92TB SSD for data disk
Debian 11.6.0 fresh install

After installing only Debian 11 OS and ZFS, ran `build_sdcard.sh` as follows:

```
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/build_sdcard.sh
bash build_sdcard.sh --github-user rootzoll --branch dev --display hdmi --wifi-region off
```

Script failed at the following point:
```
*** PREPARING BITCOIN ***
gpg: directory '/root/.gnupg' created
gpg: keybox '/root/.gnupg/pubring.kbx' created
gpg: keyserver receive failed: Server indicated a failure
# FAIL # Couldn't download Wladimir J. van der Laan's PGP pubkey
```

I checked the exit code with `echo $?` and it shows `1`.

I tried running `gpg --keyserver hkp://keyserver.ubuntu.com --recv-key "71A3 B167 3540 5025 D447 E8F2 7481 0B01 2346 C9A6"` in the terminal and it failed with the error `gpg: keyserver receive failed: Server indicated a failure`.

Then I rebooted the system and ran the same command, `gpg --keyserver hkp://keyserver.ubuntu.com --recv-key "71A3 B167 3540 5025 D447 E8F2 7481 0B01 2346 C9A6"`, which succeeded in downloading the key.

However, running `build_sdcard.sh` failed again in the same way, and re-running `gpg --keyserver hkp://keyserver.ubuntu.com --recv-key "71A3 B167 3540 5025 D447 E8F2 7481 0B01 2346 C9A6"` _**after**_ `build_sdcard.sh` caused the keyserver error message again.

It seems that something in the `build_sdcard.sh` script is preventing gpg from downloading the keys properly.

In case it is helpful, I had previously seen similar behavior when testing out the `build_sdcard.sh` script in a VM running Debian 11, and the issue seemed to be with tor - that tor was not running - but I did not get the same tor-related error messages this time.

openoms, 2023-04-14:
We should use wget to get the key(s) and have options more in the update as well.

See this issue for a working method: https://github.com/bitcoin/bitcoin/issues/25395

flyerbear, 2023-04-14:
I may have found a solution to this issue. Based on https://unix.stackexchange.com/questions/399027/gpg-keyserver-receive-failed-server-indicated-a-failure, adding port 80 to the key server (so hkp://keyserver.ubuntu.com:80) should solve the issue.

I created a custom `bitcoin.install.sh` with this fix, and it got past the failure point, though it hit other, unrelated errors later on and did not complete.

If there are other locations in the code that get keys the same way, they may need to have the port added as well.

PR#3754 created to address this.

cercatrova21, 2023-04-19:
I think this also looks similar 

https://github.com/rootzoll/raspiblitz/issues/3679

closed 2023-12-01 by rootzoll