raspiblitz/raspiblitz #3759: logrotate may not properly rotate log files / rsyslogd not propery notified of log file rotation

cricktor, 2023-04-15:
I have some evidence that with certain entries in current config files in `/etc/logrotate.d/` rsyslogd isn't properly notified when a log file rotation actually happened. The way rsyslogd should be signaled that a new fresh log file has been created doesn't seem to work with the `postrotate ... endscript` option.

This is what I observed when running RaspiBlitz for a long time without a reboot. Example log file is syslog (same for daemon.log) in `/var/log` which can easily grow quite big.
When logrotate kicks in it moves `syslog` to `syslog.1` (I omit further rotation of more existing retention files for `syslog`) and creates a new 0 Byte file `syslog`, then rsyslogd has to be notified to use the new file `syslog` as new log target file. This is done with the following statement in the config file `/etc/logrotate.d/rsyslog`
```
        postrotate
                invoke-rc.d rsyslog rotate > /dev/null
        endscript
```
As this doesn't seem to work (it returns an error if called without the redirection to `/dev/null`), rsyslogd continues to log into `syslog.1` and the file `syslog` stays empty. This empty file is a problem because the logrotate option `notifempty` is in place and instructs to do nothing when the presumed active log file is empty. So, while `syslog.1` still grows it's not rotated down anymore.

This issue is postponed to next logrotate invocation when a reboot happens as then the empty log files get in use again. But the problem comes up again when new empty log files are rotated in existence. It can become a severe issue if the microSD card is rather small and the user doesn't pay attention to free space on the microSD card and leaves his RaspiBlitz running unattended.

The working fix is to change the postrotate statement in the logrotate config files where appropriate to
```
        postrotate
                /usr/lib/rsyslog/rsyslog-rotate
        endscript
```
This properly notifies rsyslogd about the log file rotation and the new log files are then uses and filled and subsequent logrotate invocations are not hindered by zero length log files.

rootzoll, 2023-04-18:
On which version have you tested this? Did you tried with the latest v1.9.0 release candidates available on dev branch? https://github.com/rootzoll/raspiblitz/tree/dev#downloading-the-software

rootzoll, 2023-05-03:
v1.7.2 is quite an old version. There were some updates on the log rotates in recent updates - so I will close this issue. But if you or someone else can still find problems with that on recent releases - please reopen.

closed 2023-05-03 by rootzoll