raspiblitz/raspiblitz #3763: Raspberry Pi OS 64bit Lite.

dipunm, 2023-04-18:
I noticed that the docs show that you should start with a raspberry pi OS 64 bit for desktop.

I expect that this is because the Lite version has not been tested, therefore I want to do that. (raspberry pi 4 model b)

I will be running through the scripts, manually running the steps and documenting any issues I find along the way. The only thing I would expect is that I will need to install missing packages since we are starting from lite.

I do not have a display, so I will be running it headless.

Questions:

1. Is it worth doing this or has the Lite version been tested already?
2. Any thing in particular I should test?
3. Any tips/advice?

I will be starting with blockchain data on my SSD, I am not going to download that from scratch.

rootzoll, 2023-04-18:
@openoms as far as I remember you did last time test with the lite version. Any advice on this?

openoms, 2023-04-18:
Could work if not trying to use a display.

Previous attempts:
https://github.com/rootzoll/raspiblitz/issues/1184
https://github.com/rootzoll/raspiblitz/issues/1867

dipunm, 2023-04-18:
in build.sh, calling:
```
sudo update-alternatives --install /usr/bin/pip pip /usr/bin/pip3 1
```

results in the following warning:
```
update-alternatives: warning: not replacing /usr/bin/pip with a link

# Further info:
pi@raspberrypi:~ $ ls -la /usr/bin/pip*
-rwxr-xr-x 1 root root 941 Jul 13  2022 /usr/bin/pip
-rwxr-xr-x 1 root root 943 Jul 13  2022 /usr/bin/pip3
pi@raspberrypi:~ $ /usr/bin/pip --version
pip 20.3.4 from /usr/lib/python3/dist-packages/pip (python 3.9)
pi@raspberrypi:~ $ /usr/bin/pip3 --version
pip 20.3.4 from /usr/lib/python3/dist-packages/pip (python 3.9)
```

- is it worth checking for the presence of /usr/bin/pip?
- should we verify that it is the correct version as part of the script?
- should we delete the file and run again to ensure that this line has a meaningful effect?

dipunm, 2023-04-18:
The following lines:

```
# replace boot splash image when raspbian
[ "${baseimage}" = "raspios_arm64" ] && { echo "* replacing boot splash"; sudo cp /home/admin/raspiblitz/pictures/splash.png /usr/share/plymouth/themes/pix/splash.png; }
```

Results in an error because `plymouth` was never installed:

```
* replacing boot splash
cp: cannot create regular file '/usr/share/plymouth/themes/pix/splash.png': No such file or directory
```

dipunm, 2023-04-18:
In this section:

```
  # disable DRM VC4 V3D
  echo -e "\n*** DISABLE DRM VC4 V3D driver ***"
  dtoverlay=vc4-fkms-v3d
  sudo sed -i "s/^dtoverlay=${dtoverlay}/# dtoverlay=${dtoverlay}/g" /boot/config.txt
```

I noticed that dtoverlay in my `/boot/config.txt` file was `vc4-kms-v3d` (missing the f) unsure if this is a typo or not?

```
pi@raspberrypi:/home/admin $ cat /boot/config.txt | grep dtoverlay
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18
dtoverlay=vc4-kms-v3d
dtoverlay=pi3-disable-bt
dtoverlay=disable-bt
```

dipunm, 2023-04-19:
@frennkie I saw a TODO in `blitz.web.sh`.
```
# general nginx settings
  sudo sed -i "s/^user www-data;/#user www-data;/g" /etc/nginx/nginx.conf
  if ! grep -Eq '^\s*server_names_hash_bucket_size.*$' /etc/nginx/nginx.conf; then
    # ToDo(frennkie) verify this
    sudo sed -i -E '/^.*server_names_hash_bucket_size [0-9]*;$/a \\tserver_names_hash_bucket_size 128;' /etc/nginx/nginx.conf
  fi
```

Here is the before and after of `/etc/nginx/nginx.conf`:
<details>
<summary>Before</summary>
```
user www-data;
worker_processes auto;
pid /run/nginx.pid;
include /etc/nginx/modules-enabled/*.conf;

events {
	worker_connections 768;
	# multi_accept on;
}

http {

	##
	# Basic Settings
	##

	sendfile on;
	tcp_nopush on;
	types_hash_max_size 2048;
	# server_tokens off;

	# server_names_hash_bucket_size 64;
	# server_name_in_redirect off;

	include /etc/nginx/mime.types;
	default_type application/octet-stream;

	##
	# SSL Settings
	##

	ssl_protocols TLSv1 TLSv1.1 TLSv1.2 TLSv1.3; # Dropping SSLv3, ref: POODLE
	ssl_prefer_server_ciphers on;

	##
	# Logging Settings
	##

	access_log /var/log/nginx/access.log;
	error_log /var/log/nginx/error.log;

	##
	# Gzip Settings
	##

	gzip on;

	# gzip_vary on;
	# gzip_proxied any;
	# gzip_comp_level 6;
	# gzip_buffers 16 8k;
	# gzip_http_version 1.1;
	# gzip_types text/plain text/css application/json application/javascript text/xml application/xml application/xml+rss text/javascript;

	##
	# Virtual Host Configs
	##

	include /etc/nginx/conf.d/*.conf;
	include /etc/nginx/sites-enabled/*;
}

#mail {
#	# See sample authentication script at:
#	# http://wiki.nginx.org/ImapAuthenticateWithApachePhpScript
#
#	# auth_http localhost/auth.php;
#	# pop3_capabilities "TOP" "USER";
#	# imap_capabilities "IMAP4rev1" "UIDPLUS";
#
#	server {
#		listen     localhost:110;
#		protocol   pop3;
#		proxy      on;
#	}
#
#	server {
#		listen     localhost:143;
#		protocol   imap;
#		proxy      on;
#	}
#}
```
</details>
<details>
<summary>After</summary>
```
user www-data;
worker_processes auto;
pid /run/nginx.pid;
include /etc/nginx/modules-enabled/*.conf;

events {
	worker_connections 768;
	# multi_accept on;
}

http {

	##
	# Basic Settings
	##

	sendfile on;
	tcp_nopush on;
	types_hash_max_size 2048;
	# server_tokens off;

	# server_names_hash_bucket_size 64;
	server_names_hash_bucket_size 128;
	# server_name_in_redirect off;

	include /etc/nginx/mime.types;
	default_type application/octet-stream;

	##
	# SSL Settings
	##

	ssl_protocols TLSv1 TLSv1.1 TLSv1.2 TLSv1.3; # Dropping SSLv3, ref: POODLE
	ssl_prefer_server_ciphers on;

	##
	# Logging Settings
	##

	access_log /var/log/nginx/access.log;
	error_log /var/log/nginx/error.log;

	##
	# Gzip Settings
	##

	gzip on;

	# gzip_vary on;
	# gzip_proxied any;
	# gzip_comp_level 6;
	# gzip_buffers 16 8k;
	# gzip_http_version 1.1;
	# gzip_types text/plain text/css application/json application/javascript text/xml application/xml application/xml+rss text/javascript;

	##
	# Virtual Host Configs
	##

	include /etc/nginx/conf.d/*.conf;
	include /etc/nginx/sites-enabled/*;
}

#mail {
#	# See sample authentication script at:
#	# http://wiki.nginx.org/ImapAuthenticateWithApachePhpScript
#
#	# auth_http localhost/auth.php;
#	# pop3_capabilities "TOP" "USER";
#	# imap_capabilities "IMAP4rev1" "UIDPLUS";
#
#	server {
#		listen     localhost:110;
#		protocol   pop3;
#		proxy      on;
#	}
#
#	server {
#		listen     localhost:143;
#		protocol   imap;
#		proxy      on;
#	}
#}
```
<details>

dipunm, 2023-04-21:
I think this is incorrect, can someone confirm?

```
# (do last - because might trigger reboot)
if [ "${display}" != "headless" ] || [ "${baseimage}" = "raspios_arm64" ]; then
  echo "*** ADDITIONAL DISPLAY OPTIONS ***"
  echo "- calling: blitz.display.sh set-display ${display}"
  sudo /home/admin/config.scripts/blitz.display.sh set-display ${display}
  sudo /home/admin/config.scripts/blitz.display.sh rotate 1
fi
```

This is at the end of the `build_sdcard.sh` file, but I think it should read:
```
# (do last - because might trigger reboot)
if [ "${display}" != "headless" ] && [ "${baseimage}" = "raspios_arm64" ]; then
  echo "*** ADDITIONAL DISPLAY OPTIONS ***"
  echo "- calling: blitz.display.sh set-display ${display}"
  sudo /home/admin/config.scripts/blitz.display.sh set-display ${display}
  sudo /home/admin/config.scripts/blitz.display.sh rotate 1
fi
```

dipunm, 2023-04-21:
As mentioned in https://github.com/rootzoll/raspiblitz/issues/3724#issuecomment-1517064868, I noticed that there is a missing dependency for installing plugin cln-grpc... is it worth adding a fix PR for the current v1.8 branch?

dipunm, 2023-04-21:
Last thing, after a reboot, when trying to `ssh` in, I get this:
```bash
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ED25519 key sent by the remote host is
SHA256:+rix3YvznFb4w7xNUVJahdVBzDtX4dlDpCbOy1hdyE0.
Please contact your system administrator.
Add correct host key in /home/dipunm/.ssh/known_hosts to get rid of this message.
Offending ECDSA key in /home/dipunm/.ssh/known_hosts:20
Host key for raspberrypi.lan has changed and you have requested strict checking.
Host key verification failed.
```

I get that I just need to remove the lines in `known_hosts`, but what exactly causes this? I wasn't able to figure it out and is this something worth documenting or is it a bug?

rootzoll, 2023-11-14:
With RaspiIOs there are now lite images anymore: https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2023-05-03/ .. closing issue

closed 2023-11-14 by rootzoll