raspiblitz/raspiblitz #3766: Node is stuck on boot - Blockchain is 100% with Peers - Lightning stuck at Prepare Sync for 24 hours

mantiev, 2023-04-20:
Here is the log

**************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.8.0
chainnetwork: bitcoin / main
 14:22:42 up  1:43,  3 users,  load average: 1.77, 1.74, 1.73

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-04-20 12:50:10 BST; 1h 32min ago
    Process: 6432 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 6452 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 6458 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 6557 (bitcoind)
      Tasks: 18 (limit: 4531)
        CPU: 12min 33.329s
     CGroup: /system.slice/bitcoind.service
             └─6557 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Apr 20 12:41:15 raspberrypi systemd[1]: Starting Bitcoin daemon...
Apr 20 12:50:10 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Tue 2023-04-18 01:24:56 BST, ends at Thu 2023-04-20 14:22:42 BST. --
Apr 20 12:41:15 raspberrypi systemd[1]: Starting Bitcoin daemon...
Apr 20 12:50:10 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2023-04-20T12:22:32Z New outbound peer connected: version: 70016, blocks=786252, peer=68 (block-relay-only)
2023-04-20T12:23:55Z Socks5() connect to 3vsa3omspkk5fvl4fmm37yytr6isdextsis6hjnpfhchnvmhynwz4pqd.onion:8333 failed: host unreachable
2023-04-20T12:24:15Z Socks5() connect to bsm4qdggnu56grgd5bs7kam3l3l7gnwqj54vtqzajy3umbbpnt74r4ad.onion:8333 failed: host unreachable
2023-04-20T12:28:31Z Socks5() connect to ognubdrzmb2tnajptd6matxmvihnwo6l5mdfarxq7iqrkrtqhwwmpwyd.onion:8333 failed: host unreachable
2023-04-20T12:33:17Z Socks5() connect to ntkmvndauizie7lyklxghvuq6bbdkarjnsanyrgzace6pkbfl3p2whid.onion:8333 failed: host unreachable
2023-04-20T12:36:39Z New outbound peer connected: version: 70016, blocks=786252, peer=84 (block-relay-only)
2023-04-20T12:38:48Z New outbound peer connected: version: 70016, blocks=786252, peer=85 (block-relay-only)
2023-04-20T12:40:33Z UpdateTip: new best=000000000000000000018b75878ce10ef3f1c73028e2e9855b48fbdc40630b42 height=786253 version=0x29a2c000 log2_work=94.132145 tx=825714225 date='2023-04-20T12:39:56Z' progress=1.000000 cache=85.1MiB(636374txo)
2023-04-20T12:42:56Z UpdateTip: new best=000000000000000000013d6582cd6ee3d49892b9843969a81cfdb9f7a1deedd8 height=786254 version=0x319f2000 log2_work=94.132159 tx=825717148 date='2023-04-20T12:42:24Z' progress=1.000000 cache=85.6MiB(640959txo)
2023-04-20T12:50:18Z UpdateTip: new best=00000000000000000000675776ff94eb695040aa35a463b05030a0800c6707b4 height=786255 version=0x27ffe000 log2_work=94.132173 tx=825719627 date='2023-04-20T12:49:45Z' progress=1.000000 cache=86.3MiB(646625txo)
2023-04-20T12:56:30Z UpdateTip: new best=00000000000000000002e61a4b805abd8cbdbcad15724be88399d420c903592a height=786256 version=0x2000a000 log2_work=94.132187 tx=825722322 date='2023-04-20T12:56:10Z' progress=1.000000 cache=87.0MiB(651759txo)
2023-04-20T12:59:09Z New outbound peer connected: version: 70016, blocks=786256, peer=109 (outbound-full-relay)
2023-04-20T13:00:31Z New outbound peer connected: version: 70016, blocks=786256, peer=111 (block-relay-only)
2023-04-20T13:02:41Z Socks5() connect to auet46akookzvb3uqtqxclhtgthgmjic73vge4ako4g6w3yy6nx2xrid.onion:8333 failed: host unreachable
2023-04-20T13:09:35Z Socks5() connect to 5h4mnfvoy5cwjyi2msedxigcluepdmklxglgrl5utrvqni7xjmf3grad.onion:8333 failed: host unreachable
2023-04-20T13:14:10Z New outbound peer connected: version: 70016, blocks=786256, peer=129 (block-relay-only)
2023-04-20T13:18:54Z UpdateTip: new best=000000000000000000012dd3e8262d25b16330e7682f31499f335caa2f3ed6b3 height=786257 version=0x2000a000 log2_work=94.132201 tx=825725855 date='2023-04-20T13:18:28Z' progress=1.000000 cache=88.6MiB(664170txo)
2023-04-20T13:20:16Z Socks5() connect to vrbm3va6ueipesq5e2dmn3rpwxbb6ikk4unwu4lxjxrzxnchwrdmysqd.onion:8333 failed: host unreachable
2023-04-20T13:20:50Z Socks5() connect to bw66yqhqy46o2eowv6pmkgm73vvcfvvssoom7qvvdv4erhdwmaxoxbid.onion:8333 failed: host unreachable
2023-04-20T13:21:15Z Socks5() connect to 5bxn4pameych67xkx5azxpvpf2k5axdrsizdcl6omzrcg6kfj5ysimqd.onion:8333 failed: host unreachable

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Thu 2023-04-20 14:21:44 BST; 58s ago
    Process: 629891 ExecStartPre=/home/admin/config.scripts/lnd.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
    Process: 630111 ExecStart=/usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.conf (code=exited, status=1/FAILURE)
   Main PID: 630111 (code=exited, status=1/FAILURE)
        CPU: 2.671s
[... 309 lines cut ...]
btc_default_synced="1"
btc_default_peers="15"
btc_default_sync_initialblockdownload="0"
btc_default_port="8333"
btc_default_blocks_verified="786257"
btc_default_blocks_data_kb="527632912"
btc_default_sync_initial_done="1"
btc_default_blocks_headers="786257"
btc_default_sync_progress="0.9999992892556434"
btc_default_sync_percentage="100.00"
btc_default_activated="1"
btc_default_mempool_transactions="4113"

*** LOGFILES ***
Archived and active journals take up 316.0M in the file system.
729M	/var/log

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

Press ENTER to return to main menu.

mantiev, 2023-05-01:
So no clue as to why this is happening?  No suggestions as to why I have to watch my node like a toddler because things just stop working all the time....

cstenglein, 2023-06-07:
@mantiev

Your issue seems to be this:

>  2023-04-20 14:21:44.678 [ERR] LTND: Shutting down because error in main method: unable to open databases: unable to open graph DB: invalid edge key 00000000643d10b80abec30006130001 encountered

See https://github.com/lightningnetwork/lnd/issues/6058 for a solution

Closing this for now, if you still have problems please re-open.

closed 2023-06-07 by cstenglein