raspiblitz/raspiblitz #3772: Public IP Detection Fails

srw777, 2023-04-23:
Yes, I have read the previous issues that relate to this.  I don't think I'm seeing the same thing.  Running 1.8.0c clearnet.  Behind NAT with a reserved DHCP IP. (so, effectively static) and a static IP from my ISP.  After being frustrated with not seeing my public ip on the display (only 127.0.0.1) and not finding any real answers on google, I discovered I could edit /mnt/hdd/raspiblitz.conf and set my public IP.  I did this.  I opened some channels.  I sent some transactions.  Then I discovered my IP had reverted to 127.0.0.1.  I checked raspiblitz.conf and saw it had been changed back to 127.0.0.1.  So, I kept digging to figure out what might be causing this.

In the '/home/admin/config.scripts/internet.sh' script I found the 'curl' command to read from v4.ipv6-test.com.

If I run './internet.sh' global it sometimes returns my IP and other times returns 127.0.0.1.

If I run the curl command directly on the command line, I can see that often it returns a blank result. (with a 200 response)  It sometimes gets the correct external ip.

PatrickScheich, 2023-05-05:
as far as i remember the `/home/admin/config.scripts/internet.sh status global`  uses curl with the `-m 5` parameter which sets the timeout to 5 seconds. 
So if the website needs more the 5 seconds to answer it returned an error which will result in `127.0.0.1` 

You may try to increase the timeout a bit, like `-m 10` that may help

rootzoll, 2023-05-08:
increaded timeout will be part of next v1.9 release. Please feedback if this helped.

srw777, 2023-05-08:
> 

Increased timeout is not going to solve the issue I saw.  The remote server WAS responding quickly and returned a blank result instead of an IP address.  The script then used 127.0.0.1.  It didn't actually seem to be an issue with the internet.sh script (except for, perhaps, bad handling of the blank response) but more likely with the remote server or a transparent proxy somewhere between them and me.  I hardcoded my static IP into this script, so I won't see this issue again until I upgrade.

PatrickScheich, 2023-05-16:
there is an option to find the external IPv4 or IPv6 address using a special DNS query

[https://www.cyberciti.biz/faq/how-to-find-my-public-ip-address-from-command-line-on-a-linux/](https://www.cyberciti.biz/faq/how-to-find-my-public-ip-address-from-command-line-on-a-linux/)

```

# Show your public IPv4 using dig and DNS resolver
dig +short myip.opendns.com @resolver1.opendns.com
dig TXT +short o-o.myaddr.l.google.com @ns1.google.com
dig +short txt ch whoami.cloudflare @1.0.0.1

# Show your public IPv6 using dig and DNS resolver
dig -6 TXT +short o-o.myaddr.l.google.com @ns1.google.com

# You can try host command
host myip.opendns.com resolver1.opendns.com

```