raspiblitz/raspiblitz #3773: v1.8.0c: stuck at Lightning Progress  :  0 %

jalbstmeijer, 2023-04-24:
Hi,

```
Blockchain Progress : 100.00 % 14 peers 
Lightning Progress  :  0 % 
```
This has been the state now for a couple of days.

Here the debug info; https://l.termbin.com/0h89k

rootzoll, 2023-04-25:
Its a core lightning @openoms can you take a quick look?

openoms, 2023-04-25:
logs look fine.
can you type:
```
menu
``` 
and enter CLN @jalbstmeijer ?

a simple restart might set it straight also

jalbstmeijer, 2023-04-27:
@openoms

Still the same state.

Just an idea. I remember the raspiblitz web gui gave me (I suppose unintentionally) access to the dashboard while the bitcoin chain was still syncing. I played around and created a 20 SAS invoice.
Could that be in the way?

Any suggestion what I could try? Maybe reset/create a new lightning wallet?

openoms, 2023-04-27:
Is the problem that you not get entered to the menu?
what does
```
cllog
```
show? Is your CLN wallet working when you type:
```
cl getinfo
```
If it is unused you are safe to reset the CLN wallet from the CLN - REPAIR-CL menu.

jalbstmeijer, 2023-04-28:
@openoms

I kept an eye on /mnt/hdd/app-data/.lightning/bitcoin/cl.log
When "Adding block" reached the Bitcoin Blocks amount, it seems to have finished.
Would be nice if this progress would be seen on the sync progress screen too.

Now when I log in to ssh, I get into the main menu instead of the sync progress screen.

Only thing is that the web gui is not able to complete loading now, as nginx encounters timeouts;
```
2023/04/28 12:34:00 [error] 839#839: *571 upstream timed out (110: Connection timed out) while reading response header from upstream, client: xxx, server: _, request: "GET /api/v1/lightning/list-all-tx?reversed=true HTTP/1.1", upstream: "http://127.0.0.1:11111/v1/lightning/list-all-tx?reversed=true", host: "xxx", referrer: "http://xxx/home"
2023/04/28 12:34:07 [error] 839#839: *585 upstream timed out (110: Connection timed out) while reading response header from upstream, client: xxx, server: _, request: "GET /api/v1/setup/status HTTP/1.1", upstream: "http://127.0.0.1:11111/v1/setup/status", host: "xxx", referrer: "http://xxx/home"

```

Should I open a separate ticket for that?

jalbstmeijer, 2023-04-29:
@openoms

As web gui loadng issues did not resolve after waiting for a whole.
I rebooted the raspiblitz,and ended up back exactly where this ticket started.

Lightning Progress  :  0 %

and cllog Adding block's

So I have no idea where I'm waiting for and when to expect a stable running raspiblitz

openoms, 2023-04-29:
Not sure why your node is going into rescanning again.
Can try to remove the line starting with

```
rescan=
```
from the cl.conf

jalbstmeijer, 2023-04-29:
it seems to work after a new flash. This because I upgraded the OS (apt-get update), causing dependency issues see https://github.com/rootzoll/raspiblitz/issues/3779

closed 2023-04-29 by jalbstmeijer