raspiblitz/raspiblitz #3775: My node refuses to reboot - Failed to start Bitcoin daemon - pls helb

KanutoTakumi, 2023-04-25:
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.8.0
chainnetwork: bitcoin / main
 20:24:20 up 3 min,  2 users,  load average: 2.48, 1.37, 0.55

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-04-25 20:22:39 BST; 1min 40s ago
    Process: 6855 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 6866 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 6869 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=1/FAILURE)
        CPU: 709ms

Apr 25 20:22:39 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Apr 25 20:22:39 raspberrypi systemd[1]: Failed to start Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Mon 2023-04-24 05:39:35 BST, ends at Tue 2023-04-25 20:24:20 BST. --
Apr 25 20:22:39 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Apr 25 20:22:39 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Apr 25 20:22:39 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
Apr 25 20:22:39 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 5.
Apr 25 20:22:39 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Apr 25 20:22:39 raspberrypi systemd[1]: bitcoind.service: Start request repeated too quickly.
Apr 25 20:22:39 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Apr 25 20:22:39 raspberrypi systemd[1]: Failed to start Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2023-04-25T19:22:38Z Config file arg: [test] rpcport="18332"
2023-04-25T19:22:38Z Command-line arg: conf="/mnt/hdd/bitcoin/bitcoin.conf"
2023-04-25T19:22:38Z Command-line arg: daemonwait=""
2023-04-25T19:22:38Z Command-line arg: datadir="/mnt/hdd/bitcoin"
2023-04-25T19:22:38Z Command-line arg: debuglogfile="/mnt/hdd/bitcoin/debug.log"
2023-04-25T19:22:38Z Using at most 40 automatic connections (1024 file descriptors available)
2023-04-25T19:22:38Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2023-04-25T19:22:38Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2023-04-25T19:22:38Z Script verification uses 3 additional threads
2023-04-25T19:22:38Z Wallet disabled!
2023-04-25T19:22:38Z scheduler thread start
2023-04-25T19:22:38Z HTTP: creating work queue of depth 16
2023-04-25T19:22:38Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2023-04-25T19:22:38Z HTTP: starting 4 worker threads
2023-04-25T19:22:38Z Using /16 prefix for IP bucketing
2023-04-25T19:22:38Z init message: Loading P2P addresses…
2023-04-25T19:22:39Z Error: Invalid or corrupt peers.dat (Checksum mismatch, data corrupted). If you believe this is a bug, please report it to https://github.com/bitcoin/bitcoin/issues. As a workaround, you can move the file ("/mnt/hdd/bitcoin/peers.dat") out of the way (rename, move, or delete) to have a new one created on the next start.
2023-04-25T19:22:39Z Shutdown: In progress...
2023-04-25T19:22:39Z scheduler thread exit
2023-04-25T19:22:39Z Shutdown: done

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: inactive (dead)

Apr 25 20:22:34 raspberrypi systemd[1]: Dependency failed for Lightning Network Daemon.
Apr 25 20:22:34 raspberrypi systemd[1]: lnd.service: Job lnd.service/start failed with result '…ency'.
Hint: Some lines were ellipsized, use -l to show in full.
[... 313 lines cut ...]
ln_default_activated="1"
btc_default_sync_initial_done="1"
btc_default_activated="1"
btc_default_sync_percentage="0"
btc_default_ready="0"
btc_default_running="0"
btc_default_online="0"
btc_default_error_full=""
btc_default_peers="0"
btc_default_port=""
btc_default_version="v23.0.0"
btc_default_error_short=""

*** LOGFILES ***
Archived and active journals take up 304.0M in the file system.
889M	/var/log

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

Press ENTER to return to main menu.

KanutoTakumi, 2023-04-25:
Ok, this is embarrassing, I should have read the log better.

just delete the old peer.dat and restart.

`sudo mv /mnt/hdd/bitcoin/peers.dat /mnt/hdd/bitcoin/peers.dat.old`

and

`sudo reboot`

sorry for bothering, ser

rootzoll, 2023-04-26:
Thanks for reporting anyway - lets keep this issue open as an example for the peer.dat error message to be part of the future to be build auto errorn detection:

```
2023-04-25T19:22:39Z Error: Invalid or corrupt peers.dat (Checksum mismatch, data corrupted). If you believe this is a bug, please report it to https://github.com/bitcoin/bitcoin/issues. As a workaround, you can move the file ("/mnt/hdd/bitcoin/peers.dat") out of the way (rename, move, or delete) to have a new one created on the next start.
```