raspiblitz/raspiblitz #3777: BTCPayserver 502 Bad Gateway - migration to postgres

KanutoTakumi, 2023-04-28:
Hey,

I updated btcpay today a bit recklessly. Since then the server is not accessible, if I understand the log data correctly, the database must be aktuallisiert.

I have found this:

https://github.com/rootzoll/raspiblitz/pull/3756

Is there any way to get back online in the short term without the 1.9.1 update?

thx for help.

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.8.0
chainnetwork: bitcoin / main
 06:59:12 up 46 min,  2 users,  load average: 6.44, 4.49, 4.25

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2023-04-28 06:14:40 BST; 44min ago
    Process: 6269 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 6287 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 6290 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 6397 (bitcoind)
      Tasks: 18 (limit: 4531)
        CPU: 30min 13.252s
     CGroup: /system.slice/bitcoind.service
             └─6397 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Apr 28 06:14:14 raspberrypi systemd[1]: Starting Bitcoin daemon...
Apr 28 06:14:40 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Wed 2023-04-26 16:16:43 BST, ends at Fri 2023-04-28 06:59:12 BST. --
Apr 28 06:14:14 raspberrypi systemd[1]: Starting Bitcoin daemon...
Apr 28 06:14:40 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2023-04-28T05:16:22Z New outbound peer connected: version: 70016, blocks=787314, peer=6 (outbound-full-relay)
2023-04-28T05:16:30Z Socks5() connect to 6rjyz5obeg3tnqxbtgmnxqe42ze7ibeev6sltyyqed76cdqr2xwxiuid.onion:8333 failed: host unreachable
2023-04-28T05:16:38Z New outbound peer connected: version: 70016, blocks=787314, peer=7 (outbound-full-relay)
2023-04-28T05:16:41Z New outbound peer connected: version: 70016, blocks=787314, peer=8 (outbound-full-relay)
2023-04-28T05:16:47Z Socks5() connect to kwrllp6kjzzl4xvx36brie3ona6lmechbwcu5usgw3osaotnunzdjlid.onion:8333 failed: host unreachable
2023-04-28T05:16:56Z New outbound peer connected: version: 70016, blocks=787314, peer=9 (outbound-full-relay)
2023-04-28T05:17:07Z New outbound peer connected: version: 70016, blocks=787314, peer=11 (outbound-full-relay)
2023-04-28T05:17:24Z New outbound peer connected: version: 70016, blocks=787314, peer=12 (outbound-full-relay)
2023-04-28T05:22:44Z New outbound peer connected: version: 70016, blocks=787314, peer=14 (block-relay-only)
2023-04-28T05:24:21Z Imported mempool transactions from disk: 70915 succeeded, 0 failed, 0 expired, 104 already there, 0 waiting for initial broadcast
2023-04-28T05:24:21Z loadblk thread exit
2023-04-28T05:25:00Z Socks5() connect to l472yhrqkt4ylcpjtevt4smm7yow4jvqeeekeui7gkvadffrxieltyyd.onion:8333 failed: host unreachable
2023-04-28T05:27:08Z New outbound peer connected: version: 70016, blocks=787314, peer=18 (block-relay-only)
2023-04-28T05:34:44Z New outbound peer connected: version: 70016, blocks=787314, peer=29 (block-relay-only)
2023-04-28T05:39:36Z Socks5() connect to jlpmwdredbx7abrlixby2y2e5duouvnohbgftmoq3mjp4sttcezc2bid.onion:8333 failed: host unreachable
2023-04-28T05:47:10Z Potential stale tip detected, will try using extra outbound peer (last tip update: 1902 seconds ago)
2023-04-28T05:47:32Z New outbound peer connected: version: 70016, blocks=787314, peer=40 (outbound-full-relay)
2023-04-28T05:51:44Z Socks5() connect to 7sdqxwf3f63qaceuzs42fshifvktlttdcbmnm5y6au67tkkw2ow7z3id.onion:8333 failed: host unreachable
2023-04-28T05:57:41Z Potential stale tip detected, will try using extra outbound peer (last tip update: 2533 seconds ago)
[... 385 lines cut ...]
btc_default_sync_progress="0.9999998848775455"
btc_default_sync_initial_done="1"
btc_default_peers="14"
btc_default_ready="1"
btc_default_synced="1"
btc_default_mempool_transactions="76915"
btc_default_blocks_data_kb="529523476"
btc_default_blocks_behind="0"
btc_default_activated="1"
btc_default_sync_percentage="100.00"
btc_default_error_short=""
btc_default_blocks_headers="787315"

*** LOGFILES ***
Archived and active journals take up 396.0M in the file system.
889M	/var/log

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

Press ENTER to return to main menu.

rootzoll, 2023-04-28:
> I updated btcpay today a bit recklessly.

So this is not a general RaspiBlitz issue - its a personal edge case. It could possible fix itself if you install the upcoming v1.9.0 of Raspiblitz and you can try with the next ReleaseCandidate 4 (1.9.0rc4) that is targeted for next week.

KanutoTakumi, 2023-04-28:
Hey rootzoll, thanks for your comment.

because it is a personal problem, I have now uninstalled BTCPAY and reinstalled. During the uninstall I made sure that the old data is not deleted.
After the reinstallation BTCPAY was accessible again!

As far as I can see all data to the BTCPAYserver are still available. 

After a restart everything works again as before.

I am online again!

NicolasDorier, 2023-04-29:
The real root of this error is that since 1.9.0 btcpay refuses to start on sqlite or mysql.

```
Apr 28 06:15:23 raspberrypi dotnet[9537]: info: Configuration: Network: Mainnet
Apr 28 06:15:23 raspberrypi dotnet[9537]: fail: Configuration: SQLITE backend support is out of support. Please migrate to Postgres by following the following instructions (https://github.com/btcpayserver/btcpayserver/blob/master/docs/db-migration.md). If you don't want to update, you can try to start this instance by using the command line argument --deprecated
Apr 28 06:15:23 raspberrypi systemd[1]: btcpayserver.service: Succeeded.
```

openoms, 2023-05-01:
closing as resolved

closed 2023-05-01 by openoms