raspiblitz/raspiblitz #3779: v1.8.0c: errors when resetting wallet

jalbstmeijer, 2023-04-29:
As I'm struggling to get raspibliz running correctly in https://github.com/rootzoll/raspiblitz/issues/3773 I decided to reset my wallet through CLN->REPAIR->CL-RESET

The errors that follow don't make me very confident this will do any good.
Not sure I should have been able to input my choices for language and passphrases as I did not get a prompt to enter any.

```
The next step will overwrite the old Core Lightning mainnet wallet 
Press ENTER to continue or CTRL+C to abort

# Delete /home/bitcoin/.lightning/config
# Delete and recreate /home/bitcoin/.lightning/bitcoin
# Running 'cl.hsmtool.sh new mainnet'
--- Logging error ---
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 177, in emit
    self.console.print(renderable, overflow="ignore", crop=False, style=style)
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1752, in print
    extend(render(renderable, render_options))
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1390, in render
    for render_output in iter_render:
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 134, in __rich_console__
    for line in lines:
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/segment.py", line 245, in split_lines
    for segment in segments:
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/console.py", line 1368, in render
    renderable = rich_cast(renderable)
  File "/usr/local/lib/python3.9/dist-packages/pip/_vendor/rich/protocol.py", line 36, in rich_cast
    renderable = cast_method()
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/self_outdated_check.py", line 130, in __rich__
    pip_cmd = get_best_invocation_for_this_pip()
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/entrypoints.py", line 58, in get_best_invocation_for_this_pip
    if found_executable and os.path.samefile(
  File "/usr/lib/python3.9/genericpath.py", line 101, in samefile
    s2 = os.stat(f2)
FileNotFoundError: [Errno 2] No such file or directory: '/usr/bin/pip3.9'
Call stack:
  File "/usr/local/bin/pip", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/main.py", line 70, in main
    return command.main(cmd_args)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 101, in main
    return self._main(args)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/base_command.py", line 223, in _main
    self.handle_pip_version_check(options)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/cli/req_command.py", line 148, in handle_pip_version_check
    pip_self_version_check(session, options)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/self_outdated_check.py", line 237, in pip_self_version_check
    logger.info("[present-rich] %s", upgrade_prompt)
  File "/usr/lib/python3.9/logging/__init__.py", line 1442, in info
    self._log(INFO, msg, args, **kwargs)
  File "/usr/lib/python3.9/logging/__init__.py", line 1585, in _log
    self.handle(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 1595, in handle
    self.callHandlers(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 1657, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python3.9/logging/__init__.py", line 948, in handle
    self.emit(record)
  File "/usr/local/lib/python3.9/dist-packages/pip/_internal/utils/logging.py", line 179, in emit
    self.handleError(record)
Message: '[present-rich] %s'
Arguments: (UpgradePrompt(old='22.1.2', new='23.1.2'),)

# This file was placed by cl.hsmtool.sh
# Contains the seed words from which the hsm_secret in the same directory was generated from
seedwords='xxx'
seedwords6x4='xxx'
# Will be removed safely when the hsm_secret is encrypted.

Select your language:
[... 77 lines cut ...]
# Using the settings for: bitcoin mainnet
# *** INSTALL CORE LIGHTNING v0.11.2 BINARY ***
# only binary install to system
# no configuration, no systemd service
Core Lightning binary already installed - done
# Make sure bitcoin is in the debian-tor group
# Add plugin-dir: /home/bitcoin/cl-plugins-enabled
# Add plugin-dir: /home/bitcoin/cl-plugins-available
# Store the lightning data in /mnt/hdd/app-data/.lightning
# Symlink to /home/bitcoin/
ls: cannot access '/home/bitcoin/.lightning/config': No such file or directory
# Create /home/bitcoin/.lightning/config
# lightningd configuration for bitcoin mainnet

network=bitcoin
log-file=cl.log
log-level=info
plugin-dir=/home/bitcoin/cl-plugins-enabled

# Tor settings
proxy=127.0.0.1:9050
bind-addr=127.0.0.1:9736
addr=statictor:127.0.0.1:9051/torport=9736
always-use-proxy=true
```

openoms, 2023-04-29:
as you seem be running into a python dependency error a reflash is probably the beat courae of action to take.

The chain and other data will be preserved on the disk.

jalbstmeijer, 2023-04-29:
So, trying to keep the the base OS up2date with 'apt-get update' .. can be a show stopper for raspibliz?

One of the first things I do after an installation. I will try it without with a new flash.

jalbstmeijer, 2023-04-29:
Works now after a new flash to the same version. So it seems 'apt-get update' can brake the setup by introducing dependency issues with software installed or changes made outside the packaging system.

closed 2023-04-29 by jalbstmeijer