raspiblitz/raspiblitz #3792: JAM App: No connection to Backend

rootzoll, 2023-05-05:
This is how the JAM App looks on my v1.9.0rc4 after install:

[image]

openoms, 2023-05-06:
Can reproduce, but couldn't fin the cause so far.
[image]

Asked in https://t.me/JoinMarketWebUI/7223

openoms, 2023-05-06:
Likely a jm api issue, exploring further:

`curl -kv GET https://127.0.0.1:28183/api/v1/session`
```
* Could not resolve host: GET
* Closing connection 0
curl: (6) Could not resolve host: GET
*   Trying 127.0.0.1:28183...
* Connected to 127.0.0.1 (127.0.0.1) port 28183 (#1)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*  CAfile: /etc/ssl/certs/ca-certificates.crt
*  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / TLS_AES_256_GCM_SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: C=US; ST=Utah; L=Lehi; O=Your Company, Inc.; OU=IT; CN=example.com
*  start date: May  6 19:46:54 2023 GMT
*  expire date: May  3 19:46:54 2033 GMT
*  issuer: C=US; ST=Utah; L=Lehi; O=Your Company, Inc.; OU=IT; CN=example.com
*  SSL certificate verify result: self signed certificate (18), continuing anyway.
> GET /api/v1/session HTTP/1.1
> Host: 127.0.0.1:28183
> User-Agent: curl/7.74.0
> Accept: */*
> 
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* old SSL session ID is stale, removing
* Mark bundle as not supporting multiuse
< HTTP/1.1 405 Method Not Allowed
< Server: TwistedWeb/22.4.0
< Date: Sat, 06 May 2023 19:51:04 GMT
< Content-Type: text/html; charset=utf-8
< Allow: OPTIONS, HEAD, GET
< Content-Length: 153
< 
<!doctype html>
<html lang=en>
<title>405 Method Not Allowed</title>
<h1>Method Not Allowed</h1>
<p>The method is not allowed for the requested URL.</p>
* Connection #1 to host 127.0.0.1 left intact
```

openoms, 2023-05-07:
The issue is an updted dependency in JM: https://github.com/JoinMarket-Org/joinmarket-clientserver/issues/1483

fixed in the latest master so updating in joininbox.

rootzoll, 2023-05-08:
OK - so it makes sense to build an RC5 to be sure this is working with fatpack.

closed 2023-05-10 by rootzoll