raspiblitz/raspiblitz #3794: LNBits not running on 1.9rc4/dev after update to 0.10.5

kravens, 2023-05-05:
I've tried to update LNBits on Raspiblitz and it seems to have an issue with the poetry configuration now.
Under the admin account poetry is starting just fine, the lnbits account seemed to be missing a symbolic link to the right path.

I tried reinstalling lnbits on both the 1.9 branch and dev branch today.
Here the system logs:
```lnbits@raspberrypi:/home/admin/config.scripts $ ./bonus.lnbits.sh status
# Running: 'bonus.lnbits.sh status'
version='0.10.5'
installed=1
localIP='192.168.1.xx'
httpPort='5000'
httpsPort='5001'
httpsForced='1'
httpsSelfsigned='1'
authMethod='none'
publicIP='xx.xx.xx.xx'
LNBitsFunding='lnd'
```
systemctl status lnbits:
```
● lnbits.service - lnbits
     Loaded: loaded (/etc/systemd/system/lnbits.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Fri 2023-05-05 15:37:08 BST; 2s ago
    Process: 194448 ExecStartPre=/home/admin/config.scripts/bonus.lnbits.sh prestart (code=exited, status=0/SUCCESS)
    Process: 194470 ExecStart=/bin/sh -c cd /home/lnbits/lnbits && poetry run lnbits --port 5000 --host 0.0.0.0 (code=exited, status=127)
   Main PID: 194470 (code=exited, status=127)
        CPU: 159ms
```
sudo journalctl -u lnbits -f --since '2 hour ago'
```
May 05 15:39:12 raspberrypi systemd[1]: Started lnbits.
May 05 15:39:12 raspberrypi sh[198024]: /bin/sh: 1: poetry: not found
May 05 15:39:12 raspberrypi systemd[1]: lnbits.service: Main process exited, code=exited, status=127/n/a
May 05 15:39:12 raspberrypi systemd[1]: lnbits.service: Failed with result 'exit-code'.
```

I tried adding the link to poetry with ln, but lnbits account doesn't recognize it for some reason.
The main issue is lnbits not being properly started anymore after the update:
[image]

kravens, 2023-05-07:
https://github.com/rootzoll/raspiblitz/pull/3798#issuecomment-1537371172
Solved by installing the update under root account, instead of admin account!

closed 2023-05-07 by kravens

diskbuster, 2023-05-08:
I have the same issue. I made the update from the menu as admin.
How did you made it as root?

kravens, 2023-05-09:
Use: sudo su root 
And then run the same commands as before in the /config.scripts/ folder