raspiblitz/raspiblitz #3800: LND constantly shutting down after a few hours being active

JessieRaspi, 2023-05-06:
I am struggling with my node since a few weeks. LND is shutting down constantly every few hours. On the display is mentioned (in red):  
not started, not ready yet, no data.
Everytime when I perform a reboot of my Raspiblitz, everything works again, but only for a few hours,

I did an upgrade to 0.16.2 beta a few days ago, did a reboot, but after a while LND was shutting down again. I found something similar in this issue #3465 and turned autopilot on. However I only have one (unannounced) channel, so I don't know what that would do actually. It seemed to work first, but after a few hours it is shutting down again. 

Here is the debuglog.

https://l.termbin.com/v76a8

I also did a longer log of the LND.  See herunder.

sudo tail -n 100 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
2023-05-06 14:24:24.523 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:24:24.523 [INF] INVC: New invoice subscription client: id=191
2023-05-06 14:24:24.524 [INF] INVC: Cancelling invoice subscription for client=191
2023-05-06 14:24:28.065 [DBG] ATPL: Initialized simple graph with 9682 nodes and 55663 channels
2023-05-06 14:24:28.592 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:24:29.603 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:24:29.604 [INF] INVC: New invoice subscription client: id=192
2023-05-06 14:24:29.604 [INF] INVC: Cancelling invoice subscription for client=192
2023-05-06 14:24:34.102 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:24:34.662 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:24:34.662 [INF] INVC: New invoice subscription client: id=193
2023-05-06 14:24:34.662 [INF] INVC: Cancelling invoice subscription for client=193
2023-05-06 14:24:39.854 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:24:39.855 [INF] INVC: New invoice subscription client: id=194
2023-05-06 14:24:39.855 [INF] INVC: Cancelling invoice subscription for client=194
2023-05-06 14:24:39.975 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:24:45.059 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:24:45.060 [INF] INVC: New invoice subscription client: id=195
2023-05-06 14:24:45.113 [INF] INVC: Cancelling invoice subscription for client=195
2023-05-06 14:24:45.597 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:24:50.195 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:24:50.200 [INF] INVC: New invoice subscription client: id=196
2023-05-06 14:24:50.203 [INF] INVC: Cancelling invoice subscription for client=196
2023-05-06 14:24:51.749 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:24:55.287 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:24:55.296 [INF] INVC: New invoice subscription client: id=197
2023-05-06 14:24:55.297 [INF] INVC: Cancelling invoice subscription for client=197
2023-05-06 14:24:58.231 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:25:00.455 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:00.472 [INF] INVC: New invoice subscription client: id=198
2023-05-06 14:25:00.472 [INF] INVC: Cancelling invoice subscription for client=198
2023-05-06 14:25:05.685 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:25:05.741 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:05.748 [INF] INVC: New invoice subscription client: id=199
2023-05-06 14:25:05.748 [INF] INVC: Cancelling invoice subscription for client=199
2023-05-06 14:25:10.836 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:10.837 [INF] INVC: New invoice subscription client: id=200
2023-05-06 14:25:10.838 [INF] INVC: Cancelling invoice subscription for client=200
2023-05-06 14:25:11.305 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:25:16.041 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:16.062 [INF] INVC: New invoice subscription client: id=201
2023-05-06 14:25:16.090 [INF] INVC: Cancelling invoice subscription for client=201
2023-05-06 14:25:18.152 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:25:22.328 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:22.329 [INF] INVC: New invoice subscription client: id=202
2023-05-06 14:25:22.329 [INF] INVC: Cancelling invoice subscription for client=202
2023-05-06 14:25:24.002 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:25:27.407 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:27.408 [INF] INVC: New invoice subscription client: id=203
2023-05-06 14:25:27.408 [INF] INVC: Cancelling invoice subscription for client=203
2023-05-06 14:25:29.786 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:25:32.501 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:32.501 [INF] INVC: New invoice subscription client: id=204
2023-05-06 14:25:32.508 [INF] INVC: Cancelling invoice subscription for client=204
2023-05-06 14:25:35.479 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-06 14:25:37.614 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:37.615 [INF] INVC: New invoice subscription client: id=205
2023-05-06 14:25:37.615 [INF] INVC: Cancelling invoice subscription for client=205
2023-05-06 14:25:42.657 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:42.668 [INF] INVC: New invoice subscription client: id=206
2023-05-06 14:25:42.668 [INF] INVC: Cancelling invoice subscription for client=206
2023-05-06 14:25:47.751 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:47.752 [INF] INVC: New invoice subscription client: id=207
2023-05-06 14:25:47.752 [INF] INVC: Cancelling invoice subscription for client=207
2023-05-06 14:25:52.922 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:52.923 [INF] INVC: New invoice subscription client: id=208
2023-05-06 14:25:52.923 [INF] INVC: Cancelling invoice subscription for client=208
2023-05-06 14:25:58.013 [DBG] RPCS: [/lnrpc.Lightning/SubscribeInvoices] requested
2023-05-06 14:25:58.013 [INF] INVC: New invoice subscription client: id=209
2023-05-06 14:25:58.014 [INF] INVC: Cancelling invoice subscription for client=209
2023-05-06 14:25:58.616 [DBG] ATPL: Got scores for 5535 nodes
2023-05-06 14:25:58.686 [INF] ATPL: Attempting to execute channel attachment directives: (map[autopilot.NodeID]*autopilot.AttachmentDirective) (len=1) {
 (autopilot.NodeID) (len=33 cap=33) {
  00000000  02 ed 6d 58 db 33 69 dc  2f 27 4a a7 c8 f7 1c 64  |..mX.3i./'J....d|
  00000010  7e 9f 52 b0 1b 40 4b 55  bc 01 9a c4 f0 ba b5 f0  |~.R..@KU........|
  00000020  ed                                                |.|
 }: (*autopilot.AttachmentDirective)(0x400c0fa000)({
  NodeID: (autopilot.NodeID) (len=33 cap=33) {
   00000000  02 ed 6d 58 db 33 69 dc  2f 27 4a a7 c8 f7 1c 64  |..mX.3i./'J....d|
   00000010  7e 9f 52 b0 1b 40 4b 55  bc 01 9a c4 f0 ba b5 f0  |~.R..@KU........|
   00000020  ed                                                |.|
  },
  ChanAmt: (btcutil.Amount) 0.00052974 BTC,
  Addrs: ([]net.Addr) (len=1 cap=1) {
   (*tor.OnionAddr)(0x400dd882d0)(nuw3plwhxmfox5isrna2u5lmra6quwijto6i46e3imhyghgpny32fcid.onion:9735)
  }
 })
}

2023-05-06 14:25:58.698 [DBG] ATPL: Manager stopped autopilot agent
2023-05-06 14:25:58.708 [INF] RPCS: Stopping ChainRPC Sub-RPC Server
2023-05-06 14:25:58.715 [INF] RPCS: Stopping PeersRPC Sub-RPC Server
2023-05-06 14:25:58.721 [INF] RPCS: Stopping NeutrinoKitRPC Sub-RPC Server
2023-05-06 14:25:58.728 [INF] RPCS: Stopping WatchtowerRPC Sub-RPC Server
2023-05-06 14:25:58.731 [INF] RPCS: Stopping WatchtowerClientRPC Sub-RPC Server
2023-05-06 14:25:58.736 [INF] TORC: Stopping tor controller
2023-05-06 14:25:58.742 [DBG] TORC: removing serviceID:bf43pxxvugf5ibr2id2yjtmxe3npaby5pwbov2is6drcw32kimrqu5id from tor controller
2023-05-06 14:25:58.725 [DBG] SRVR: Connecting to 02ed6d58db3369dc2f274aa7c8f71c647e9f52b01b404b55bc019ac4f0bab5f0ed@nuw3plwhxmfox5isrna2u5lmra6quwijto6i46e3imhyghgpny32fcid.onion:9735
2023-05-06 14:25:58.946 [INF] LTND: Shutdown complete

openoms, 2023-05-06:
Can't see what causing the problem, but would keep the autopilot off (you really don't want it to open some random channels).

I can see that there is low memory (only 365 MB available). Could test: https://github.com/rootzoll/raspiblitz/pull/3780#issuecomment-1531381381
and /or deactivate Mempool to reduce the memory usage.

JessieRaspi, 2023-05-07:
I deactivated Mempool this morning. My node is running just smoothly for 12 hours now, so I guess Mempool caused the problem. (And I turned off that autopilot indeed).

You mean by low memory, the 365 MB available, the memory on the SD card, the system ram available?
On my display I now have 1415M/3844M and HDDuse 630G (73%).

I haven't test your modification yet, being a little nervous to doing something wrong and not knowing how to fix it in that case....

But thanks for the help!

JessieRaspi, 2023-06-09:
I upgraded to 1.9.0, the minimal version. LND worked well, but after a few hours I had the same problem: LND not started,  not ready yet, no data (in red on my LCD display). 
I used a brandnew official Samsung SD Pro Endurance 32 GB card and an updated Balena Etcher for the upgrade.

This morning, (hoping to get the problem solved), I updated LND to the newest version 0.16.3 .  The rebooting process lasted very long, After two hours, the display still wasn't white. However activity light was black and SSD light constantly blue, so I disconnected the power. After reconnection it immediately started to reboot and all was fine after the reboot. 

LND worked for about 8 hours, but now I have the same problem as before. 
Electrum connects to Electrs, so no problem there. 

Don't know anymore what I can do next....I don't like to reboot the Raspiblitz every few hours. Is there a command, that only restarts LND, but with the same wallet? 

Here is my debug log.
https://l.termbin.com/cd61

JessieRaspi, 2023-07-05:
just an update:
I downgraded LND about a week ago to 0.16.2 beta, after reading about memory leak problems in 0.16.3. 

Every night LND is locking. I am able to SSH in. After restarting LND and unlocking the wallet, it works again.

I have switched off most services and only have Electrs, LNBits, RTLweb and the whitepaper running.

Electrs works fine, it's only LND that has problems.

JessieRaspi, 2023-07-13:
update:
I have set the wallet to "auto unlock" and it seems, that the problem is gone. 

LND is active now for 5 days continually.

rootzoll, 2023-08-05:
OK can you test if the available v1.10rc2 is fixing this - it has a new lnd version - download here: https://github.com/raspiblitz/raspiblitz/tree/v1.10#downloading-the-software

JessieRaspi, 2023-08-05:
Hi,
OK!
Yesterday I removed the wallet and Lnd completely, so I only have a bitcoin
node running at the moment. I had the idea to install CLN instead.
But I'll try tomorrow and let you know!

JessieRaspi, 2023-08-06:
Today I have the minimum version installed and all is running again (with Electrs, LND and RTL).
Few things: 
1. I had to remove (in windows 10) the SSH key twice during the process and 
2. Electrum wouldn't connect to Electrs. The properties in the Electrum shortcut were right written, however in the config file was written: "oneserver": false. Setting that to "true" didn't help, so I did set it back to "false". Then, I blanked the line in the Electrum shortcut properties in Windows and filled the same line again. That helped!  Now, fingers crossed that LND keeps running!