raspiblitz/raspiblitz #3806: Blockchain Progress: X restarts (endless loop)

velgard1, 2023-05-07:
A while ago I realized that my node is facing an issue (status screen) and decided to reboot (the counter is going up after some seconds).
Since then I'm facing endless restart loops to no avail, what I tried so far:
- Flashing the SD card (v1.8a)
- Using another SD card (only 16GB) since the other faced errors when flashing with balenaEtcher
- Using another slot of my card reader via an SD-Card-Adapter
- Flashing with v1.8c
- Updating LND
- Updating Bitcoin Core
- Deactivating TOR

Stiil the same issue, I hope someone has an idea?
I'd also like to check if the SSD is still ok - but apart from the restarts everything seems to work fine...
To check if the blockchain etc. is still ok could also be useful since starting from scratch takes quite some time...

Thanks

velgard1, 2023-05-09:
Here the log some minutes and some loops after restart) - hope this helps to find the problem, thanks for your support!

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.8.0
chainnetwork: bitcoin / main
 20:07:06 up 11 min,  2 users,  load average: 2.31, 2.31, 1.50

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: activating (start) since Tue 2023-05-09 20:06:55 BST; 10s ago
    Process: 77299 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 77304 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
Cntrl PID: 77305 (bitcoind)
      Tasks: 13 (limit: 4531)
        CPU: 10.855s
     CGroup: /system.slice/bitcoind.service
             ├─77305 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log
             └─77306 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

May 09 20:06:55 raspberrypi systemd[1]: Starting Bitcoin daemon...

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Wed 2022-07-20 12:10:49 BST, ends at Tue 2023-05-09 20:07:06 BST. --
May 09 20:06:54 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
May 09 20:06:55 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
May 09 20:06:55 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
May 09 20:06:55 raspberrypi systemd[1]: bitcoind.service: Consumed 15.606s CPU time.
May 09 20:06:55 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 38.
May 09 20:06:55 raspberrypi systemd[1]: Stopped Bitcoin daemon.
May 09 20:06:55 raspberrypi systemd[1]: bitcoind.service: Consumed 15.606s CPU time.
May 09 20:06:55 raspberrypi systemd[1]: Starting Bitcoin daemon...

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2023-05-09T19:06:55Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2023-05-09T19:06:55Z HTTP: starting 4 worker threads
2023-05-09T19:06:55Z Using wallet directory /mnt/hdd/bitcoin
2023-05-09T19:06:55Z init message: Verifying wallet(s)…
2023-05-09T19:06:55Z Warning: Skipping -wallet path that doesn't exist. Failed to load database path '/mnt/hdd/bitcoin/wallet.dat'. Path does not exist.
2023-05-09T19:06:55Z Using /16 prefix for IP bucketing
2023-05-09T19:06:55Z init message: Loading P2P addresses…
2023-05-09T19:06:56Z Loaded 16406 addresses from peers.dat  236ms
2023-05-09T19:06:56Z init message: Loading banlist…
2023-05-09T19:06:56Z SetNetworkActive: true
2023-05-09T19:06:56Z Cache configuration:
2023-05-09T19:06:56Z * Using 2.0 MiB for block index database
2023-05-09T19:06:56Z * Using 127.8 MiB for transaction index database
2023-05-09T19:06:56Z * Using 8.0 MiB for chain state database
2023-05-09T19:06:56Z * Using 886.2 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
2023-05-09T19:06:56Z init message: Loading block index…
2023-05-09T19:06:56Z Switching active chainstate to Chainstate [ibd] @ height -1 (null)
2023-05-09T19:06:56Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2023-05-09T19:06:56Z Opened LevelDB successfully
2023-05-09T19:06:56Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000

*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
[... 254 lines cut ...]
system_ram_available_mb="2775"
ln_default_error_full=""
ln_default_running="0"
ln_default_locked="0"
ln_default_ready="0"
ln_default_error_short=""
ln_default_activated="1"
ln_default_sync_initial_done="0"
ln_default_version="0.15.4-beta"
ln_default_online="0"
btc_default_error_full=""
btc_default_activated="1"
btc_default_sync_initial_done="0"
btc_default_sync_percentage="0"
btc_default_peers="0"
btc_default_error_short=""
btc_default_ready="0"
btc_default_version="v23.0.0"
btc_default_online="0"
btc_default_port=""
btc_default_running="0"

*** LOGFILES ***
Archived and active journals take up 320.0M in the file system.
332M    /var/log

rootzoll, 2023-05-11:
I can see the bitcoind restarting, but would need a bigger chunk of the bitcoind logs to see why. Can you post the output from this command also `sudo tail -n 200 /mnt/hdd/bitcoin/debug.log`

velgard1, 2023-05-12:
Thanks, I'll need some days to check and come back then...

rootzoll, 2023-05-24:
Best to update to latest v1.9.0 raspiblitz and then use command `repair` to get access to the reindex options.

velgard1, 2023-05-25:
The update to v1.9.0 worked - I only was confused since a key has changed for the ssh login...
I tried the reindex option via the menu, but it did not work.

The "fast" version told me it's not possible due to -txindex and to I tried the first entry which gave me the following error message: Job for bitcoind.service failed because the control process exited with error code. See ... for details.

I'll have a closer look the next days, but if it's already clear what can be done without further infomation an answer would be appreciated...

velgard1, 2023-05-31:
With the two mentioned commands I get those logs:

systemctl status bitcoind.service
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: activating (start) since Wed 2023-05-31 12:47:15 BST; 7s ago
    Process: 50943 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status>
    Process: 50956 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
Cntrl PID: 50964 (bitcoind)
      Tasks: 13 (limit: 4531)
        CPU: 7.874s
     CGroup: /system.slice/bitcoind.service
             ├─50964 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin >
             └─50983 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin >

journalctl -xe
Hint: You are currently not seeing messages from other users and the system.
      Users in groups 'adm', 'systemd-journal' can see all messages.
      Pass -q to turn off this notice.
May 31 12:46:34 raspberrypi systemd[46016]: Started Sound Service.
░░ Subject: A start job for unit UNIT has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit UNIT has finished successfully.
░░
░░ The job identifier is 19.
May 31 12:46:34 raspberrypi systemd[46016]: Reached target Main User Target.
░░ Subject: A start job for unit UNIT has finished successfully
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit UNIT has finished successfully.
░░
░░ The job identifier is 1.
May 31 12:46:34 raspberrypi systemd[46016]: Startup finished in 681ms.
░░ Subject: User manager start-up is now complete
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The user manager instance for user 1001 has been started. All services queued
░░ for starting have been started. Note that other services might still be starting
░░ up or be started at any later time.
░░
░░ Startup of the manager took 681615 microseconds.
May 31 12:46:34 raspberrypi sudo[46162]:    admin : TTY=pts/0 ; PWD=/home/admin ; USER=root ; COMMAND=/usr/bin/df
May 31 12:46:34 raspberrypi sudo[46162]: pam_unix(sudo:session): session opened for user root(uid=0) by admin(uid=1001)
May 31 12:46:34 raspberrypi sudo[46162]: pam_unix(sudo:session): session closed for user root
lines 1091-1119/1119 (END)

Hope that helps, thanks!

velgard1, 2023-06-21:
Sadly it does not work - but it's likely another issue now.
I'll close the topic and open another now.

closed 2023-06-21 by velgard1