raspiblitz/raspiblitz #3809: baseimage not supported (debian)

Minituff, 2023-05-08:
Hello, 

I am trying to install raspiblitz on my debian proxmox machine and I keep getting the following error:

```bash
> mempool-backend@2.4.0 tsc
> ./node_modules/typescript/bin/tsc

# blitz.display.sh set-display lcd
baseimage(debian)
# old(headless)
# new(lcd)
err='baseimage not supported'
```

I have tried following both of these guides:
* https://github.com/rootzoll/raspiblitz/tree/v1.9/alternative.platforms#virtual-machine
* https://github.com/rootzoll/raspiblitz/tree/dev/alternative.platforms/Proxmox#readme

Minituff, 2023-05-09:
Update, I have gotten raspiblitz to install with this command

`sudo bash build_sdcard.sh -f true -b dev -d headless -t false -w off`

The new issue is whenever I attach a drive in proxmox ([variant 1](https://github.com/rootzoll/raspiblitz/tree/dev/alternative.platforms/Proxmox#variant-1-external-hard-disk)) then the system just endlessly reboots.

rootzoll, 2024-09-12:
The problem was that on fatpack the LCD was always installed even when signaled `-d headless`. The lastest build script in `dev` branch now respects the `-d headless` on fatpack. Closing the issue.

closed 2024-09-12 by rootzoll