raspiblitz/raspiblitz #3821: raspiblitz-1.8.0c lnbits gui publicly accessible

jalbstmeijer, 2023-05-11:
Hi,

Is it correct, that both the local (port 5001) as the tor lnbits url is accessible without any authentication?, 

Gr, J

rootzoll, 2023-05-11:
Thats basically correct. lnbits creates new subaccounts that are reachable thru a unique URL - whoever has that URL can control that subaccount. Thats makes it easy to use on the one hand, but not very secure on the other .. so should not be used to manage big amounts of sats.

jalbstmeijer, 2023-05-11:
Yes, not only control an existing sub account.. but also create new ones.
Did not know that an onion address was seen as some kind of security by obscurity.

So nothing that can be added to increase security?

rootzoll, 2023-05-15:
> Did not know that an onion address was seen as some kind of security by obscurity.

The onion address just lets people call the lnbits main page - to control an existing sub account the sub account as part of the url is needed.

As much as I like to advice in best practices on lnbits ... its just an app running on RaspiBlitz. Best would be to to connect to the lnbits project/community directly to get your questions answered. They have a Telegram group: https://t.me/lnbits

jalbstmeijer, 2023-05-16:
@rootzoll 
I seem not to have explained my self very well.

Raspblitz exposes lnbits on port 5001.
It also exposes this port through Tor on a separate onion address than the main one. (need to go in to the cli menu to get this onion address)

The lnbits web gui does not seem to offer any authentication for the gui.
So anyone aware of the onion address can make new sub accounts/wallets.

(Am I the only one that sometimes inputs a onion address to a non Tor browser, by mistake resulting in posting it to google as search string?)

As it feels a bit "open" and unnecessary... so I was thinking... when my lnbits sub account users  use a vpn connection to port 5001. Could I just stop exposing lnbits through Tor?

Does that make sense? maybe even an config/setting option to consider for Raspblitz?

rootzoll, 2023-05-16:
Yes your right the onion address gives unprotected access to the LNbits UI. I can see this feeling a bit "unprotected" but can you again make clear what the attack scenario you fear for? There is no access to you funds - worst thing that could happen that somebody creates a sub account and sends you money .. use you as a custodian.

If you abolutly dont want that you can manually deactivate the Tor hidden service for lndbits.

rootzoll, 2023-05-16:
> Could I just stop exposing lnbits through Tor?

run from the command line: `/home/admin/config.scripts/tor.onion-service.sh off lnbits` you need to do this manually after every sd card update or recovery

jalbstmeijer, 2023-05-16:
@rootzoll 

> Yes your right the onion address gives unprotected access to the LNbits UI. I can see this feeling a bit "unprotected" but can you again make clear what the attack scenario you fear for? There is no access to you funds (thru the UI) - worst thing that could happen that somebody creates a sub account and sends you money .. use you as a custodian.

I think I agree funds are probably not affected directly in the scenario.

But from a security point of view, any interface you expose could potentially be an entry point to the Raspiblitz node when this code can be exploited. (This probably also applies to parts of the general Raspiblitz web gui)

And if not exploitable currently, it might at least offer a way to trigger a DDOS.

- Can it withstand many calls?
- What would happen if I would create a script to keep on creating wallets and or invoices in a loop?

So in general I tend not to expose anything if there is no absolute need for it.

> run from the command line: /home/admin/config.scripts/tor.onion-service.sh off lnbits you need to do this manually after every sd card update or recovery

Thank you, that looks better than editing the Tor config.

Gr, Justin

cstenglein, 2023-06-07:
@jalbstmeijer Did rootzoll answer your questions?

Closing this for now, please re-open if you got more questions.

closed 2023-06-07 by cstenglein