raspiblitz/raspiblitz #3831: Lightning Terminal error when trying to create custodial account on version v1.9.0rc5

FlorianHubl, 2023-05-14:
When I try to create a custodial account on the Lightning Terminal I get a error: "Unable to register custodial account, proxy error with context lit: invalid macaroon: permission denied" on the Raspiblitz version v1.9.0rc5.
[image]

rootzoll, 2023-05-16:
Steps to reproduce:

https://docs.lightning.engineering/lightning-network-tools/lightning-terminal/accounts#docs-internal-guid-587703e5-7fff-5b2b-8389-3bd9e7bf6d0b

rootzoll, 2023-05-16:
@guggero I created fresh macaroons but still getting this error - can you give an hint what else to try to make this this work?

rootzoll, 2023-05-16:
@guggero I did a complete refresh of all lnd macaroons - especially the admin.macaroon that is named in `litd.conf` with `remote.lnd.macaroonpath=/home/lit/.lnd/data/chain/bitcoin/mainnet/admin.macaroon`, still the same error - where are the litd macaroons are located

guggero, 2023-05-16:
Not sure where they are on RaspiBlitz, the default location is `~/.lit/<network>/lit.macaroon`.

guggero, 2023-05-16:
Yes, you can bake a new one (see `lncli listpermissions` and `lncli bakemacaroon`) and then delete the old one.

rootzoll, 2023-05-16:
@guggero Thx. 

TODO for v1.10.0: rewrite `lnd.credentials.sh` to rebacke/sync macaroons without restarting lnd

rootzoll, 2023-05-26:
Solution so far: On RaspiBlitz SSH menu under "CONNECT > RESET" which then will also reset the litd macaroon now.

closed 2024-04-23 by rootzoll