raspiblitz/raspiblitz #3837: CLN v23.05 needs clrest update and status screen fix

openoms, 2023-05-15:
it is due to chnages how the msat amounts are returned
Related: #3835

surfac3, 2023-05-15:
[image]

When CLN is done finished scanning there is a balance inside channels and onchain - `0` 

feereport is still showing correct amount and number of channels also is show correctly

openoms, 2023-05-17:
Example output on v23.02:
```
₿ scl listfunds
{
   "outputs": [
      {
         "txid": "fcc72185ce859a4da427e85f36562fbb892332f0d8a711c6c0051bd3422e78eb",
         "output": 1,
         "value": 4918,
         "amount_msat": "4918000msat",
         "scriptpubkey": "00145592458b5af67897bc7248b7ef41ef15af2a5218",
         "address": "tb1q2kfytz667euf00rjfzm77s00zkhj55sc8xzrwt",
         "status": "confirmed",
         "blockheight": 109683,
         "reserved": false
      },
      {
         "txid": "fd4fdac26ed079182eef4290154f96363809c7127d74c7b39c262a6b2ff10fd1",
         "output": 1,
         "value": 84950,
         "amount_msat": "84950000msat",
         "scriptpubkey": "0014b749274a26602e23847f1c139cc89f0e6ea98fde",
         "address": "tb1qkayjwj3xvqhz8prlrsfeejylpeh2nr77llcddx",
         "status": "confirmed",
         "blockheight": 130651,
         "reserved": false
      }
   ],
   "channels": []
}
```
In the `cl.monitor.sh`:
getting the value with:
```
cl listfunds | jq .outputs[] | jq 'select(.status=="confirmed")' | grep value | awk '{print $2}' | cut -d, -f1
```

Will simplify this and fix for the new format once I get a balance on an updated node or someone could post the sample output of listfunds on v23.05

surfac3, 2023-05-23:
when running `cl listfunds`

```
{
   "outputs": [
      {
         "txid": "d0fda55b92d6077cf74f6f58a32dab8ab3c577705b20dcaa00d4bf6c7b2fcfc0",
         "output": 0,
         "amount_msat": 658000,
         "scriptpubkey": "0014488f94ce2361000e2b9ae9699965558b9f238ffc",
         "address": "bc1qfz8efn3rvyqqu2u6a95eje243w0j8rlurlzvqv",
         "status": "unconfirmed",
         "reserved": false
      },
      {
         "txid": "f7b2cfa2c06a07829861a8f4547c1182902ec4d2d0043c485f91e95542c08e0b",
         "output": 1,
         "amount_msat": 47304000,
         "scriptpubkey": "0014eac76a7c9c9867b47de54bbd61d3f81169280d56",
         "address": "bc1qatrk5lyunpnmgl09fw7kr5lcz95jsr2kljm3rx",
         "status": "confirmed",
         "blockheight": 782468,
         "reserved": false
      },
      {
         "txid": "3cee0466727e8a486a372bc937599160df0e3fa798ec926a98c64c315c11a2dd",
         "output": 3,
         "amount_msat": 3880851000,
         "scriptpubkey": "00207e13dd1db35b00cf853039d9cda1bbdf2faa1fa63e77b37c25f642a5af1cc721",
         "address": "bc1q0cfa68dntvqvlpfs88vumgdmmuh658ax8emmxlp97ep2ttcucusslfjlfn",
         "status": "confirmed",
         "blockheight": 789705,
         "reserved": false
      },
      {
         "txid": "b3deb0ecd7dab559f8293dc306d0fe32baf965afb863da4c886c7dac31506637",
         "output": 1,
         "amount_msat": 596267000,
         "scriptpubkey": "0014b248ee00ed64d0a61c92a8e95db32ae62be4fee3",
         "address": "bc1qkfywuq8dvng2v8yj4r54mve2uc47flhrgju60g",
         "status": "confirmed",
         "blockheight": 789757,
         "reserved": false
      },
      {
         "txid": "18bdc828d90fd2f34cf2c6af2822b8e99c0828a06b439d0f5a724709bdc4893f",
         "output": 0,
         "amount_msat": 587958000,
         "scriptpubkey": "0014a4addd4f24b4e7b042fdce0ae3e40c0638599783",
         "address": "bc1q5jka6neyknnmqshaec9w8eqvqcu9n9urpjsxuf",
         "status": "confirmed",
         "blockheight": 789777,
         "reserved": false
      },
      {
         "txid": "843ae16acec78d3cbc9c3084c5f2dbcf0bdd8bb35d4f831d766067ebad96c458",
         "output": 0,
         "amount_msat": 521873000,
         "scriptpubkey": "0014beb52bd81e7da55fb3f2eae2b9c8bf2a71f7f564",
         "address": "bc1qh66jhkq70kj4lvljat3tnj9l9fcl0atyxvcc2u",
         "status": "confirmed",
         "blockheight": 790032,
         "reserved": false
      },
      {
         "txid": "0134ad0543bef80b8e9a87ffc052e656bbdde6cbb40f34b0b2d33991177b9a03",
         "output": 1,
         "amount_msat": 768000,
         "scriptpubkey": "0014d3397ad77db7ff68b6aaf71601a51f83296b2056",
         "address": "bc1q6vuh44makllk3d427utqrfglsv5kkgzky66dr0",
[... 132 lines cut ...]
         "funding_output": 0
      },
      {
         "peer_id": "026165850492521f4ac8abd9bd8088123446d126f648ca35e60f88177dc149ceb2",
         "connected": true,
         "state": "CHANNELD_NORMAL",
         "channel_id": "86df42ef99361af704cdf67fd1142b42bcd05f5ef1e5ec45c8879acfd10340bf",
         "short_channel_id": "721071x1583x0",
         "our_amount_msat": 1259381435,
         "amount_msat": 1295816000,
         "funding_txid": "bf4003d1cf9a87c845ece5f15e5fd0bc422b14d17ff6cd04f71a3699ef42df86",
         "funding_output": 0
      }
   ]
}
```
when running `cl listfunds | jq .outputs[] | jq 'select(.status=="confirmed")' | grep value | awk '{print $2}' | cut -d, -f1`

[image]

and when running `cl listfunds | jq '.outputs[] | select(.status == "confirmed").value'`

[image]

How did i doxxed my channels now .. oh boy.

openoms, 2023-07-09:
@surfac3 to test the balance fixes:
patch to:
REPO: `raspiblitz`
BRANCH: `clrest-patches` (or dev once #4005 is merged)

reinstalling cl-rest and RTL should fix the issues with those as well.
If not updated yet uninstall RTL and c-lightning-REST with:
```
config.scripts/bonus.rtl.sh off purge
config.scripts/cl.rest.sh off purge
```
for a clean install

rootzoll, 2023-07-29:
PR #4005 merged. Closing Issue. Please reopen is problem still exists in next RC release.

closed 2023-07-29 by rootzoll