raspiblitz/raspiblitz #3848: Bitcoin Core not starting after reboot (Fresh install)

judesen, 2023-05-17:
I had an issue when booting the RaspiBlitz. It kept stating "Blockchain Progress: 5 restart", so I decided to do a fresh install and resync everything. After the sync was completed, everything was working great again.

Then I did the following steps, which has now caused the exact same issue.
1. When into the bitcoin.conf and added an additional ip address to rpcallowip, so it now stated "rpcallowip=127.0.0.1 xxx.xxx.xxx.xxx". Notice that I used a space.
2. Then I rebooted using "menu" and navigated to reboot.
3. When it was up and running again it gave me this issue.
4. I tried changing back the bitcoin.conf so it again said "rpcallowip=127.0.0.1".
5. I rebooted and I was hit by the same issue.

The log just seems to state that Bitcoin Core experienced an issue and therefore exitted.
You can find the log here: https://l.termbin.com/5opj

judesen, 2023-05-17:
From what I understand the "wallet.dat" file has been corrupted. How would this be corrupted based on the above steps?
Still trying to figure out a way to repair the file.

airs0urce, 2023-05-31:
After upgrade to 1.9.0 my wallet.dat got corrupted too:

In /mnt/hdd/bitcoin/debug.log I see
```
2023-05-31T07:47:25Z Error: Failed to load database path '/mnt/hdd/bitcoin/wallet.dat'. Data is not in recognized format.
2023-05-31T07:47:25Z Shutdown: In progress...
2023-05-31T07:47:25Z scheduler thread exit
2023-05-31T07:47:25Z Shutdown: done```

openoms, 2023-05-31:
Not sure what causing this but is likely a configuration issue.

If you get this error can simply move the wallet.dat file to a backup (it is not holding funds):
`sudo mv /mnt/hdd/bitcoin/wallet.dat /mnt/hdd/bitcoin/wallet.dat.backup`

then restart bitcoind:
`sudo systemctl restart bitcoind `

the new wallets, used by joinmarket for example are created as a directory.

Please post what your `bitcoin.conf` contains (except the rpcpassword) to try spot the problem.

airs0urce, 2023-05-31:
@openoms 
I just renamed the wallet.dat file. 

Now I get this error message:
> 2023-05-31T14:57:35Z Error: Failed to load database path '/mnt/hdd/bitcoin'. Data is not in recognized format.

This is content of the bitcoin folder:
```
admin@192.168.88.253:~ ₿ ls -lah /mnt/hdd/bitcoin
total 125M
drwxr-xr-x 8 bitcoin bitcoin 4.0K May 31 15:58 .
drwxr-xr-x 9 root    root    4.0K May 31 15:57 ..
-rw------- 1 bitcoin bitcoin    0 May 31 15:57 .lock
-rw------- 1 bitcoin bitcoin    0 May 30 16:14 .walletlock
-rw------- 1 bitcoin bitcoin   31 Nov  9  2022 banlist.json
-rw-r--r-- 1 bitcoin bitcoin  885 May 30 16:14 bitcoin.conf
drwxr-xr-x 3 bitcoin bitcoin 224K May 30 17:59 blocks
drwxr-xr-x 2 bitcoin bitcoin 120K May 30 14:22 chainstate
drwx------ 2 bitcoin bitcoin 4.0K May 30 16:14 database
-rw------- 1 bitcoin bitcoin 102K May 31 07:06 db.log
-rw------- 1 bitcoin bitcoin 106K May 31 15:58 debug.log
-rw------- 1 bitcoin bitcoin 243K May 30 14:13 fee_estimates.dat
drwx------ 4 bitcoin bitcoin 4.0K May  8 11:01 indexes
-rw------- 1 bitcoin bitcoin 123M May 30 14:13 mempool.dat
-rw------- 1 bitcoin bitcoin   99 May 30 14:16 onion_v3_private_key
-rw------- 1 bitcoin bitcoin 1.5M May 23 19:14 peers.070a
drwx------ 2 bitcoin bitcoin 4.0K May 30 14:16 satstack
-rw-r--r-- 1 bitcoin bitcoin   57 May 31 15:58 settings.json
drwx------ 3 bitcoin bitcoin 4.0K May  8 12:51 specter
-rw------- 1 bitcoin bitcoin 8.6K May 30 14:17 wallet.dat-journal
-rw------- 1 bitcoin bitcoin  12K May 30 14:17 wallet.dat.backup
```

Here is full bitcoind startup log:
```
2023-05-31T14:57:35Z Bitcoin Core version v24.0.1 (release build)
2023-05-31T14:57:35Z InitParameterInteraction: parameter interaction: -bind set -> setting -listen=1
2023-05-31T14:57:35Z InitParameterInteraction: parameter interaction: -proxy set -> setting -upnp=0
2023-05-31T14:57:35Z InitParameterInteraction: parameter interaction: -proxy set -> setting -natpmp=0
2023-05-31T14:57:35Z InitParameterInteraction: parameter interaction: -proxy set -> setting -discover=0
2023-05-31T14:57:35Z Using the 'standard' SHA256 implementation
2023-05-31T14:57:35Z Default data directory /home/bitcoin/.bitcoin
2023-05-31T14:57:35Z Using data directory /mnt/hdd/bitcoin
2023-05-31T14:57:35Z Config file: /mnt/hdd/bitcoin/bitcoin.conf
2023-05-31T14:57:35Z Config file arg: blockfilterindex="1"
2023-05-31T14:57:35Z Config file arg: daemon="1"
2023-05-31T14:57:35Z Config file arg: datadir="/mnt/hdd/bitcoin"
2023-05-31T14:57:35Z Config file arg: dbcache="512"
2023-05-31T14:57:35Z Config file arg: deprecatedrpc="addresses"
2023-05-31T14:57:35Z Config file arg: disablewallet="0"
2023-05-31T14:57:35Z Config file arg: dns="0"
2023-05-31T14:57:35Z Config file arg: dnsseed="0"
2023-05-31T14:57:35Z Config file arg: maxconnections="40"
2023-05-31T14:57:35Z Config file arg: maxmempool="300"
2023-05-31T14:57:35Z Config file arg: maxorphantx="10"
2023-05-31T14:57:35Z Config file arg: maxuploadtarget="5000"
2023-05-31T14:57:35Z Config file arg: onlynet="onion"
2023-05-31T14:57:35Z Config file arg: peerbloomfilters="1"
2023-05-31T14:57:35Z Config file arg: proxy="127.0.0.1:9050"
2023-05-31T14:57:35Z Config file arg: rpcallowip="127.0.0.1"
2023-05-31T14:57:35Z Config file arg: rpcallowip="192.168.88.0/24"
2023-05-31T14:57:35Z Config file arg: rpcpassword=****
2023-05-31T14:57:35Z Config file arg: rpcuser=****
2023-05-31T14:57:35Z Config file arg: server="1"
2023-05-31T14:57:35Z Config file arg: testnet="0"
2023-05-31T14:57:35Z Config file arg: txindex="1"
2023-05-31T14:57:35Z Config file arg: whitelist="download@127.0.0.1"
2023-05-31T14:57:35Z Config file arg: whitelist="127.0.0.1"
2023-05-31T14:57:35Z Config file arg: zmqpubrawblock="tcp://127.0.0.1:28332"
2023-05-31T14:57:35Z Config file arg: zmqpubrawtx="tcp://127.0.0.1:28333"
[... 59 lines cut ...]
maxconnections=40
maxuploadtarget=5000
# tor by default
whitelist=download@127.0.0.1
main.wallet=wallet.dat
rpcallowip=192.168.88.0/24
main.rpcbind=192.168.88.253
whitelist=127.0.0.1
blockfilterindex=1
deprecatedrpc=addresses
onlynet=onion
proxy=127.0.0.1:9050
main.bind=127.0.0.1
test.bind=127.0.0.1
dnsseed=0
dns=0
```

Also I remember that right after update everything was working. 
Then I requested install of Join Market using WebUI. 
When I tried to open JoinMarket web interface it was showing me error like server doesn't respond on port #N. 
I saw that some processing goes there because CPU usage was near 100% (saw in WebUI) and I left it alone.

Next time I connected it this is what I saw:  
[image]

openoms, 2023-05-31:
@airs0urce is seems there is something wrong with the satstack wallet file also. Try you rename or move that too and restart bitcoind.

airs0urce, 2023-05-31:
@openoms thank you!

cstenglein, 2023-06-07:
Looks like it's fixed @airs0urce ?

=> closed, if it still persists, please re-open

closed 2023-06-07 by cstenglein