raspiblitz/raspiblitz #3856: CL: Could not find backup.lock

rootzoll, 2023-05-21:
```
*** LAST CORE LIGHTNING (MAINNET) INFO LOGS ***
For details also use command --> cllog
sudo tail -n 50 /home/bitcoin/.lightning/bitcoin/cl.log

2023-05-21T07:58:10.449Z INFO    plugin-backup.py: Could not find backup.lock in the lightning-dir
2023-05-21T07:58:11.452Z INFO    plugin-backup.py: Killing process lightningd (3346584)
```

openoms, 2023-05-21:
Looks like the backup plugin initialization failed for some reason.
Does this happen every time?
Is it solved by running the init again:
```
config.scripts/cl-plugin.backup.sh on
```
?

rootzoll, 2023-05-21:
@openoms When running the command `config.scripts/cl-plugin.backup.sh on` is the following correct?

```
Processing triggers for man-db (2.9.4-2) ...
# Stop the lightningd.service
# Initialize the backup plugin
Command not found: ./backup-cli
# The crontab for admin now is:
@weekly /home/admin/config.scripts/cl-plugin.backup.sh backup-compact mainn
```

The script ran thru. I restarted and still get the CL error:

```
2023-05-21T18:09:52.969Z INFO    plugin-backup.py: Could not find backup.lock in the lightning-dir
2023-05-21T18:09:53.974Z INFO    plugin-backup.py: Killing process lightningd (37607)
```

rootzoll, 2023-05-21:
@openoms basically `/home/admin/config.scripts/cl-plugin.backup.sh on` is called as part of the `cl.install.sh` script:

https://github.com/rootzoll/raspiblitz/blob/da66fb9724ed7d894ac79eb4499fe0f3261e798c/home.admin/config.scripts/cl.install.sh#L307

and also as part of `_provision.setup.sh` its calling `/home/admin/config.scripts/cl-plugin.backup.sh on mainnet`
https://github.com/rootzoll/raspiblitz/blob/da66fb9724ed7d894ac79eb4499fe0f3261e798c/home.admin/_provision.setup.sh#L381

so i think its getting run, but something is missing.

openoms, 2023-05-22:
Seems 
```
Command not found: ./backup-cli
```
is the issue.
When I am running on my testnode with the previous rc it works ok:
```
₿ config.scripts/cl-plugin.backup.sh on
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
Updating 3000327..eee5e90
Fast-forward
 .github/workflows/main.yml      |   8 +++---
 .mergify.yml                    |  12 ++++----
 README.md                       |   2 ++
 backup/poetry.lock              |  14 ++++++----
 clearnet/README.md              |  10 +++++++
 clearnet/clearnet.py            | 114 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 clearnet/requirements.txt       |   1 +
 clearnet/test_clearnet.py       |  22 +++++++++++++++
 currencyrate/README.md          |   4 +--
 currencyrate/currencyrate.py    |   6 ----
 donations/poetry.lock           |  12 ++++----
 feeadjuster/feeadjuster.py      | 100 ++++++++++++++++++++++++++++++++++++++----------------------------
 feeadjuster/test_feeadjuster.py |  69 ++++++++++++++++++++++++++++++++++++----------
 historian/poetry.lock           |  12 ++++----
 noise/README.org                |  14 +++++++---
 noise/noise.py                  |  13 +++++++++
 noise/requirements-dev.txt      |   2 ++
 noise/requirements.txt          |   1 +
 noise/test_chat.py              |   4 +--
 nostrify                        |   2 +-
 summary/summary.py              |  39 ++++++++++++++++----------
 summary/test_summary.py         |  22 +++++++++++++++
 22 files changed, 371 insertions(+), 112 deletions(-)
 create mode 100644 clearnet/README.md
 create mode 100755 clearnet/clearnet.py
 create mode 100644 clearnet/requirements.txt
 create mode 100644 clearnet/test_clearnet.py
 create mode 100644 noise/requirements-dev.txt
 create mode 100644 noise/requirements.txt
HEAD is now at 3000327 summary: check and dismiss old datastores
# The backup plugin is already loaded
# Stop the lightningd.service
# Backup the existing old backup on the SDcard
/home/bitcoin/.lightning/bitcoin/backup.lock
# Initialize the backup plugin
Found an existing database at /home/bitcoin/.lightning/bitcoin/lightningd.sqlite3, initializing the backup with a snapshot
Successfully written initial snapshot to file:///home/bitcoin/lightningd.sqlite3.backup
Initialized backup backend file:///home/bitcoin/lightningd.sqlite3.backup, you can now start Core-Lightning
# The crontab for admin now is:
@weekly /home/admin/config.scripts/cl-plugin.backup.sh backup-compact mainnet
@weekly /home/admin/config.scripts/cl-plugin.backup.sh backup-compact testnet
@weekly /home/admin/config.scripts/cl-plugin.backup.sh backup-compact signet

# Started the lightningd.service
```

Checking if the file is there sn poetry is available for the bitcoin user:
```
sudo -u bitcoin poetry run /home/bitcoin/cl-plugins-available/plugins/backup/backup-cli
Usage: backup-cli [OPTIONS] COMMAND [ARGS]...

Options:
  --help  Show this message and exit.

Commands:
  init
  restore
  server
```
Will need to do a clean install to test.

rootzoll, 2023-05-22:
@openoms This is what it looks like on a clean cln install an my test machine:

[image]

Because its a poetry command - can it have anything todo with me changing the virtialenv after poetry installs? see changes I did here: https://github.com/rootzoll/raspiblitz/pull/3850/files

openoms, 2023-05-22:
@rootzoll thsi pyproject.toml error is not the problem . it can be solved if running the comanfd in the right directory (like in the script).

So the correct test to reproduce the issue is:
```
cd /home/bitcoin/cl-plugins-available/plugins/backup/
sudo -u bitcoin poetry run ./backup-cli
```

just running a new install to reproduce.

openoms, 2023-05-22:
Good news!
using the full path seems to work:
```
cd /home/bitcoin/cl-plugins-available/plugins/backup/
sudo -u bitcoin poetry run  /home/bitcoin/cl-plugins-available/plugins/backup//backup-cli
```

rootzoll, 2023-05-22:
@openoms you have a PR incomming to test?

PR #3861 (2023-05-22): fix: use absolute path for cln backup-cli
fix: #3856 
running:
```
./config.scripts/cl-plugin.backup.sh on
```
works now and fixes CLN.
Will run a test with a new install.
closed 2023-05-23 by rootzoll