raspiblitz/raspiblitz #3857: v1.9 datadrive.sh script won't find the largest partition for x86 devices

BTCNautilus, 2023-05-21:
I am building raspiblitz on a NUC, with a 2TB nvme drive as the data drive. I kept getting an error going through the process, and noticed that the datadrive.sh status was seemingly finding the wrong partition for my data drive.

sda
├─sda1      vfat
├─sda2      ext4
└─sda3      swap
nvme0n1
├─nvme0n1p1 btrfs  BLITZDATA
├─nvme0n1p2 btrfs  BLITZSTORAGE
└─nvme0n1p3 vfat   BLITZTEMP
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=0
isBTRFS=1

# SETUP INFO
isSMART=0
isSSD=1
hddCandidate='nvme0n1'
hddBytes=2000398934016
hddGigaBytes=1863
hddPartitionCandidate='nvme0n1p3'
hddDataPartitionBytes=36507205120
hddPartitionGigaBytes=33
hddFormat='vfat'
hddRaspiData=0
hddBlocksBitcoin=0
hddGotBlockchain=0
hddAdapterUSB=''
hddAdapterUSAP=0

I looked over the datadrive.sh script, and found that the section that checks whether the current line in lsblk is largest is missing the size comparison for x86 devices. 

 if [ "$(uname -m)" = "x86_64" ]; then
	if [ $(echo "$testpartition" | grep -c "nvme")  = 0 ]; then
          testParentDisk=$(echo "$testpartition" | sed 's/[^a-z]*//g')
	else
          testParentDisk=$(echo "$testpartition" | sed 's/\([^p]*\).*/\1/')
   	fi
	if [ $(echo "$OSPartition" | grep -c "nvme")  = 0 ]; then
          OSParentDisk=$(echo "$OSPartition" | sed 's/[^a-z]*//g')
	else
          OSParentDisk=$(echo "$OSPartition" | sed 's/\([^p]*\).*/\1/')
        fi
        if [ $(echo "$bootPartition" | grep -c "nvme")  = 0 ]; then	
          bootParentDisk=$(echo "$bootPartition" | sed 's/[^a-z]*//g')
	else
	  bootParentDisk=$(echo "$bootPartition" | sed 's/\([^p]*\).*/\1/')
	fi
		  
        if [ "$testdevice" != "$OSParentDisk" ] && [ "$testdevice" != "$bootParentDisk" ];then
          sizeDataPartition=${testsize}
          hddDataPartition="${testpartition}"
          hdd="${testdevice}"
        fi

It seems it is simply iterating and just adding the latest partition as the candidate. I've added the following to it (same as the non x86 version), and got the right status from the script:

if [ "$testdevice" != "$OSParentDisk" ] && [ "$testdevice" != "$bootParentDisk" ];then
           # make sure to use the biggest
          if [ ${testsize} -gt ${sizeDataPartition} ]; then
            sizeDataPartition=${testsize}
            hddDataPartition="${testpartition}"
            hdd="${testdevice}"
          fi
        fi
        
Not sure if it actually fixes my issue, as my install is now busted and I'll have to re-image it. Will update if it does (although I think that's somewhat irrelevant).

BTCNautilus, 2023-05-22:
The script doesn't like blunt force simple fix, and is causing a reboot during the formatting step, I'll try harder.

rootzoll, 2023-08-05:
@openoms I have no x86 platform here to test. Is this something to still fix for v1.10.0?

rootzoll, 2024-10-01:
Supporting multiple drive/partition layouts will be addressed in upcomming v1.12.0 release - after v1.11.3

rootzoll, 2024-11-14:
Closing this issue because we consider to remove BTRFS from RaspiBlitz - see: https://github.com/raspiblitz/raspiblitz/issues/2045#issuecomment-2471845122

closed 2025-05-15 by rootzoll