raspiblitz/raspiblitz #3873: Suez wont launch on v1.9 after update

surfac3, 2023-05-24:
When attempting to launch suez after update the tool is reinstalled 

Once installation is done following is displayed in terminal

[image]

Worth to add i am running reckless 23.05 CLN version.

rootzoll, 2023-05-24:
CLM 23.05 could be another problem, they have some breaking changes. 

For the suez on v1.9.0 please run once: `patch code` on command line (or SSHMENU > UPDATE > PATCH > PATCHMENU > PATCH) & reboot.

surfac3, 2023-05-24:
Now i am getting different errors 

I fear its connected with the CLN23.05

[image]