raspiblitz/raspiblitz #3876: Blockchain Sync Stuck at Same Block Height for Days now

uncleJim21, 2023-05-27:
Not sure what happened but I suspect there's something wrong with either read/write or corruption of the magic number. Here are the logs I have:
https://l.termbin.com/1tbcg

openoms, 2023-05-29:
keep monitoring the bitcoind logs to see what's going wrong - find it in menu - SYSTEM

uncleJim21, 2023-05-29:
Bitcoind keeps running on a loop saying that leveldb's magic number is corrupted

openoms, 2023-05-29:
Will need to resync to solve that. Can use the REPAIR menu to reset the chain.

cstenglein, 2023-06-09:
@uncleJim21 is that still an issue?

Closing for now, if it still is, please re-open.

closed 2023-06-09 by cstenglein

uncleJim21, 2023-06-09:
@cstenglein I mean it's still an issue from the standpoint that the DB corrupts and has no recourse without hours of effort. I am just pulling the plug on Raspibitz it's not worth the effort for me. Thanks.