raspiblitz/raspiblitz #3878: Update to ver 1.9 my LND is not working?

Omicron59, 2023-05-27:
Hello after an update to ver 1.9 my LND no longer launches, I feel like there is an error in my lnd.conf

https://l.termbin.com/l8bj

after running this command i get an error

sudo journalctl  -n100 -fu lnd

May 27 20:40:47 raspberrypi systemd[1]: lnd.service: Consumed 1.034s CPU time.
May 27 20:41:47 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 22.
May 27 20:41:47 raspberrypi systemd[1]: Stopped Lightning Network Daemon.
May 27 20:41:47 raspberrypi systemd[1]: lnd.service: Consumed 1.034s CPU time.
May 27 20:41:48 raspberrypi lnd[158375]: failed to load config: /home/bitcoin/.lnd/lnd.conf:70: unknown option: routerrpc.apriorihopprob
May 27 20:41:47 raspberrypi systemd[1]: Starting Lightning Network Daemon...
May 27 20:41:48 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
May 27 20:41:48 raspberrypi systemd[1]: lnd.service: Failed with result 'exit-code'.
May 27 20:41:48 raspberrypi systemd[1]: Failed to start Lightning Network Daemon.
May 27 20:41:48 raspberrypi systemd[1]: lnd.service: Consumed 1.074s CPU time.

I can't find what's wrong
if you have any solution

 thank you

openoms, 2023-05-29:
Remove the deprecated lines form your lnd.conf:
```
failed to load config: /home/bitcoin/.lnd/lnd.conf:70: unknown option: routerrpc.apriorihopprob
```

Omicron59, 2023-05-29:
hello openoms thank you for your answer I had to deactivate line by line so that LND restarts without error

##############################
# Optimise for cheap routes #
##############################
[routerrpc]
# Make sure that LND is the binary release or built with the routerrpc tag
# Set default chance of a hop success
#routerrpc.apriorihopprob=0.5
# Start to ignore nodes if they return many failures (set to 1 to turn off)
#routerrpc.aprioriweight=0.75
# Set minimum desired savings of trying a cheaper path
# The (virtual) fixed cost in sats of a failed payment attempt (default: 100)
#routerrpc.attemptcost=10
# The (virtual) proportional cost in ppm of the total amount of a failed payment
# attempt (default: 1000)
#routerrpc.attemptcostppm=10
# Set the number of historical routing records
#routerrpc.maxmchistory=10000
# Set the min confidence in a path worth trying
#routerrpc.minrtprob=0.005
# Set the time to forget past routing failures
#routerrpc.penaltyhalflife=6h0m0s

all seem to be working again

I don't need this is optimization configuration?
Why did I get this crash after the update?

thank you

May 29 20:38:13 raspberrypi systemd[1]: Starting Lightning Network Daemon...
May 29 20:38:23 raspberrypi systemd[1]: Started Lightning Network Daemon.
May 29 20:46:58 raspberrypi lnd[10335]: 2023/05/29 20:46:58 http: TLS handshake error from 45.79.181.251:15814: EOF
May 29 20:50:11 raspberrypi systemd[1]: Stopping Lightning Network Daemon...
May 29 20:50:36 raspberrypi systemd[1]: lnd.service: Succeeded.
May 29 20:50:36 raspberrypi systemd[1]: Stopped Lightning Network Daemon.
May 29 20:50:36 raspberrypi systemd[1]: lnd.service: Consumed 4min 31.852s CPU time.
May 29 20:50:36 raspberrypi systemd[1]: Starting Lightning Network Daemon...
May 29 20:50:44 raspberrypi systemd[1]: Started Lightning Network Daemon.

openoms, 2023-05-29:
that's a custom config you have set and LND changed the options.

Omicron59, 2023-05-29:
I use the same manipulation for raspiblitz updates, suddenly is it possible to put back an original configuration not personalized?

openoms, 2023-05-29:
find the defaults (except the Tor settings) in
https://github.com/rootzoll/raspiblitz/blob/v1.9/home.admin/assets/lnd.bitcoin.conf

closed 2023-06-09 by Omicron59