raspiblitz/raspiblitz #3881: switching between CLN<->LND

jalbstmeijer, 2023-05-30:
I'm currently running version 1.8 with CLN.

Can I just switch to LND without loosing anything?
How would I do such switch?

wchancao, 2023-05-30:
You cant switch between lnd and cln... You gotta close channels and create a new node with lnd

jalbstmeijer, 2023-05-30:
@wchancao thank you for your reply.

I have no channels yet.

So I hoped someone could assure me the balance on my wallet will not be affected.
And if there are some commands or menu options to switch to LND without re-installing raspiblitz.

wchancao, 2023-05-30:
Then you dont have problems, what you gotta do is activate lnd, generate an onchain address within lnd and send all your sats to that lnd address from your cln onchain wallet, then you can deactivate cln. You dont have to reinstall raspiblitz

closed 2023-06-17 by rootzoll