raspiblitz/raspiblitz #3884: V1.9.0 CORE LIGHTNING turns to "Not Started | Not Ready Yet | No Data" after a few hours

dan-str, 2023-05-30:
The first few hours node is running fine and working.
After app. 6 hours CLN stops. The failure was repeated multiple times already on this node.

The status screen then says:

Blocks xxxxxy/xxxxxy Sync OK 100.00%
   
CLN v23.02.2 
Not Started | Not Ready Yet | No Data

[image]

When returning to the menu it shows Blockchain 100% and  Lightning 100%.

[image]

Debug -l:
https://l.termbin.com/fmji

 I saw the issue #3879 and #3882 but they are with LND. This one is with CLN.

wchancao, 2023-05-30:
You run out of space on your sd card... Thats why your cln backup plugin says theres an error writing in the database

dan-str, 2023-05-31:
Looks good with a bigger SD-Card. I now tried a SanDisk Ultra 128GB microSDXC and it's running.
But question is, what can be removed from the 32GB SD card to make it run again? Or does the recommendation needs to be changed to bigger SD cards?

surfac3, 2023-05-31:
i think this issue is connected to hsm secret gets encrypted .. do you by any chance get this error? in the logs? 

Did the CLN node by any chance stop with error in the log like this one? 
2023-05-27T17:43:36.445Z DEBUG   hsmd: pid 2555751, msgfd 48
2023-05-27T17:43:36.501Z DEBUG   hsmd: capability +WIRE_HSMD_CHECK_PUBKEY
2023-05-27T17:43:36.501Z DEBUG   hsmd: capability +WIRE_HSMD_SIGN_ANY_DELAYED_PAYMENT_TO_US
2023-05-27T17:43:36.695Z INFO    plugin-backup.py: root        : Comparing backup version 3858405 versus first write version 3858406
2023-05-27T17:43:36.696Z INFO    plugin-backup.py: Comparing backup version 3858405 versus first write version 3858406
2023-05-27T17:43:36.696Z INFO    plugin-backup.py: root        : Versions match up
2023-05-27T17:43:36.696Z INFO    plugin-backup.py: Versions match up

BitromNO, 2023-06-01:
CLN sync on new node with proxmox is on day 3 now. 98%.. not that slow last time i checked, a friend of mine gave up and removed CLN..

dan-str, 2023-06-03:
[image]

It's also filling the 128 GB the SD-card after some days, again same failure.

openoms, 2023-06-07:
we will need to find which file is growing to be able to take care of it.

Can list the files bigger than 500MB (eith the `/mnt/hdd` dir excluded):
```
sudo find / -type d \( -path "/mnt/hdd" -o -path "/mnt/hdd/*" \) -prune -o -type f -size +500M -exec ls -lh {} \;
```
Or use the tool ncdu for a graphical file browser:
```
# installl
sudo apt-get install ncdu
# run, srating in the root directory
sudo ncdu /
```

wisegorilla, 2023-06-07:
@openoms @dan-str On my install, the issue was two log files in /var/log growing very quickly, filling with constant errors that appeared to be related to the Blitz API/WebUI.  I disabled that option in the console menu and the issue went away.  I can try to re-enable and get those errors again if that would help reproduce the issue.

giybs372, 2023-06-10:
> @openoms @dan-str On my install, the issue was two log files in /var/log growing very quickly, filling with constant errors that appeared to be related to the Blitz API/WebUI. I disabled that option [...]

I can't even get into the console menu now - it keeps stopping at 100% on both - I've reflashed and installed again but no different, it has synced a couple times but after a while it always goes back to the 100% progress with 16 or 17 peers. the LCD shows "Not Started | Not Ready Yet | No Data"

cstenglein, 2023-06-12:
Most likely also related to #3879

rootzoll, 2023-08-05:
OK can you test if the available v1.10rc2 is fixing this - download here: https://github.com/raspiblitz/raspiblitz/tree/v1.10#downloading-the-software

psalzAppDev, 2023-08-31:
> @openoms @dan-str On my install, the issue was two log files in /var/log growing very quickly, filling with constant errors that appeared to be related to the Blitz API/WebUI. I disabled that option [...]

Thanks, disabling Blitz API and WebUI seems to have fixed this issue for me. (Raspiblitz v1.9.0)

closed 2023-09-14 by rootzoll