raspiblitz/raspiblitz #3885: After Update to Raspiblitz 1.9, LND stucks at "prepare syny"; RPC service not available

Marcoblue, 2023-05-31:
I did update Raspiblitz to 1.9 with new SD-card and FATPACK today.

After restart I got stuck at:
[image]

At lndlog I see every second:
2023-05-31 15:30:24.072 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-05-31 15:30:24.072 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available

My debug log:
https://l.termbin.com/7u0n

Can you help me to find the problem? Thanks.

Marcoblue, 2023-05-31:
journalctl -u lnd:
May 31 14:17:46 raspberrypi systemd[1]: Starting Lightning Network Daemon...
May 31 14:37:47 raspberrypi systemd[1]: lnd.service: start operation timed out. Terminating.
May 31 15:01:28 raspberrypi lnd[11525]: error creating wallet config: unable to set up wallet password listeners: shutting down
May 31 15:01:28 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
May 31 15:01:28 raspberrypi systemd[1]: lnd.service: Failed with result 'timeout'.
May 31 15:01:28 raspberrypi systemd[1]: Failed to start Lightning Network Daemon.
May 31 15:01:28 raspberrypi systemd[1]: lnd.service: Consumed 9min 12.287s CPU time.
May 31 15:02:28 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 1.
May 31 15:02:28 raspberrypi systemd[1]: Stopped Lightning Network Daemon.
May 31 15:02:28 raspberrypi systemd[1]: lnd.service: Consumed 9min 12.287s CPU time.
May 31 15:02:28 raspberrypi systemd[1]: Starting Lightning Network Daemon...
May 31 15:22:29 raspberrypi systemd[1]: lnd.service: start operation timed out. Terminating.
May 31 15:40:27 raspberrypi lnd[254908]: error creating wallet config: unable to set up wallet password listeners: shutting down
May 31 15:40:27 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
May 31 15:40:27 raspberrypi systemd[1]: lnd.service: Failed with result 'timeout'.
May 31 15:40:27 raspberrypi systemd[1]: Failed to start Lightning Network Daemon.
May 31 15:40:27 raspberrypi systemd[1]: lnd.service: Consumed 9min 5.759s CPU time.
May 31 15:41:27 raspberrypi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 2.
May 31 15:41:27 raspberrypi systemd[1]: Stopped Lightning Network Daemon.
May 31 15:41:27 raspberrypi systemd[1]: lnd.service: Consumed 9min 5.759s CPU time.
May 31 15:41:27 raspberrypi systemd[1]: Starting Lightning Network Daemon...
May 31 16:01:28 raspberrypi systemd[1]: lnd.service: start operation timed out. Terminating.
May 31 16:18:38 raspberrypi lnd[478855]: error creating wallet config: unable to set up wallet password listeners: shutting down
May 31 16:18:38 raspberrypi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
May 31 16:18:38 raspberrypi systemd[1]: lnd.service: Failed with result 'timeout'.
May 31 16:18:38 raspberrypi systemd[1]: Failed to start Lightning Network Daemon.

Marcoblue, 2023-06-01:
I downloaded the package again (with SHA256 check), wrote it to another SD-Card (previewsly used) and started again. But after some houres its the same status again.

Marcoblue, 2023-06-02:
I have started a new attempt. This time with the minimal version. Again I can't get past "prepare sync", but the logs look similar. 
https://l.termbin.com/aa6s

I can't get any further. Any ideas to localize the problem (RPC service not available)?

rootzoll, 2023-06-03:
Please try: CTRL+c to exit to terminal. the `menu` to force start the SSH menu. The choose "PASSWORD" > "RPC/APP Password" to reset your password B. Then restart.

Marcoblue, 2023-06-04:
Thanks rootzoll. But nothing changed. It also did not ask about PW B after restart.

New log:
https://l.termbin.com/7917t

rootzoll, 2023-06-04:
@openoms I just checked the `lnd.conf` on my v1.9.0 test raspiblitz - it does not look good and it could be related to thsi issue I guess:
[image]
[image]

@Marcoblue can you also check your `lnd.conf` with `﻿sudo cat /mnt/hdd/lnd/lnd.conf`

Marcoblue, 2023-06-04:
```
rpcmiddleware.enable=true
# lnd configuration
[Application Options]
accept-keysend=true
listen=0.0.0.0:9735
tlsdisableautofill=1
tlsautorefresh=1
stagger-initial-reconnect=1
ignore-historical-gossip-filters=1
debuglevel=debug
# maxpendingchannels=5
alias=sample
color=#6113CB
nat=false

# The maximum number of incoming pending channels permitted per peer.
maxpendingchannels=1

# The smallest channel size (in satoshis) that we should accept. Incoming
# channels smaller than this will be rejected, default value 20000.
minchansize=490000

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# REST open to all connections on Port 8080
restlisten=0.0.0.0:8080
# Domain, could use https://freedns.afraid.org
#tlsextradomain=lightning.yourhost.com

# If true, our node will allow htlc forwards that arrive and depart on the same
# channel.
allow-circular-route=true

# The smallest HTLC we are willing to accept on our channels, in millisatoshi.
# bitcoin.minhtlc=999

[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

# Fee estimate mode for bitcoind. It must be either "ECONOMICAL" or "CONSERVATIVE".
# If unset, the default value is "CONSERVATIVE".
#bitcoind.estimatemode=ECONOMICAL

[autopilot]
autopilot.active=0
autopilot.maxchannels=5
autopilot.allocation=0.5

# The smallest channel that the autopilot agent should create (default: 20000)
#autopilot.minchansize=1000000

# The largest channel that the autopilot agent should create (default: 16777215)
#autopilot.maxchansize=2000000

# The maximum number of incoming pending channels permitted per peer.
#maxpendingchannels=1

# Fee estimate mode for bitcoind. It must be either "ECONOMICAL" or "CONSERVATIVE".
# If unset, the default value is "CONSERVATIVE".
#bitcoind.estimatemode=ECONOMICAL

[Wtclient]
wtclient.active=1

# Specify the fee rate with which justice transactions will be signed. This fee
# rate should be chosen as a maximum fee rate one is willing to pay in order to
# sweep funds if a breach occurs while being offline. The fee rate should be
# specified in sat/byte, the default is 10 sat/byte.
wtclient.sweep-fee-rate=99

[Watchtower]
watchtower.active=1

[bitcoind]
bitcoind.rpchost=127.0.0.1:8332
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=Co53AgnXcT32bPmF5V
bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332
bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333

[tor]
tor.active=true
tor.v3=true
tor.streamisolation=true
tor.privatekeypath=/mnt/hdd/lnd/v3_onion_private_key
tor.socks=9050
tor.control=9051

[bolt]
db.bolt.auto-compact=true
db.bolt.auto-compact-min-age=672h

[workers]
workers.sig=4
workers.write=4

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

  [rpcmiddleware]

```

nickshredaway, 2023-06-07:
I think I had this issue. I tried to solve by going back to v1.8, as I just couldnt seem v1.9 to play nice. v1.8 loads up well but can't connect to LND, because it doesnt want to go back in versions. Is there a way to boot v1.8 and then update to latest LND version?

nickshredaway, 2023-06-07:
FWIW when I try to boot v1.9 it either just disconnects me from SSH during setup 'closed by remote host' or gets stuck on 'preparing provision' but I cant even seem to get a fresh v1.9 flash to work. again 1.8 works but with the LND version problem stated above

openoms, 2023-06-16:
apart from applying the patch in #3892

can clean the lnd conf with:
```
#remove "  [rpcmiddleware]"
sudo sed -i '^  /\[rpcmiddleware\]/d' /home/bitcoin/.lnd/lnd.conf
# remove empty lines
sudo sed -i '/^\s*$/d' /home/bitcoin/.lnd/lnd.conf
```

Marcoblue, 2023-06-17:
Thank you.

1. I changed lnd.check.sh as described in https://github.com/raspiblitz/raspiblitz/issues/3885
2. I cleand lnd.conf as desribed above
3. I did a reboot
4. After some houres it syned the blockchain again, but stucked with lnd. So I changed RPC password and did reboot again. (Because it still never asked about PW_B)
5. New debug-log: https://l.termbin.com/o5s1

sudo cat /mnt/hdd/lnd/lnd.conf
```
rpcmiddleware.enable=true
# lnd configuration
[Application Options]
accept-keysend=true
listen=0.0.0.0:9735
tlsdisableautofill=1
tlsautorefresh=1
stagger-initial-reconnect=1
ignore-historical-gossip-filters=1
debuglevel=debug
# maxpendingchannels=5
alias=OneWorld2199
color=#6113CB
nat=false
# The maximum number of incoming pending channels permitted per peer.
maxpendingchannels=1
# The smallest channel size (in satoshis) that we should accept. Incoming
# channels smaller than this will be rejected, default value 20000.
minchansize=490000
# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# REST open to all connections on Port 8080
restlisten=0.0.0.0:8080
# Domain, could use https://freedns.afraid.org
#tlsextradomain=lightning.yourhost.com
# If true, our node will allow htlc forwards that arrive and depart on the same
# channel.
allow-circular-route=true
# The smallest HTLC we are willing to accept on our channels, in millisatoshi.
# bitcoin.minhtlc=999
[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1
# Fee estimate mode for bitcoind. It must be either "ECONOMICAL" or "CONSERVATIVE".
# If unset, the default value is "CONSERVATIVE".
#bitcoind.estimatemode=ECONOMICAL
[autopilot]
autopilot.active=0
autopilot.maxchannels=5
autopilot.allocation=0.5
# The smallest channel that the autopilot agent should create (default: 20000)
#autopilot.minchansize=1000000
# The largest channel that the autopilot agent should create (default: 16777215)
#autopilot.maxchansize=2000000
# The maximum number of incoming pending channels permitted per peer.
#maxpendingchannels=1
# Fee estimate mode for bitcoind. It must be either "ECONOMICAL" or "CONSERVATIVE".
# If unset, the default value is "CONSERVATIVE".
#bitcoind.estimatemode=ECONOMICAL
[Wtclient]
wtclient.active=1
# Specify the fee rate with which justice transactions will be signed. This fee
# rate should be chosen as a maximum fee rate one is willing to pay in order to
# sweep funds if a breach occurs while being offline. The fee rate should be
# specified in sat/byte, the default is 10 sat/byte.
wtclient.sweep-fee-rate=99
[Watchtower]
watchtower.active=1
[bitcoind]
bitcoind.rpchost=127.0.0.1:8332
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=Co53AgnXcT32bPmF5V1
bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332
bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333
[tor]
tor.active=true
tor.v3=true
tor.streamisolation=true
tor.privatekeypath=/mnt/hdd/lnd/v3_onion_private_key
tor.socks=9050
tor.control=9051
[bolt]
db.bolt.auto-compact=true
db.bolt.auto-compact-min-age=672h
[workers]
workers.sig=4
workers.write=4

[rpcmiddleware]
```

rootzoll, 2023-06-17:
pr #3892 merged - also add clean up on bootstrap

Marcoblue, 2023-06-18:
I'm not very confident anymore that I can get this to work again. What is the best way to save my channels and onchainfunds? 

At the beginning I had 2 open channels but one was closed 2 days ago by the partner. 

I have the lnd-recue-file available.

Unfortunately I don't have a 2nd raspiblitz available. And before i flatten this one, i want to know what exactly i have to do then.

I think there is no point to import the backup on this broken raspiblitz? Or can I just force close the channel here? How can I transfer the onchain funds then?

openoms, 2023-07-02:
> I'm not very confident anymore that I can get this to work again. What is the best way to save my channels and onchainfunds?
>
> At the beginning I had 2 open channels but one was closed 2 days ago by [...]

@Marcoblue this question is about standard LND recovery options covred in the FAQ. Hope you made it work since.

openoms, 2023-07-02:
> apart from applying the patch in #3892
> 
> can clean the lnd conf with:
> 
> ```
> # remove "  [rpcmiddleware]"
> sudo sed -i '/^  \[rpcmiddleware\]/d' /home/bitcoin/.lnd/lnd.conf
> # remove empty lines
> sudo sed -i '/^\s*$/d' /home/bitcoin/.lnd/lnd.conf
> ```

@rootzoll these might be useful to have in the lnd.check.sh also to clear the existing lnd.conf files automatically in the next bugfix release.

mr-q17, 2023-07-05:
I have the same problem... after some off-days of a working Raspiblitz on RPI4B.
I´ve done the workaround form @openoms like he wrote here: https://github.com/raspiblitz/raspiblitz/issues/3885#issuecomment-1616449810

plus: reset password b (that is not asked within the startup before and after the changes)
plus: Touchscreen turned off by menu

But that changes nothing with the main problem.

Snippet (LND part) from the debug report: menu -> repair -> software
```
*** LND (MAINNET) SYSTEMD STATUS ***
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
     Active: activating (start) since Wed 2023-07-05 12:31:18 BST; 10min ago
    Process: 17009 ExecStartPre=/home/admin/config.scripts/lnd.check.sh prestart mainnet (code=exited, status=0/SUCCESS)
   Main PID: 17303 (lnd)
      Tasks: 13 (limit: 4531)
        CPU: 10.446s
     CGroup: /system.slice/lnd.service
             └─17303 /usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.conf

Jul 05 12:31:18 raspberrypi systemd[1]: Starting Lightning Network Daemon...

*** LAST LND (MAINNET) ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Journal begins at Thu 2023-06-29 08:25:24 BST, ends at Wed 2023-07-05 12:41:30 BST. --
Jul 05 12:31:18 raspberrypi systemd[1]: Starting Lightning Network Daemon...

*** LAST LND (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
2023-07-05 12:40:41.484 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:41.484 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:40:43.494 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:43.494 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:40:45.507 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:45.507 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:40:47.519 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:47.520 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:40:49.532 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:49.532 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:40:51.543 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:51.543 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:40:53.559 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:53.560 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:40:55.592 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:55.592 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:40:57.632 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:57.632 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:40:59.649 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:40:59.649 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:01.670 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:01.671 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:03.682 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:03.682 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:05.697 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:05.697 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:07.710 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:07.710 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:09.721 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:09.721 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:11.732 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:11.732 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:13.749 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:13.749 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:15.761 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:15.762 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:17.783 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:17.783 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:19.825 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:19.825 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
[... 6 lines cut ...]
2023-07-05 12:41:27.902 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:27.902 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available
2023-07-05 12:41:29.918 [DBG] RPCS: [/lnrpc.Lightning/GetInfo] requested
2023-07-05 12:41:29.918 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: waiting to start, RPC services not available

*** CORE LIGHTNING (MAINNET) SYSTEMD STATUS ***
- not activated -

*** BLOCKCHAIN (TESTNET) SYSTEMD STATUS ***
- OFF by config -
*** LND (TESTNET) SYSTEMD STATUS ***
- OFF by config -

*** CORE LIGHTNING (TESTNET) SYSTEMD STATUS ***
- not activated -

*** BLOCKCHAIN (SIGNET) SYSTEMD STATUS ***
- OFF by config -
*** LND (SIGNET) SYSTEMD STATUS ***
- OFF by config -

*** CORE LIGHTNING (SIGNET) SYSTEMD STATUS ***
- not activated -
```
[image]

Marcoblue, 2023-07-24:
I try to do it like described here: https://github.com/raspiblitz/raspiblitz/blob/v1.9/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

But: then run: /home/admin/config.scripts/lnd.backup.sh lnd-import-gui
gives: error='missing parameter'

Is it tha same as from this menu:
[image]

Marcoblue, 2023-07-24:
I used LNDRESCUE and Channel.db import from the menu above. After everything was done, I got the same error as above (RPC service not available). So I did the same again with the following changes:
- Raspiblitz 1.8.0c
- Used the same node name as the old one
- Used identical passwords like the old one
After it was running and channel.db was imported, I updated lnd to 0.16.4 from the menu, because the versions were not compatible.

After reboot LND is running again. However, it still hangs at scanning blocks 100.00 (yellow). I think this will take some time. But I could already access the onchain coins.

rootzoll, 2023-08-05:
OK can you test if the available v1.10rc2 is fixing this - download here: https://github.com/raspiblitz/raspiblitz/tree/v1.10#downloading-the-software

PR #3892 (2023-06-17): fix indentation of lnd.conf
fixes: #3885
closed 2023-09-14 by rootzoll