raspiblitz/raspiblitz #3891: CPU-Load Monitoring & Warning

rootzoll, 2023-06-04:
If the CPU load gets too high (reported constant values on 8) the RaspberryPi gets hot and even more throttles down resulting in possible HDD/SSD operations failing.

The idea would be to add a CPU monitorinf and give warining to "reduce number of apps" to the user.

rootzoll, 2023-06-04:
And/or if CPU load shows +4 on a raspberrypi - the SSH or WebUI should maybe ask if the user really wants to install further apps.

quarreller, 2023-06-06:
My english is not very good :)
But since updating to v1.9,0 final the cpu load is often about 100% and i have only installed RTL an Mempool.space and BTC Exploreer ...on a Raspi 4 8GB....right now 112% by 15 in and 10 outgoing connections,

woeisme, 2023-07-01:
Came here to say similar, upgraded last night and 24 hours later the cpu load and temp are pegged high and I cannot figure out why.