raspiblitz/raspiblitz #3949: 1.9 Raspiblitz on Proxmox syncs Bitcoin Core 100% and cannot connect to peers after a few days

BitromNO, 2023-06-08:
Is it a network related issue? and why does this happen after a day or two? 

2023-06-07T17:55:52Z Config file arg: peerbloomfilters="1"
2023-06-07T17:55:52Z Config file arg: rpcallowip="127.0.0.1"
2023-06-07T17:55:52Z Config file arg: rpcallowip="enp0s18...0/24"
2023-06-07T17:55:52Z Config file arg: rpcpassword=****
2023-06-07T17:55:52Z Config file arg: rpcuser=****
2023-06-07T17:55:52Z Config file arg: server="1"
2023-06-07T17:55:52Z Config file arg: testnet="0"
2023-06-07T17:55:52Z Config file arg: txindex="0"
2023-06-07T17:55:52Z Config file arg: zmqpubrawblock="tcp://127.0.0.1:28332"
2023-06-07T17:55:52Z Config file arg: zmqpubrawtx="tcp://127.0.0.1:28333"
2023-06-07T17:55:52Z Config file arg: [main] rpcbind=****
2023-06-07T17:55:52Z Config file arg: [main] rpcbind=****
2023-06-07T17:55:52Z Config file arg: [main] rpcport="8332"
2023-06-07T17:55:52Z Config file arg: [main] wallet="wallet.dat"
2023-06-07T17:55:52Z Config file arg: [test] rpcbind=****
2023-06-07T17:55:52Z Config file arg: [test] rpcport="18332"
2023-06-07T17:55:52Z Command-line arg: conf="/mnt/hdd/bitcoin/bitcoin.conf"
2023-06-07T17:55:52Z Command-line arg: daemonwait=""
2023-06-07T17:55:52Z Command-line arg: datadir="/mnt/hdd/bitcoin"
2023-06-07T17:55:52Z Command-line arg: debuglogfile="/mnt/hdd/bitcoin/debug.log"
2023-06-07T17:55:52Z Using at most 40 automatic connections (1024 file descriptors available)
2023-06-07T17:55:52Z Using 16 MiB out of 16 MiB requested for signature cache, able to store 524288 elements
2023-06-07T17:55:52Z Using 16 MiB out of 16 MiB requested for script execution cache, able to store 524288 elements
2023-06-07T17:55:52Z Script verification uses 3 additional threads
2023-06-07T17:55:52Z scheduler thread start
2023-06-07T17:55:52Z Error: Invalid -rpcallowip subnet specification: enp0s18...0/24. Valid are a single IP (e.g. 1.2.3.4), a network/netmask (e.g. 1.2.3.4/255.255.255.0) or a network/CIDR (e.g. 1.2.3.4/24).
2023-06-07T17:55:52Z Error: Unable to start HTTP server. See debug log for details.
2023-06-07T17:55:52Z Shutdown: In progress...
2023-06-07T17:55:52Z scheduler thread exit
2023-06-07T17:55:52Z Shutdown: done

dalamanster, 2023-06-09:
I assume these two lines are related. Can you fix the IP reference?

```
2023-06-07T17:55:52Z Config file arg: rpcallowip="enp0s18...0/24"
....
2023-06-07T17:55:52Z Error: Invalid -rpcallowip subnet specification: enp0s18...0/24. Valid are a single IP (e.g. 1.2.3.4), a network/netmask (e.g. 1.2.3.4/255.255.255.0) or a network/CIDR (e.g. 1.2.3.4/24).

```

BitromNO, 2023-06-09:
> I assume these two lines are related. Can you fix the IP reference?
>
> ```
> 2023-06-07T17:55:52Z Config file arg: rpcallowip="enp0s18...0/24"
> ....
> 2023-06-07T17:55:52Z Error: Invalid -rpcallowip subnet [...]

Setting static ip somehow?

rootzoll, 2024-09-12:
@BitromNO I fixed the localIP range detection for Proxmox for the next RaspiBlitz version 1.11.3 to prevent faulty entries like your in the `bitcoin.conf`

If you still have that problem you need to edit your bitcoin.conf (MAINMENU > SYSTEM > bitcoinCONF) manually and remove the line with `rpcallowip=`

PR #4736 (2024-09-12): #3949 fix vm IPrange
fixes #3949
closed 2024-09-12 by rootzoll