raspiblitz/raspiblitz #3966: migration-failed from umbrel to raspiblitz

BTCtoTheMooon, 2023-06-14:
hi together,

i have a problem.. I want to migrate my umbrel to raspiblitz and did everything like described in the github. i even deinstalled everything on my umbrel except of bitcoin fullnode, mempool and electrum, to make the migration as easy as possible (what i thought). I even deleted my lightning LND app on umbrel after taking all the funds to another wallet.

I use a raspberry pi 4 8GM RAM with an 1TB SSD (with the blockchain on it) and a 64GB SD-card

i did it like described:

-shut down umbrel -took out the SD 
-flashed SD-card with balenaEtcher with the latest version of raspiblitz (i used the fat-version) after deleting the umbrel data from the SD 
-SD back to my raspberry pi 
-look after the IP with angry IP Scanner 
-open IP Adress with webbrowser / started with ssh (i tried both)
-setting the new passwords 
-start migrating 
(everything fine to this point, except of a few Problems with finding the IP, which i quickly fixed)

But than after 2min: “error migration-failed” with a loading symbol in the webbrowser
and an error-text in the terminal, when i tried with ssh (after i fixed the host change issue with "ssh-keygen -R "ip-adress"").

I would be so glad if someone could help me. i really dont want to download the blockchain again haha it took me 6 days.

thanks in advance and have a awesome day!

### DEBUG.LOG:

# redacting file: /var/cache/raspiblitz/debug.log
backup info: network
backup info: chain

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.9.0
chainnetwork: bitcoin / test
 20:13:01 up 57 min,  2 users,  load average: 1.08, 1.09, 1.08

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> migration
state--> error
/dev/sda1      983378332 648690500 284661288  70% /mnt/hdd
tmpfs             805088        16    805072   1% /run/user/1001
# lsblk -o NAME,FSTYPE,LABEL
NAME        FSTYPE LABEL
sda
└─sda1      ext4   umbrel
mmcblk0
├─mmcblk0p1 vfat   bootfs
└─mmcblk0p2 ext4   rootfs
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
# migrationFile()
# Sourcing /mnt/hdd/raspiblitz.conf
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
# setting PASSWORD A
# salt file exists
# hashedPasswordSalt(3e6dba4f94c67f06)
runningUser='0'
# OK - password A changed for user ***, root, admin & bitcoin
error=''
Calling _provision.migration.sh for possible migrations ..

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Thu 2023-05-11 16:58:36 BST, ends at Wed 2023-06-14 20:13:01 BST. --
[... 188 lines cut ...]

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0

*** SYSTEM CACHE STATUS ***
system_init_time="1686766572"
system_ups_status="OFF"
system_temp_fahrenheit="118"
system_ups_battery=""
system_count_start_lightning="0"
system_count_longscan="0"
system_count_start_tui="0"
system_ram_available_mb="7470"
system_ram_gb="8"
system_ram_mb="7862"
system_up="3448"
system_temp_celsius="48"
system_vm_vagrant="0"
system_cpu_load=" 1.27, 1.13, 1.09"
system_count_undervoltage="0"
system_board="rp4"
system_count_start_blockchain="0"
btc_default_sync_percentage="0"
btc_default_peers="0"
btc_default_port=""

rootzoll, 2023-08-28:
please retry with the latest v1.10.0rc4 that you can download here: https://github.com/raspiblitz/raspiblitz/tree/v1.10#downloading-the-software

tlindi, 2023-11-16:
> @tlindi I have same issue, did you find any solutions?

Not anything I could recommend. (Cause I don't remember). Anyhow it was manual work on commandline related to file / folder locations or chmod & chown.

Anyhow result was success without resync.