raspiblitz/raspiblitz #3968: After update to v1.9.0 - CL Lightning Wallet can not be unlocked

Minervaaaaa, 2023-06-16:
This message I get:
A dependency job for lightningd.service failed. See 'journalctl -xe' for details.

This shows my command line:
[image]

[image]

The correct password C was used and I restarted the raspiblitz already via command line. 
Any ideas what could help?

openoms, 2023-06-17:
The following log lines would help a lot. Please share here if not sensitive.

Minervaaaaa, 2023-06-18:
[image]

[image]

Minervaaaaa, 2023-06-18:
[image]

Minervaaaaa, 2023-06-18:
> The following log lines would help a lot. Please share here if not sensitive.
Are these the right logs you need to look through? Thanks a lot!

openoms, 2023-06-18:
> Are these the right logs you need to look through? Thanks a lot!

Not really. When you enter the CLN menu after a minute there is there are the last lines of the journal for CLN and cl.log are shown. Have a look at and paste thise please.

Minervaaaaa, 2023-06-22:
[image]

Looks like I get the same message...
Is any fix for this already available?

0xtrr, 2023-06-22:
I ended up with doing a backup of the CLN data, deleting all data from the HDD besides the blockchain, and then reinstalling the node as a "fresh" node. After the reinstall I restored the CLN node with the backup file. Seems to work just fine now.

rootzoll, 2023-08-11:
@openoms is in this issue still something relevant for v1.10.0?

openoms, 2023-08-11:
Didn't get to reproduce this problem. Closing, please reopen if comes up again.

closed 2023-08-11 by openoms