raspiblitz/raspiblitz #3970: [v1.9.0] Data is not in recognized format (wallet.dat)

adevoss, 2023-06-18:
After upgrading from v1.8 with existing data, provisioning was successful.
Now there are problems with syncing.

Reading other issues it is similar to others but no issues with disk space.
Running on a amd64 VM. I don't think that is relevant here but to be sure.

debug log shows:
`Error: Failed to load database path '/mnt/hdd/bitcoin/wallet.dat'. Data is not in recognized format.`

`-rw------- 1 bitcoin bitcoin 12K Jun 10 12:41 /mnt/hdd/bitcoin/wallet.dat
`
Blockchain Progress : 179 restarts
Lightning Progress  : prepare sync

Info screen:
both bitcoin and LND
Not Started | Not Ready Yet | No Data

openoms, 2023-06-18:
Not sure what is causing this but is resolved by simply deleting the wallet and restart bitcoind.
```
sudo mv /mnt/hdd/bitcoin/wallet.dat.backup
sudo systemcctl restart bitcoind
```

adevoss, 2023-06-19:
When I do that my sats are lost. Don't like that. Can I safely downgrade to v1.8.0 if `/mnt/hdd/raspiblitz.conf` is v1.9.0?

openoms, 2023-06-21:
> Is it safe to actually delete the wallet.dat file @openoms ?

move it to a .backup file if unsure.

The default bitcoind wallet is not used to store private keys.
Onpy Joinmarket/Jam uses it as watch only, but obviously it wouldn't wouldn't work if corrupted.
Worst case is that need would need to do an import and rescan from Joininbox to recover the watch only address balances.

0xtrr, 2023-06-21:
> move it to a .backup file if unsure.
>
> The default bitcoind wallet is not used to store private keys. Onpy Joinmarket/Jam uses it as watch only, but obviously it wouldn't wouldn't work if corrupted. [...]

I did this but I'm still getting the same error after restarting the service
```
2023-06-21T16:19:50Z Error: Failed to load database path '/mnt/hdd/bitcoin'. Data is not in recognized format.
```

adevoss, 2023-06-22:
Saw the `blitz.backupdrive.sh` script. It does not solve this issue but that started me thinking of a good backup procedure.
Now working on a few small pr's for that script and a backup script.

0xtrr, 2023-06-23:
I got it working again: https://github.com/raspiblitz/raspiblitz/issues/3968#issuecomment-1602783058

adevoss, 2023-06-24:
I use lnd instead of cln...

Pubskiste, 2023-07-03:
Any updates here? Luckily this only happened on my test node. And only, when I flashed the SD for a second time after already ran successfully on v1.9.0.
``` 
sudo mv /mnt/hdd/bitcoin/wallet.dat /mnt/hdd/bitcoin/wallet.dat.backup
sudo systemctl restart bitcoind
```
just changes the error slightly to:
```
Error: Failed to load database path '/mnt/hdd/bitcoin'. Data is not in recognized format.
```
(missing the term `wallet.dat`)
I don't mind using the LNDRESCUE backup, but I don't like the idea getting into trouble next time updating my main node. To know the root cause and how to avoid it in future would be great.

openoms, 2023-07-03:
Unfortunately if the bitcoin data directory appears corrupted there is no other way apart from resyncing the chain. 
Can try to reset the only the blockchain from the REPAIR menu or do a migration / lightning database backup and format the disk to clean the whole filesystem.

The cause of this is usually an unclean shutdown, power problems or hardware failure. Can't tell remotely from limited info.

openoms, 2023-07-04:
Got this after a recovery to v1.9:
```
2023-07-04T07:26:28Z New outbound peer connected: version: 70016, blocks=797109, peer=0 (block-relay-only)
2023-07-04T07:26:38Z New outbound peer connected: version: 70016, blocks=797109, peer=2 (block-relay-only)
2023-07-04T07:27:02Z IsBDBFile: Is a directory /mnt/hdd/bitcoin/wallet.dat
2023-07-04T07:27:02Z IsSQLiteFile: Is a directory /mnt/hdd/bitcoin/wallet.dat
2023-07-04T07:27:02Z Using SQLite Version 3.38.5
2023-07-04T07:27:02Z Using wallet /mnt/hdd/bitcoin
2023-07-04T07:27:02Z SQLite Error. Code: 14. Message: cannot open file at line 41010 of [78d9c993d4]
2023-07-04T07:27:02Z SQLite Error. Code: 14. Message: os_unix.c:41010: (21) open(/mnt/hdd/bitcoin/wallet.dat) - Is a directory
2023-07-04T07:27:10Z New outbound peer connected: version: 70016, blocks=797109, peer=4 (outbound-full-relay)
2023-07-04T07:27:18Z New outbound peer connected: version: 70016, blocks=797109, peer=5 (outbound-full-relay)
2023-07-04T07:27:32Z New outbound peer connected: version: 70016, blocks=797109, peer=6 (outbound-full-relay)
2023-07-04T07:27:43Z IsBDBFile: Is a directory /mnt/hdd/bitcoin/wallet.dat
2023-07-04T07:27:43Z IsSQLiteFile: Is a directory /mnt/hdd/bitcoin/wallet.dat
2023-07-04T07:27:43Z Using SQLite Version 3.38.5
2023-07-04T07:27:43Z Using wallet /mnt/hdd/bitcoin
2023-07-04T07:27:43Z SQLite Error. Code: 14. Message: cannot open file at line 41010 of [78d9c993d4]
2023-07-04T07:27:43Z SQLite Error. Code: 14. Message: os_unix.c:41010: (21) open(/mnt/hdd/bitcoin/wallet.dat) - Is a directory
```

Pubskiste, 2023-07-04:
@openoms 
Doesn't work for me. Regardless of 
```
main.walletdir=/mnt/hdd/bitcoin
```
set in `bitcoin.conf` or not, the error is
```bash
Error: Failed to load database path '/mnt/hdd/bitcoin'. Data is not in recognized format.
```
i case the is no file named `wallet.dat` because of `sudo mv /mnt/hdd/bitcoin/wallet.dat /mnt/hdd/bitcoin/wallet.dat.backup`.
And it's
```bash
Error: Failed to load database path '/mnt/hdd/bitcoin/wallet.dat'. Data is not in recognized format.
```
in case `wallet.dat` is still in place.

openoms, 2023-07-04:
@Pubskiste try to use an absolute path for your wallet like:
```
main.wallet=/mnt/hdd/bitcoin/wallet.dat
```

Could experiment with different things and see what works.

Pubskiste, 2023-07-06:
## Solved it!

`/mnt/hdd/bitcoin/settings.json` had this entry:
```
{
    "wallet": [
        ""
    ]
}
```
Removing this empty wallet list resulting in
```
{
}
```
did the trick. 
After restarting bitcoind
```
sudo systemctl restart bitcoind
```
**My node is back! :-)**

### Remark
There is still the following error message persisting in `/mnt/hdd/bitcoin/debug.log`, but this seams to be okay:
```
Warning: Skipping -wallet path that doesn't exist. Failed to load database path '/mnt/hdd/bitcoin/wallet.dat'. Path does not exist.
```

## Root cause
I'm not sure by now. Especially because updating from v1.8.0 to v1.9.0 worked for the first time. Then an additional flashing of the sd card broke it. So when and why was `settings.json` changed?

AdamFerguson, 2023-07-09:
I was able to resolve this issue by modifying the bitcoin.conf and moving the existing wallet.dat. Previously the setting was:

```
main.wallet=wallet.dat
```

I commented this line out and then moved the `wallet.dat` to `wallet.dat.backup`

And then ran: `sudo systemctl restart bitcoind`

After restarting bitcoind regenerated the `wallet.dat`. If I left the file in place and only commented out that setting, it would still try to use the existing `wallet.dat` and fail. If I left the configuration in place but moved the file, it seems like bitcoind expected to find the wallet.dat file and failed.

adevoss, 2023-07-09:
> `/mnt/hdd/bitcoin/settings.json` had this entry:

Didn't worked for me. Also tried a reboot.

adevoss, 2023-07-09:
> I commented this line out and then moved the `wallet.dat` to `wallet.dat.backup`

Syncing now.

md-weber, 2023-07-17:
So it seems to be still an issue, after running all the things described here after 2 days of happy running node I am back at the 100 % problem.
[image]

### Logs
Sorry I forgot the command to upload the logs, but here are my debug.log.
```
2023-07-17T19:11:39Z [i2p] Error connecting to fhzlp3xroabohnmjonu5iqazwhlbbwh5cpujvw2azcu3srqdceja.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=fhzlp3xroabohnmjonu5iqazwhlbbwh5cpujvw2azcu3srqdceja.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=fhzlp3xroabohnmjonu5iqazwhlbbwh5cpujvw2azcu3srqdceja.b32.i2p"
2023-07-17T19:11:56Z Socks5() connect to wefeqxh5anummmg6qnf7dmttrm4ig4xd42ojpmjyrcranryjrxngeayd.onion:8333 failed: host unreachable
2023-07-17T19:12:56Z Socks5() connect to wefeqxh5anummmg6qnf7dmttrm4ig4xd42ojpmjyrcranryjrxngeayd.onion:8333 failed: host unreachable
2023-07-17T19:14:19Z [i2p] Error connecting to wjrul5jwwb4vqdmkkrjbmly7osj6amecdpsac5xvaoqrti4nb3ha.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=wjrul5jwwb4vqdmkkrjbmly7osj6amecdpsac5xvaoqrti4nb3ha.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=wjrul5jwwb4vqdmkkrjbmly7osj6amecdpsac5xvaoqrti4nb3ha.b32.i2p"
2023-07-17T19:15:50Z Socks5() connect to weystplbjpu3fuqnxmcuphzbnst47nrlwygv3yhdxm2zg7dilwyxytid.onion:8333 failed: host unreachable
2023-07-17T19:16:35Z [i2p] Error connecting to pohfcrfc7prn4bvn4xstw6nt3e7hjmb7kuj4djtsfqsskwhmhnna.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-07-17T19:16:52Z Socks5() connect to detb33r4ikug7mux6y4wk3rmb5tdf6s74jzi7umfehu7pyw6p5tqiead.onion:8333 failed: host unreachable
2023-07-17T19:18:59Z [i2p] Error connecting to yc4xwin5ujenvcr6ynwkz7lnmmq3nmzxvfguele6ovqqpxgjvonq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-07-17T19:21:52Z [i2p] Error connecting to 3gocb7wc4zvbmmebktet7gujccuux4ifk3kqilnxnj5wpdpqx2hq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-07-17T19:23:23Z [i2p] Error connecting to lrah7acdsgopybg43shadwwiv6igezaw64i6jb5muqdg7dmhj3la.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-07-17T19:24:37Z UpdateTip: new best=00000000000000000001f2e9ce69f708f1382807979873ce0c352c4ac404c587 height=799112 version=0x22b86000 log2_work=94.305939 tx=866141818 date='2023-07-17T19:24:14Z' progress=1.000000 cache=33.9MiB(183367txo)
2023-07-17T19:26:45Z [i2p] Error connecting to 4osyqeknhx5qf3a73jeimexwclmt42cju6xdp7icja4ixxguu2hq.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=4osyqeknhx5qf3a73jeimexwclmt42cju6xdp7icja4ixxguu2hq.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=4osyqeknhx5qf3a73jeimexwclmt42cju6xdp7icja4ixxguu2hq.b32.i2p"
2023-07-17T19:29:53Z [i2p] Error connecting to m6v454xd6p3bt5swujgmveklsp7lzbkqlqqfc2p36cjlwv5dbucq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-07-17T19:29:57Z Socks5() connect to wzxmhn3tlcxbz5wv3nzj7yx6jfg4zqvwtkrpuzejkiw6dsytkyyh2sad.onion:8333 failed: host unreachable
2023-07-17T19:30:06Z Socks5() connect to pzunpwvmeg3h7uwksesltmzfckoywjml4wbppr4ep7y45ygit6df4oad.onion:8333 failed: host unreachable
2023-07-17T19:30:16Z New outbound peer connected: version: 70016, blocks=799112, peer=6263 (outbound-full-relay)
2023-07-17T19:31:07Z UpdateTip: new best=00000000000000000001c689da27a6b19197d9f3c4b27b5a5beee30a33a51b65 height=799113 version=0x2e0ac000 log2_work=94.305952 tx=866145032 date='2023-07-17T19:30:19Z' progress=1.000000 cache=35.3MiB(194097txo)
2023-07-17T19:32:18Z [i2p] Error connecting to c4gfnttsuwqomiygupdqqqyy5y5emnk5c73hrfvatri67prd7vyq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-07-17T19:33:55Z UpdateTip: new best=00000000000000000002daa1e708138fa3549a6f076693ec12c3fe64f1d6437d height=799114 version=0x24928000 log2_work=94.305966 tx=866146432 date='2023-07-17T19:34:15Z' progress=1.000000 cache=35.4MiB(194765txo)
2023-07-17T19:35:14Z [i2p] Error connecting to liu75cvktv4icbctg72w7nxbk4eibt7wamizfdii4omz7gcke5vq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
2023-07-17T19:35:48Z UpdateTip: new best=00000000000000000004092d1c9d2eea725f8f99d6d4763ac784636bb7989f56 height=799115 version=0x20e00000 log2_work=94.305980 tx=866150394 date='2023-07-17T19:35:15Z' progress=1.000000 cache=36.3MiB(202103txo)
2023-07-17T19:35:57Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-17T19:35:57Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-17T19:36:25Z Socks5() connect to ylf6skso4urpfthjhkf2ivxcblbhqxnudrabt4iv5icuzoxs4jm324id.onion:8333 failed: InterruptibleRecv() timeout or other failure
2023-07-17T19:38:38Z Socks5() connect to q5bdxtyqdqpmgwqbvxkr4bsip2kbfcjplpbdgfjnagrju3dw42tr7nid.onion:8333 failed: InterruptibleRecv() timeout or other failure
2023-07-17T19:38:57Z [i2p] Error connecting to fhzlp3xroabohnmjonu5iqazwhlbbwh5cpujvw2azcu3srqdceja.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=fhzlp3xroabohnmjonu5iqazwhlbbwh5cpujvw2azcu3srqdceja.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=fhzlp3xroabohnmjonu5iqazwhlbbwh5cpujvw2azcu3srqdceja.b32.i2p"
2023-07-17T19:41:31Z [i2p] Error connecting to wjrul5jwwb4vqdmkkrjbmly7osj6amecdpsac5xvaoqrti4nb3ha.b32.i2p:0: Unexpected reply to "NAMING LOOKUP NAME=wjrul5jwwb4vqdmkkrjbmly7osj6amecdpsac5xvaoqrti4nb3ha.b32.i2p": "NAMING REPLY RESULT=INVALID_KEY NAME=wjrul5jwwb4vqdmkkrjbmly7osj6amecdpsac5xvaoqrti4nb3ha.b32.i2p"
2023-07-17T19:43:25Z UpdateTip: new best=00000000000000000001ad3ef3065adbaeccf45f292396d465d21398df230c3c height=799116 version=0x20a00000 log2_work=94.305993 tx=866153714 date='2023-07-17T19:42:18Z' progress=1.000000 cache=37.3 [... cut ...]
2023-07-17T19:43:52Z UpdateTip: new best=00000000000000000003691b3262d512abab74536a35a23939eeec1969e63a3b height=799117 version=0x2000c000 log2_work=94.306007 tx=866155702 date='2023-07-17T19:42:50Z' progress=1.000000 cache=37.8MiB(214611txo)
2023-07-17T19:44:13Z [i2p] Error connecting to du5kydummi23bjfp6bd7owsvrijgt7zhvxmz5h5f5spcioeoetwq.b32.i2p:0: "STREAM STATUS RESULT=CANT_REACH_PEER"
```

Not sure how I go forward from here, probably will try to downgrade to version 1.8 somehow. If you have ideas I am happy to try it.

md-weber, 2024-01-11:
Not sure about the others, but after upgrading to the latest release everything works as expected. No more issues.

overcuriousity, 2024-02-21:
I also receive this error. I am on the current "dev" branch and could resolve the error temporarily by taking the above steps (cannot reproduce exactly), but after a reboot it comes back.

dinnerisserved, 2024-03-06:
> ## Solved it!
>
> `/mnt/hdd/bitcoin/settings.json` had this entry:
>
> ```
> {
>     "wallet": [
>         ""
>     ]
> }
> ```
>
> Removing this empty wallet list resulting in
>
> ```
> {
> }
> ```
>
> did the trick. After [...]

This worked for me, and bitcoind now works fine. The core issue here is however that there are expected defaults in place for the node to run. It should be able to run headless with no prerequisites of wallet files upon boot.

ytrezq, 2024-12-23:
> > Is it safe to actually delete the wallet.dat file @openoms ?
> 
> The default bitcoind wallet is not used to store private keys. Only Joinmarket/Jam uses it as watch only, but obviously it wouldn't wouldn't work if corrupted.

@openoms so where private keys are stored by default ?