raspiblitz/raspiblitz #3973: ext4 repair options

rootzoll, 2023-06-21:
Its quite common that people run into problems with there ext4 ssd filesystem. Under repair we shoudl at least offer the most basic repair and analysis options:

* search the issues & forums for cases and best practices (fsck.ext4)
* check if disk is full
* health of disk
* ...

velgard1, 2023-06-21:
It's just coincidence that it seems I actually have an issue related to this topic and wanted to open a new topic right now...

Have having a loop rebooting issue it was recommended to reindex - and this also work for a while, but then two issues occured at around 28% progress:
1) ext4 read error on swap device (at least something like that
2) complete freeze of v1.9.0

When I unplug the Blitz and restart it works again for a while - I'm not sure if the time it's running decreases right now.
The 1TB disk has around 650GB if I remember correctly, at least the message does also not indicate that it's full since it's a read issue...
Obvious for the freeze but also with the read error it's stuck and I'm not able to access the Bitz any more.

So what's mentioned is exactly what I want to know: How to check the SSD health in the first step?

Thanks for any helpful suggestions!

velgard1, 2023-08-01:
At first I did not realize that the initial comment was edited and then it took me some time to work on the issue...

Anyway, finally I managed it and at least for now it's working - what I did:
1) Download Knoppix 9.1 DVD version via knopper.net
2) Flash it to a 64GB USB Stick with balenaEtcher
3) Boot with that from an older Laptop (it is already UEFI and so there has been a message on first boot - but how to deal with that is mentioned on knopper.net, by some reason I had to boot 2 or 3 times until everything worked fine)
4) Managed to find the Raspiblitz SSD via the file manager and execute "sudo fsck <device>" - quite fast and seemed to have no errors
5) After some research I tried "sudo fsck -c <device>" to check for bad blocks (read test, took apx. 7h on an USB2 port) - there seemed to be errors (message "(13/0/0 errors)", but I was not able to figure out the meaning
6) Tried an read/write test on USB3 via "sudo e2fsck -fccky <device>" - this was much slower and seemed to have the same result as mentioned under 5) - I assumed that it would be 0/0/0 now
7) Booted Raspiblitz and it works as before - syncing slowly continuing

I will give an update if it keeps working or not - if someone knows the meaning of "(13/0/0 errors)" clarification is appreciated...

velgard1, 2023-10-23:
I bought a 2TB WD Green and the suggested enclosure but did not manage to copy the SSD in a way that worked with the Raspiblitz (via Knoppix on a Laptop).
When 1.10 was released I flashed the SDCard and to my surprise it worked. After a while it was stuck again and worked after flashing another time.
Managed to fully sync that way and it was running for a while - until today.

I'll open another issue and consider that topic closed for me regarding the SSD error - at least a proper description how to copy to another SSD would be nice for people that are not so experienced with Linux, thanks.