raspiblitz/raspiblitz #3974: which image for upgrades?

woeisme, 2023-06-24:
The documentation on the main page does not make it clear which sd card image is best for upgrades from previous versions. 

The new scheme (fat) (min) does not illustrate which type of image came before either.

adevoss, 2023-06-25:
You can choose which image you use. The important thing is the version number.

There is no upgrade in the way that that the sd-card in use gets (hot)fixes.
You should use another sd-card, flash a new version (fat or min) and replace the old sd-card.

Don't know about downgrading.

openoms, 2023-07-08:
This is correct as covered in the readme. Please detail if you need more clarification.

rootzoll, 2023-07-29:
You can upgrade with both images fat & min - but only the fat offers the WebUI for recovery/update. When in doubt use fatpack image.

closed 2023-07-29 by rootzoll