raspiblitz/raspiblitz #3983: add overcommit_memory to redis config

cstenglein, 2023-06-29:
From the redis logs under `/var/log/redis/redis-server.log`:

```
WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
```

See also: https://redis.io/docs/management/admin/

rootzoll, 2023-07-29:
_When overcommit_memory is set to 0 (the default), the kernel uses a heuristic algorithm to determine whether to allow a process to allocate more memory than is available. This can lead to situations where a process is allowed to allocate more memory than is physically available, which can cause the system to become unstable or crash.

When overcommit_memory is set to 1, the kernel will always allow a process to allocate memory up to the limit of its address space, regardless of how much physical memory is available. This can prevent the system from becoming unstable or crashing due to memory allocation issues, but it can also cause processes to fail if they try to allocate more memory than is available._

To set overcommit_memory=1 seems like the more stable option.

rootzoll, 2023-07-29:
@openoms can you recommend the changes in #4041

openoms, 2023-07-29:
Agree, redis failing is the less evil.

rootzoll, 2023-07-31:
PR merged - now on final testing

closed 2023-09-14 by rootzoll