raspiblitz/raspiblitz #3984: Changing what is seen on display

kahlenberg, 2023-06-30:
Hi,

Is it possible to use smaller LCD Screen and can I change what is displayed on the screen?

Thanks.

fusion44, 2023-07-02:
Hi,

currently there is no way of changing what is shown without changing the shell script. In future we are planning to make it more configurable what is shown. What information is most important to you?

semaf, 2023-07-03:
So there is no way to read the values blockheight, channels active, onion/adresse or capacity to write own python code?

md-weber, 2023-07-03:
In order to change what is visible on the display you will have to modify the `00infoBlitz.sh`

You will probably need some API call that you can make to receive the information that you would like to show.

fusion44, 2023-07-03:
Like @md-weber  mentioned, you can modify the `00infoBlitz.sh` script at around line 390:

https://github.com/raspiblitz/raspiblitz/blob/21d2593b901e1279f15b7ba6d131ef21336cf6e3/home.admin/00infoBlitz.sh#L320

With a bit of luck all information is already available in the script and you just have to adapt the representation string.

closed 2023-07-29 by rootzoll