raspiblitz/raspiblitz #3987: Investigate replacing Redis with Dragonfly DB

fusion44, 2023-07-02:
Redis consumes quite a bit of memory and CPU time which we might reduce by replacing it with Dragonfly DB. Dragonfly DB is a drop in replacement for Redis which claims to be less resource intensive than Redis. Normally there shouldn't be any code changes necessary except what binary we install for a DB.

https://github.com/dragonflydb/dragonfly/

I'm unsure about the license though: https://github.com/dragonflydb/dragonfly/blob/main/LICENSE.md
Sounds like we are fine as long as we don't offer a in memory DB as a service.

fusion44, 2023-07-02:
I've completely stopped Redis on my dev machine and could use the normal Redis CLI just fine when running Dragonfly as a replacement. Will do some performance tests on an actual RaspiBlitz later this evening.

fusion44, 2024-03-23:
Okay, this got a new urgency since Redis basically switched their license to a crippled "source available" license. 

Instead of Dragonfly, I'd switch to [KeyDB](https://docs.keydb.dev), which is a fork of Redis, and remains under the original BSD license. I don't think it's in the official Ubuntu repositories yet, but I'm sure it's just a matter of time. There is an official [PPA](https://docs.keydb.dev/docs/download/) we can use for now.

rootzoll, 2024-03-24:
The redis cache is a quite critical infrastructure and as a basic plan I agree that changing to KeyDB is a good plan because of the license. So v1.12.0 is a good horizon to aim for this - as a quick solution. 

But if possible I would like to include as less as external PPAs for the minimal build as possible. So if we consider to change the cache system - maybe we can even downsize a bit on our tech stack. Basically we just need a keyvalue-store - in the shell scripts this is wrapped by the `_cache.sh` script and could even be moved to a textfile on the RAM disk (just to describe how low tech this can go). I mark this as a `teamdiscuss` to see what our options are here.

frennkie, 2024-03-24:
> But if possible I would like to include as less as external PPAs for the minimal build as possible. 

I absolutely agree on this! :-D

> and could even be moved to a textfile on the RAM disk

I think one thing where I was considering something like Redis to be a better solution compared to a "textfile on RAM disk" was: How to deal with concurrent read and **write** access to the file(s). Is it a problem if a metric or settings is read (e.g. for the info screen) and an update is attempted? With a "proper backend" (e.g. like Redis) this is clearly defined.
Not a "no-go" when going with textfile on RAM disk - but something to consider/evaluate.