raspiblitz/raspiblitz #3990: CHANNELD_AWAITING_LOCKIN Problem low fee for closing unconfirmed funding tx channel.

smamamo, 2023-07-05:
I habe since 3 month a channel in "CHANNELD_AWAITING_LOCKIN" status
Geh funding tx is unconfirmed in block explorer 
Funding TXid: b832f3e2ab2eb572e1cbae5034ecfeb6a08e806c2ff15a4363f6bda94e684686
When I reinstall core lightning  I get a message : 
closingd-chan#1: Our ideal fee is 828sat (1500 sats/perkw), but our maximum is 700sat: using that  

I try to find a solution with lightning-cli help - I found 
- setchannel id [feebase] [feeppm] [htlcmin] [htlcmax] [enforcedelay] ?
anyone know the right syntax for this command
lightning-cli setchannel id  (channel id) ? ? ? ? ? ? ?

openoms, 2023-07-08:
This setchannel command is to set the routing fees, that's not what you need.

The tx you have linked is not found: https://mempool.space/tx/b832f3e2ab2eb572e1cbae5034ecfeb6a08e806c2ff15a4363f6bda94e684686

If you have only onchain funds best to restore from the seed only and wait for the rescan to finish to get your funds.

If you have channels do not reset the node, but spend all the onchain funds from the wallet (can simply send all of it to a new address in CLN with RTL) to make sure this outdated transaction doesn't get broadcasted.

smamamo, 2023-07-17:
After recover from seed I get loop 
DEBUG   gossipd: seeker: no peers, waiting
DEBUG   plugin-bcli: Feerate min_acceptable raised from 2569 perkb to floor of 3112
DEBUG   plugin-bcli: Feerate min_acceptable raised from 2569 perkb to floor of 3112
??No funds in RTL