raspiblitz/raspiblitz #3991: LNBits: no extension installation possible

cstenglein, 2023-07-06:
See https://github.com/cstenglein/raspiblitz-web/issues/631

openoms, 2023-07-08:
Copying over the manual solution for this:

>you need to be admin/superuser for that it seems:
https://docs.lnbits.org/guide/admin_ui.html
echo "LNBITS_ADMIN_UI=true" >> /home/lnbits/lnbits/.env
sudo systemctl restart lnbits
cat /home/lnbits/lnbits/.super_user
https://xxx:5001/wallet?usr=admin_id

Would this be the best to be an option in the LNbits menu?

rootzoll, 2023-08-01:
OK I adapted the LNbits install script.

- on install the adminUI will get activated
- the "bonus.lnbits.sh status" will give the needed admin path as "authMethod"
- the API pushes this thru to the WEBUI like this:

```
{
  "id": "lnbits",
  "version": "0.10.9",
  "installed": true,
  "status": "online",
  "address": "https://192.168.178.61:5001",
  "httpsForced": "1",
  "httpsSelfsigned": "1",
  "hiddenService": "dg33nu64yhb5y5esyo3mygs55tfiegpl3igumvh672qnucngghzbqsqd.onion",
  "authMethod": "/wallet?usr=7f5c24ca95da436082e8bc273c935ddd",
  "details": {},
  "error": ""
}
```

@cstenglein so what you need to do in the WEBUI would be to detect that a `authMethod` starts with `/` (signals its a path login) and the add the whole path to the URL (either IP or onion) after the port (is needed):

So with the example above the final URL would be:
`https://192.168.178.61:5001/wallet?usr=7f5c24ca95da436082e8bc273c935ddd`
thats the url that should be opened by the WebUI - it will be the admin page where the user can install also extensions.

rootzoll, 2023-08-01:
Its merged into dev @cstenglein if you want to try this out. LNbits have to be installed freshly for Adminpage to be activated.

cstenglein, 2023-08-02:
@rootzoll implemented it in https://github.com/raspiblitz/raspiblitz-web/pull/649 but I don't think it's a good solution, since `authMethod` is usually the method of authentication, not a URL.

rootzoll, 2023-08-03:
To make it prettier in the future we could introduce a a authData field but for now its tested and works with the new rc2.

closed 2023-08-03 by rootzoll