raspiblitz/raspiblitz #4003: status screen: tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

openoms, 2023-07-08:
on my CLN node running bitcoind v24.0.1 and then v25.0  after a node restart the debug.log file was not created.
Needed to restart bicoind once to solve this.

After having:
```
main.debuglogfile=/mnt/hdd/bitcoin/debug.log
```
in `/mnt/hdd/bitcoin/bitcoin.conf` the problem didn't resurface.

If more users have this problem the above line could be part of the default bitcoin.con and possibly inserted with a `bitcoind.check.sh` to existing setups.

nolykhemin, 2023-07-24:
I have this problem too but with
bitcoin 21.0.1
LND 0.16.2-beta 
will see if this resurfaces again.

pabo3000, 2023-07-31:
Here the same

bitcoin 24.0.1
LND 0.16.2-beta

rootzoll, 2023-07-31:
I started todo multiple changes with the following PR:

- created a bitcoin.check.sh that will run as a prestart in bitcoid systemd services
- removed the bitcoin.services from assests - will now all created by bitcoin.install.sh
- removed the debuglog path from service file (should now only be done by config)
- made sure that the bitcoin.check.sh as a prestart will always add the correct debug log path

rootzoll, 2023-07-31:
OK would recommend merging PR #4051 for next release cadidate and test also with a fresh setup BTC node and a recovered one if the new service setup works.

tulkasBTC, 2023-08-17:
GM or GN FRENS,

Experiencing similar problems here with a v1.9.0 ProxMox install:

Fresh install, just trying to create a more redundant setup. After copying the timechain from another blitz on the same LAN, I keep getting errors acquiring logs for bitcoind and lnd.

After checking here and #3970, added `main.debuglogfile=/mnt/hdd/bitcoin/debug.log` and `main.walletdir=/mnt/hdd/bitcoin` to the bitcoin.conf file... Got it partially working.

I can not install mempool.space or btc-explorer, it renders me a FAIL message asking to reboot and try to install it again manually.

```
#FAIL on Mempool Explorer install # 

Try manual install on terminal after reboot with:

/home/admin/config.scripts/bonus.mempool.sh on
```

Sadly bitcoind and lnd are not surving reboots.

debug -l: https://l.termbin.com/7day

₿ /home/admin/config.scripts/bonus.mempool.sh on
# making sure services are not running
# *** Activate MEMPOOL ***
# mempool already installed.
# OK - the mempool.service is enabled, system is on ready so starting service
result='mempool service not active'

Help me, @cercatrova21?

rootzoll, 2023-08-25:
This is to be tested wioth the next v1.10.0rc4

closed 2023-09-14 by rootzoll