raspiblitz/raspiblitz #4008: Check for RPi CPU throttling

openoms, 2023-07-09:
The summer has come and my RPi-s are running hot.
Checking for throttling with this script works:
```
#!/usr/bin/env python3

import subprocess

GET_THROTTLED_CMD = 'vcgencmd get_throttled'
MESSAGES = {
    0: 'Under-voltage!',
    1: 'ARM frequency capped!',
    2: 'Currently throttled!',
    3: 'Soft temperature limit active',
    16: 'Under-voltage has occurred since last reboot.',
    17: 'Throttling has occurred since last reboot.',
    18: 'ARM frequency capped has occurred since last reboot.',
    19: 'Soft temperature limit has occurred'
}

print("Checking for throttling issues since last reboot...")

throttled_output = subprocess.check_output(GET_THROTTLED_CMD, shell=True)
throttled_output = throttled_output.decode('utf-8')  # decode bytes to string
throttled_binary = bin(int(throttled_output.split('=')[1], 0))

warnings = 0
for position, message in MESSAGES.items():  # use items() instead of iteritems()
    # Check for the binary digits to be "on" for each warning message
    if len(throttled_binary) > position and throttled_binary[0 - position - 1] == '1':
        print(message)
        warnings += 1

if warnings == 0:
    print("Looking good!")
else:
    print("Houston, we may have a problem!")
```

Raw command:
```
₿ sudo vcgencmd get_throttled       
throttled=0x80000
```
Interpretation (temp is at 77 -78 C):
```
₿ sudo python throttle.py
Checking for throttling issues since last reboot...
Soft temperature limit has occurred
Houston, we may have a problem!
```

 Could make this a warning on the display.
 
 Source (converted for python 3): https://blog.mccormack.tech/shell/2019/01/05/monitoring-raspberry-pi-power-and-thermal-issues.html

rootzoll, 2023-07-29:
add later to debug / system state info

rootzoll, 2023-11-14:
related #1564