raspiblitz/raspiblitz #4009: Lightning sync stuck at 99.99%

BTC1M, 2023-07-10:
I am using Raspiblitz 1.7.1
I finished syncing the blockchain 2 days ago. 
However, lightning progress is still stuck at 99.99%.
I already tried to restart several times.

[image]

[image]

TrezorHannes, 2023-07-11:
Post the output link of `debug -l` here please

BTC1M, 2023-07-12:
[image]

Here are the logs on the link https://termbin.com/vuvr : 

```
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.1
chainnetwork: bitcoin / main
 12:14:37 up 9 min,  2 users,  load average: 3.75, 3.59, 2.01

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin-Daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2023-07-12 12:06:05 BST; 8min ago
  Process: 2841 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 2846 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
 Main PID: 2903 (bitcoind)
    Tasks: 18 (limit: 4532)
   CGroup: /system.slice/bitcoind.service
           └─2903 /usr/local/bin/bitcoind -daemon -conf=/mnt/hdd/bitcoin/bitcoin.conf -pid=/mnt/hdd/bitcoin/bitcoind.pid -debuglogfile=/mnt/hdd/bitcoin/debug.log

Jul 12 12:06:04 raspberrypi systemd[1]: Starting Bitcoin-Daemon...
Jul 12 12:06:05 raspberrypi systemd[1]: Started Bitcoin-Daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Tue 2023-07-11 00:17:01 BST, end at Wed 2023-07-12 12:14:37 BST. --
Jul 12 12:06:04 raspberrypi systemd[1]: Starting Bitcoin-Daemon...
Jul 12 12:06:05 raspberrypi systemd[1]: Started Bitcoin-Daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2023-07-12T11:13:19Z UpdateTip: new best=000000000000000000036bf8e6438b588614ea43319951c0c2187deee453c9f0 height=798231 version=0x2cab6000 log2_work=94.293956 tx=863221464 date='2023-07-11T07:02:52Z' progress=0.999715 cache=69.6MiB(460036txo)
2023-07-12T11:13:23Z UpdateTip: new best=00000000000000000001473b69583937c24670ffaafba5bb5092e6f54332d400 height=798232 version=0x20f5e000 log2_work=94.293969 tx=863224329 date='2023-07-11T07:08:40Z' progress=0.999716 cache=70.9MiB(469939txo)
2023-07-12T11:13:26Z UpdateTip: new best=000000000000000000019e5e80c9e72b4e63762e161e2142e922812182c67ea7 height=798233 version=0x20400000 log2_work=94.293981 tx=863228244 date='2023-07-11T07:16:17Z' progress=0.999717 cache=72.2MiB(478462txo)
2023-07-12T11:13:28Z UpdateTip: new best=00000000000000000003a4b11a9484cc117f5130d7c2688af13a312405b226e3 height=798234 version=0x201bc000 log2_work=94.293994 tx=863233603 date='2023-07-11T07:21:06Z' progress=0.999718 cache=72.8MiB(483135txo)
2023-07-12T11:13:29Z UpdateTip: new best=00000000000000000001d790c2e4e7138f05e3176e4bc6bfe1531258b5c7a8c8 height=798235 version=0x20002000 log2_work=94.294007 tx=863238059 date='2023-07-11T07:25:38Z' progress=0.999719 cache=73.8MiB(490648txo)
2023-07-12T11:13:30Z UpdateTip: new best=00000000000000000001b6fcd7f2943d9f5a228711ecec67beb6bbc2b260c6e9 height=798236 version=0x20200000 log2_work=94.294020 tx=863242661 date='2023-07-11T07:27:17Z' progress=0.999719 cache=75.1MiB(498466txo)
2023-07-12T11:13:36Z UpdateTip: new best=000000000000000000019c008d31c4ea981124893ae4289108292b1dbbd12b99 height=798237 version=0x3388c000 log2_work=94.294033 tx=863244712 date='2023-07-11T07:40:19Z' progress=0.999721 cache=76.3MiB(508034txo)
2023-07-12T11:13:40Z UpdateTip: new best=00000000000000000004e0eb14aa8c089835feae006efd6e1c5834aa5b8991dd height=798238 version=0x20200000 log2_work=94.294046 tx=863248173 date='2023-07-11T07:46:37Z' progress=0.999722 cache=77.3MiB(514909txo)
2023-07-12T11:13:44Z UpdateTip: new best=00000000000000000002e415b69fd8806be2e057dfba0271e17509750b521a8a height=798239 version=0x2000c000 log2_work=94.294059 tx=863250511 date='2023-07-11T08:06:07Z' progress=0.999726 cache=78.3MiB(523147txo)
2023-07-12T11:13:48Z UpdateTip: new best=0000000000000000000228ccc055d379ff87b7b1c80fb5bd2145e04ae0a7be66 height=798240 version=0x2849e000 log2_work=94.294072 tx=863251432 date='2023-07-11T08:08:08Z' progress=0.999726 cache=78.9MiB(527729txo)
2023-07-12T11:13:54Z UpdateTip: new best=000000000000000000022fbfbcc15a13aaa656404b24d38f0055995cefe96e04 height=798241 version=0x3fffe000 log2_work=94.294085 tx=863254535 date='2023-07-11T08:22:57Z' progress=0.999729 cache=79.9MiB(535117txo)
2023-07-12T11:13:56Z UpdateTip: new best=0000000000000000000419494edeff39ff3b9cc9122aaf23ee4ddeddcc83102d height=798242 version=0x31b28000 log2_work=94.294098 tx=863257254 date='2023-07-11T08:28:55Z' progress=0.999730 cache=80.8MiB(541202txo)
2023-07-12T11:14:00Z UpdateTip: new best=00000000000000000001a175d382a8110893763ccd465f3075e39011f419dc28 height=798243 version=0x20600000 log2_work=94.294111 tx=863260580 date='2023-07-11T08:52:43Z' progress=0.999734 cache=82.1MiB(551550txo)
2023-07-12T11:14:05Z UpdateTip: new best=00000000000000000002b0013375b6369514d0d88bf2ce447a5f223a9c6af1aa height=798244 version=0x20e00000 log2_work=94.294124 tx=863263520 date='2023-07-11T09:07:27Z' progress=0.999736 cache=83.1MiB(560057txo)
[... cut ...]
initialSync=1
syncProgress=99.97 
startcountLightning=2
lndActive=1
walletLocked=0
lndPeers=0
syncedToChain=0
scanTimestamp=1689070321
scanDate='Tue 11 Jul 11:12:01 BST 2023'
scanProgress=99.98
lndRPCReady=1
clActive=0
CLwalletLocked=0
blitzTUIActive=0
blitzTUIRestarts=0
vagrant=0
scriptRuntime=4

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
debug | torsocks nc termbin.com 9999
```

TrezorHannes, 2023-07-12:
It might be that you need to update your bitcoin version. This one is the culprit:

```
Jul 12 12:06:37 raspberrypi lnd[3077]: unable to create chain control: -28: Verifying blocks…
```

What does `bitcoind -version` and `lncli --version` say?

NilByte, 2023-07-12:
Most likely it's the outdated LND version causing trouble. There was a bug a while back stopping the sync.

Guide for updating LND to latest v0.16.4-beta:

https://github.com/openoms/lightning-node-management/blob/en/hardware/raspiblitz/lnd.updates.md

Or use the UPDATE menu in terminal:

LND      Interim LND Update Options

BTC1M, 2023-07-12:
It say : 
`Bitcoin Core version v22.0.0`
`lncli version 0.13.3-beta commit=v0.13.3-beta`

I tried to update using the guide on the link, but i get the following error :

[image]

So instead i updated LND through the UPDATE menu.
Now the output of `lncli--version` is : 
`lncli version 0.16.4-beta commit=v0.16.4-beta`

But it looks like it is still stuck at 99.99%

The output of `debug -l` is : 
```
*** LAST LND (MAINNET) ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Wed 2023-07-12 18:19:41 BST, end at Wed 2023-07-12 18:24:48 BST. --
Jul 12 18:20:40 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Jul 12 18:20:40 raspberrypi systemd[1]: Started LND Lightning Daemon.
1689182542 ERROR: [lncli] rpc error: code = Unknown desc = the RPC server is in the process of starting up, but not yet ready to accept calls
```

TrezorHannes, 2023-07-12:
It might take a while, the RPC error pops up while LND is loading. 
What does `sudo systemctl status lnd` say?

I'd also consider updating raspiblitz at some point, cause a number of things have improved, and your `lnd.conf` might be missing some additional, useful configurations which were added by LND, and henceforth included in later raspiblitz versions.

Not pressing, let's see if LND is booting again, but I'd check out to get another SD and prepare a flashing of the newest blitz-version.

BTC1M, 2023-07-12:
> What does sudo systemctl status lnd say?

[image]

> It might take a while, the RPC error pops up while LND is loading. I'd also consider updating raspiblitz at some point, cause a number of things have improved, and your lnd.conf might be missing some additional, useful configurations which were added by LND, and henceforth included in later raspiblitz versions.

Ok i'll let syncing during one whole night. If after this night it is still stuck at 99.99%, i will update raspiblitz. Which one should i update from the menu, RELEASE or PATCH ?

> Not pressing, let's see if LND is booting again, but I'd check out to get another SD and prepare a flashing of the newest blitz-version.

I will need some guidance for that. Does it mean i will have to buy another SD card, and re-sync from the beginning ? This option is difficult to envisage for me, because it would take me a very long time as i can't sync during the day, because when my raspiblitz is on, it makes my connection very slow and almost unusable. Is it normal by the way ?

TrezorHannes, 2023-07-13:
It's running, so you should be fine.

For the raspiblitz update, don't confuse SD and SSD. You only need to flash
a second SD with the newest release.
All the Blockchain and lightning data is on the SSD, you don't need to
start from scratch

BTC1M, 2023-07-23:
Please

TrezorHannes, 2023-08-01:
I don't think updating raspiblitz will necessarily fix the issue you're having. The most delicate system modules underneath are LND and Bitcoind. You can update both first, and then identify the issue why it's stuck. And only once it's fixed, I'd recommend to update, because otherwise you risk to make things worse.

So first off, please provide what versions you're running
  - `lncli --version`
  - `bitcoin-cli -version` 

If LND is < lower than 16.3, and bitcoin is < 24.0, I would recommend to update (@NilByte provided [update infos here](https://github.com/raspiblitz/raspiblitz/issues/4009#issuecomment-1632700921).

Lastly, if it's stuck, please provide the termbin-link you'll get after running `debug -l` here, so we can look into the relevant logs.

openoms, 2024-08-10:
closing as outdated

closed 2024-08-10 by openoms