raspiblitz/raspiblitz #4011: check if mempool STDOUT_LOG_MIN_PRIORITY is set

openoms, 2023-07-12:
found a large syslog and daemon.log:
[image]

Looks like it is this setting not applied for mempool:
```
    "STDOUT_LOG_MIN_PRIORITY": "info"
 ```
https://github.com/raspiblitz/raspiblitz/pull/3426/files

shouldcheck and correct with a script option in systemd.

rootzoll, 2023-07-29:
Following PR should fix that log can grow that big:
https://github.com/raspiblitz/raspiblitz/pull/4038

rootzoll, 2023-07-29:
@openoms do you also want to fix that mempool logging level?

openoms, 2023-07-29:
on a recovery it should be applied.
I think here a dirty update was the issue.
Can be closed.

closed 2023-07-29 by openoms