raspiblitz/raspiblitz #4012: Apps tab won't load on WebUI (#3608)

BlueberryGH, 2023-07-13:
I am having the same issue with Apps not showing on the Apps tab.
Running on v 1.9.0
Would appreciate any idea how to fix this other than waiting for the next update to be installed.

Many thanks!

rootzoll, 2023-07-29:
Is this still happening?

BlueberryGH, 2023-07-29:
Yes, still happening.

rootzoll, 2023-07-29:
Can you go to "Settings" in the WebUI and "Generate Debug Report" & paste here?

cstenglein, 2023-07-29:
@fusion44 could it be a problem with the API?
Usually the webui waits for the installed app message, which doesn't seem to arrive here.

fusion44, 2023-07-30:
It is possible, will check on my node later today.

BlueberryGH, 2023-07-31:
Here comes the debug report:

[stdout]

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.9.0
chainnetwork: bitcoin / main
 14:19:20 up 18 days,  3:18,  1 user,  load average: 4.21, 3.62, 3.35

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-07-13 11:03:18 BST; 2 weeks 4 days ago
    Process: 6111 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 6137 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 6142 ExecStart=/usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 6205 (bitcoind)
      Tasks: 18 (limit: 9352)
        CPU: 1w 3d 17h 46min 53.536s
     CGroup: /system.slice/bitcoind.service
             └─6205 /usr/local/bin/bitcoind -daemonwait -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

Warning: journal has been rotated since unit was started, output may be incomplete.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n20
-- Journal begins at Sun 2023-07-30 07:28:54 BST, ends at Mon 2023-07-31 14:19:20 BST. --
-- No entries --

*** LAST BLOCKCHAIN (MAINNET) INFO LOGS ***
sudo tail -n 50 /mnt/hdd/bitcoin/debug.log
2023-07-31T11:28:20Z UpdateTip: new best=000000000000000000038cb8836602604dcadf838df4d0e2eac6d1a6361227a3 height=801062 version=0x28796000 log2_work=94.332020 tx=872507067 date='2023-07-31T11:27:53Z' progress=1.000000 cache=30.5MiB(151146txo)
2023-07-31T11:28:20Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T11:28:20Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T11:28:20Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T11:28:20Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T11:40:08Z UpdateTip: new best=00000000000000000002015a2714b19cccf1bf1cdafd18aa2b1a43c9b9ae2144 height=801063 version=0x22000000 log2_work=94.332033 tx=872510809 date='2023-07-31T11:39:51Z' progress=1.000000 cache=32.3MiB(164375txo)
2023-07-31T11:41:49Z UpdateTip: new best=00000000000000000002dc8e2ccb02e847b5589897a0e95afcb074e2ddc2263a height=801064 version=0x20006000 log2_work=94.332046 tx=872515388 date='2023-07-31T11:41:38Z' progress=1.000000 cache=33.7MiB(174083txo)
2023-07-31T11:41:49Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T11:44:43Z UpdateTip: new best=00000000000000000003767af0e795e3d7b0d1a89704bab49f8fdcbc03d214e7 height=801065 version=0x20342000 log2_work=94.332059 tx=872521494 date='2023-07-31T11:44:09Z' progress=1.000000 cache=34.5MiB(181735txo)
2023-07-31T11:44:51Z New outbound peer connected: version: 70016, blocks=801065, peer=21987 (outbound-full-relay)
2023-07-31T11:50:11Z Socks5() connect to xzwbohlaf4wazh2rjbd3rc5lmwgp5lxn67dpppbqq5hcanqvympvw2yd.onion:8333 failed: host unreachable
2023-07-31T11:53:45Z New outbound peer connected: version: 70016, blocks=801065, peer=21992 (block-relay-only)
2023-07-31T11:58:28Z Socks5() connect to m6neeyo23eqim6zk4orda54bl7vw7umpscwceg54llmzk5t2wdt3ebyd.onion:8333 failed: host unreachable
2023-07-31T11:59:16Z Socks5() connect to bt3gwklouhhsxmby6m5e7sd5yb47zst6feklics6ancigcfocwrrbqad.onion:8333 failed: host unreachable
2023-07-31T12:03:04Z UpdateTip: new best=00000000000000000002375bdaff6d517ef40275ba9a2b212bceba9b76d242f0 height=801066 version=0x20200004 log2_work=94.332072 tx=872524314 date='2023-07-31T12:02:50Z' progress=1.000000 cache=37.7MiB(206779txo)
2023-07-31T12:03:04Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T12:03:04Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T12:03:04Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T12:03:04Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T12:17:17Z UpdateTip: new best=000000000000000000041394182f95f4eaf0e61196d203da50615e1dda27e165 height=801067 version=0x2001c000 log2_work=94.332085 tx=872526871 date='2023-07-31T12:16:39Z' progress=1.000000 cache=39.8MiB(223707txo)
2023-07-31T12:17:17Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T12:17:17Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T12:17:17Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T12:17:17Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2023-07-31T12:17:59Z UpdateTip: new best=0000000000000000000477aae925a24f795fa63b4c95dc6c528b28f57a4b80c7 height=801068 version=0x299ce000 log2_work=94.332098 tx=872531679 date='2023-07-31T12:17:45Z' progress=1.000000 cache=40.4MiB(227904txo)
[... cut ...]
btc_default_sync_initialblockdownload="0"
btc_default_mempool_transactions="126259"
btc_default_sync_percentage="100.00"
btc_default_blocks_headers="801073"
btc_default_peers="21"

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

fusion44, 2023-07-31:
It works on my node. I can't see anything suspicious in the logs above either. Maybe the API hangs when it tries to get install info from REDIS.

scubafly, 2023-08-01:
I have the same issue on 1.9.0
Also I'm not seeing the installed apps on the left side.
Under settings the version number is missing.
And on home there is a spinner in the block left to the "Hardware Details"

The last thing I did was install BTCpay.
I will disable BTCpay trough ssh when I'm home and let you know if this fixes it for me.
Logs are here: https://pastebin.com/QnkT2Xpe

cstenglein, 2023-08-01:
@fusion44 I got a similar problem on my testnode as well. 

The only messages I get are `system_startup_info`, `ping` and `btc_info`.

Node & blitz api running without issues. Load is at `3.01` so not super busy.

scubafly, 2023-08-02:
I removed BTCPay, shutdown, reconnected al the wires. Now It looks like it is stable. uptime: up 11:38,  3 users,  load average: 2.01, 2.02, 1.99. Will update if I see this issue again. If not, it will be resolved for me.

BlueberryGH, 2023-08-02:
Some additional info:

The Apps menu worked just fine before I have started the Electrum server on the Raspi.

The Activation process did not run through properly, so that I have powered off and restarted the Raspi after several hours of waiting.

Perhaps an open file was damaged at this point?

 

Is there anything I can do (re-install, update,…)?

Would appreciate any help.

cstenglein, 2023-10-10:
@BlueberryGH is this still an issue with 1.10?

scubafly, 2023-10-11:
For mee the 1.10 version looks to be more stable. I still have to try if it keeps stable if I instal the BTCPay server.

Radiokot, 2023-11-16:
I also have this problem, albeit electrs is still indexing.

rwernerar, 2023-12-05:
I'm having the same problem with the latest software version. The Apps tab keeps loading and never finishes! When I initially installed the software, it was working, so it must be some setting that I changed later. I will probably start a new, clean installation and make changes step by step to identify where the problem is coming from!

helishmopter, 2024-02-14:
Hi, having the same issue, apps tab keeps loading with spinning yellow circle. Fresh install, NO lightning, running BTC only on version 1.10. Blockchain is 100 % synced. I installed/indexed electrs from the command line and it seems to be working, but it's not available in webUI (spinning circle).

helishmopter, 2024-02-16:
Having the same issue as above in 1.10. Apps tab won't stop loading in webUI. No LN installed, just BTC. Blockchain 100 % synced. No other apps running. Tried re-installing 1.10. with same result (app tab not loading). Also attempted re-activating 'Blitz API + WebUI' in Node Settings & Options, with the result of being completely locked out of the webUI (502 Bad Gateway). Any help is much appreciated!

cstenglein, 2024-02-17:
@helishmopter @Radiokot @rwernerar Can you check if the blitzapi is running?

Log in via SSH and type `sudo service blitzapi status`

If it displays something like `EXITED` like this:

[image]

Then please post the output of the following command here:

`sudo journalctl -fu blitzapi`

@fusion44 On installation I get the following error:

```
Feb 17 09:23:16 raspberrypi blitz.web.api.sh[80768]: # '.env' config updates - blitzapi maybe needs to be restarted
Feb 17 09:23:16 raspberrypi systemd[1]: Started BlitzBackendAPI.
Feb 17 09:23:16 raspberrypi python[80845]: /home/blitzapi/blitz_api/venv/bin/python: No module named uvicorn
```

helishmopter, 2024-02-17:
@cstenglein  I get the following:
[image]

sudo journalctl -fu blitzapi

Feb 16 07:23:46 raspberrypi python[760]: 2024-02-16 07:23:46 | ⚠️ | utils.py:69 | Key 'hdd_capacity_bytes' not found in Redis DB.
Feb 16 07:23:46 raspberrypi python[760]: 2024-02-16 07:23:46 | ⚠️ | utils.py:69 | Key 'hdd_free_bytes' not found in Redis DB.
Feb 16 07:23:48 raspberrypi python[760]: 2024-02-16 07:23:48 | ⚠️ | utils.py:69 | Key 'hdd_capacity_bytes' not found in Redis DB.
Feb 16 07:23:48 raspberrypi python[760]: 2024-02-16 07:23:48 | ⚠️ | utils.py:69 | Key 'hdd_free_bytes' not found in Redis DB.
Feb 16 07:24:27 raspberrypi python[760]: 2024-02-16 07:24:27 | ℹ️ | service.py:46 | Bitcoin Core initializing, waiting 10 seconds...
Feb 16 07:24:37 raspberrypi python[760]: 2024-02-16 07:24:37 | ℹ️ | service.py:46 | Bitcoin Core initializing, waiting 10 seconds...
Feb 16 07:24:47 raspberrypi python[760]: 2024-02-16 07:24:47 | ℹ️ | service.py:46 | Bitcoin Core initializing, waiting 10 seconds...
Feb 16 07:24:57 raspberrypi python[760]: 2024-02-16 07:24:57 | ℹ️ | service.py:46 | Bitcoin Core initializing, waiting 10 seconds...
Feb 16 07:25:07 raspberrypi python[760]: 2024-02-16 07:25:07 | ℹ️ | service.py:46 | Bitcoin Core initializing, waiting 10 seconds...
Feb 16 07:25:17 raspberrypi python[760]: 2024-02-16 07:25:17 | ✔️ | service.py:39 | Bitcoin repository initialized

fusion44, 2024-02-17:
> @helishmopter @Radiokot @rwernerar Can you check if the blitzapi is running?
> 
> Log in via SSH and type `sudo service blitzapi status`
> 
> If it displays something like `EXITED` like this:
> 
> [image]
> 
> Then please post the output of the following command here:
> 
> `sudo journalctl -fu blitzapi`
> 
> Thank you!
> 
> ----
> 
> @fusion44 On installation I get the following error:
> 
> ```
> Feb 17 09:23:16 raspberrypi blitz.web.api.sh[80768]: # '.env' config updates - blitzapi maybe needs to be restarted
> Feb 17 09:23:16 raspberrypi systemd[1]: Started BlitzBackendAPI.
> Feb 17 09:23:16 raspberrypi python[80845]: /home/blitzapi/blitz_api/venv/bin/python: No module named uvicorn
> ```

Seems like it's not an API error. Something went wrong during install. The unicorn package was not installed properly. Did you try a second time?

ray-bun, 2024-02-28:
Same issue here, try uninstall and reinstall web interface

rootzoll, 2024-03-19:
So far this looks good on my testmachines on RC4. Closing this issue for release. Please comment/reopen when this error happens with v1.11.0rc4 or later.

closed 2024-03-19 by rootzoll