raspiblitz/raspiblitz #4013: electrs in restart loop after upgrade to 1.9

david-bakin, 2023-07-13:
Upgraded from 1.7.1 (or maybe 1.7.2) to 1.9 after being offline for 9mo.  Bitcoind sync'd fine (took 3.5days) but electrs is in a restart loop.  Used systemctl to stop/start it, no difference.  Rebooted, no difference.  Uninstalled/reinstalled electrs, no difference.  (Used to work for me on 1.7.x.) 4Gb Pi4.

Debug log [at termbin](https://l.termbin.com/mdzeo).  Do you need anything else?

See also another issue [TBD] that is also happening for me post-upgrade.

(BTW, electrs sure takes a heck of a long time to build, doesn't it?  You might think it was some kind of massive C++ program ...)

openoms, 2023-07-25:
Any update on this?
Looks like bitcoind was still syncing.

david-bakin, 2023-07-25:
Update is: No, bitcoind had sync'd fine and in fact was going great, for hours.  As seen, for example, via mempool visualizer on the box.  It updated as blocks came in, in sync with https://mempool.space.  No problems.  Also, lnd sync'd to 100% although that took a very very long time.   Only problem is electrs which no matter what, no matter how many times I stopped/started it via systemctl or rebooted never did anything but its reboot loop. I even completely uninstalled it and reinstalled it (via the menu) _twice_: no change.

I have no further updates. (I uninstalled it and have left it uninstalled.)  Will try it again at 1.9.1.

(Unless there's some other log you'd like to see.  I could reinstall it and pull the log, if you told me which one.)