raspiblitz/raspiblitz #4023: 1.9.0 syslog and daemon.log consuming all free space

lewildbeast, 2023-07-20:
Two Files syslog.1 (18.1GiB) and daemon.log.1 (18.1GiB) grow at a prolific rate (that file size is after 2 days) consuming all space on /dev/root which is mounted on /

This eventually results in Raspiblitz getting stuck at the screen "Blockchain Progress/Lightning Progress" just before the main menu (via SSH) and the status display page on the built in screen.

Running Bitcoin(Mainnet)/cLN/Mempool/BTC Explorer/RTL/Electrum Rust Server/TOR
Upgraded from at least 3 previous versions, not fresh install.

Output sample of these log files follow below.

############
############
############
############

daemon.log.1 
Jul 20 08:44:34 raspberrypi python[805]:     │      └ <function Handle._run at 0x7fb7846a60>
Jul 20 08:44:34 raspberrypi python[805]:     └ <Handle <TaskWakeupMethWrapper object at 0x7fb4273f40>(<Future finis... ...}, ...]}}>)>
Jul 20 08:44:34 raspberrypi python[805]:   File "/usr/lib/python3.9/asyncio/events.py", line 80, in _run
Jul 20 08:44:34 raspberrypi python[805]:     self._context.run(self._callback, *self._args)
Jul 20 08:44:34 raspberrypi python[805]:     │    │            │    │           │    └ <member '_args' of 'Handle' objects>
Jul 20 08:44:34 raspberrypi python[805]:     │    │            │    │           └ <Handle <TaskWakeupMethWrapper object at 0x7fb4273f40>(<Future finis... ...}, ...]}}>)>
Jul 20 08:44:34 raspberrypi python[805]:     │    │            │    └ <member '_callback' of 'Handle' objects>
Jul 20 08:44:34 raspberrypi python[805]:     │    │            └ <Handle <TaskWakeupMethWrapper object at 0x7fb4273f40>(<Future finis... ...}, ...]}}>)>
Jul 20 08:44:34 raspberrypi python[805]:     │    └ <member '_context' of 'Handle' objects>
Jul 20 08:44:34 raspberrypi python[805]:     └ <Handle <TaskWakeupMethWrapper object at 0x7fb4273f40>(<Future finis... ...}, ...]}}>)>
Jul 20 08:44:34 raspberrypi python[805]:   File "#033[32m./app/lightning/#033[0m#033[32m#033[1mservice.py#033[0m", line #033[33m236#033[0m, in #033[35m_handle_invoice_listener#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[35m#033[1masync#033[0m #033[35m#033[1mfor#033[0m #033[1mi#033[0m #033[35m#033[1min#033[0m #033[1mln#033[0m#033[35m#033[1m.#033[0m#033[1mlisten_invoices#033[0m#033[1m(#033[0m#033[1m)#033[0m#033[1m:#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m               │  └ #033[0m#033[36m#033[1m<function LnNodeCLNjRPCBlitz.listen_invoices at 0x7fb50401f0>#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m               └ #033[0m#033[36m#033[1m<app.lightning.impl.specializations.cln_jrpc_blitz.LnNodeCLNjRPCBlitz object at 0x7fb5becdf0>#033[0m
Jul 20 08:44:34 raspberrypi python[805]:   File "#033[32m./app/lightning/impl/specializations/#033[0m#033[32m#033[1mcln_jrpc_blitz.py#033[0m", line #033[33m175#033[0m, in #033[35mlisten_invoices#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[35m#033[1masync#033[0m #033[35m#033[1mfor#033[0m #033[1mi#033[0m #033[35m#033[1min#033[0m #033[1msuper#033[0m#033[1m(#033[0m#033[1m)#033[0m#033[35m#033[1m.#033[0m#033[1mlisten_invoices#033[0m#033[1m(#033[0m#033[1m)#033[0m#033[1m:#033[0m
Jul 20 08:44:34 raspberrypi python[805]:   File "#033[32m./app/lightning/impl/#033[0m#033[32m#033[1mcln_jrpc.py#033[0m", line #033[33m666#033[0m, in #033[35mlisten_invoices#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[35m#033[1mawait#033[0m #033[1mself#033[0m#033[35m#033[1m.#033[0m#033[1m_refresh_invoice_sub#033[0m#033[1m(#033[0m#033[36m#033[1mTrue#033[0m#033[1m)#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m      │    └ #033[0m#033[36m#033[1m<function LnNodeCLNjRPC._refresh_invoice_sub at 0x7fb503e280>#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m      └ #033[0m#033[36m#033[1m<app.lightning.impl.specializations.cln_jrpc_blitz.LnNodeCLNjRPCBlitz object at 0x7fb5becdf0>#033[0m
Jul 20 08:44:34 raspberrypi python[805]:   File "#033[32m./app/lightning/impl/#033[0m#033[32m#033[1mcln_jrpc.py#033[0m", line #033[33m946#033[0m, in #033[35m_refresh_invoice_sub#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[1minvoices#033[0m #033[35m#033[1m=#033[0m #033[35m#033[1mawait#033[0m #033[1mself#033[0m#033[35m#033[1m.#033[0m#033[1mlist_invoices#033[0m#033[1m(#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m                 │    └ #033[0m#033[36m#033[1m<function LnNodeCLNjRPCBlitz.list_invoices at 0x7fb503e8b0>#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m                 └ #033[0m#033[36m#033[1m<app.lightning.impl.specializations.cln_jrpc_blitz.LnNodeCLNjRPCBlitz object at 0x7fb5becdf0>#033[0m
Jul 20 08:44:34 raspberrypi python[805]: > File "#033[32m./app/lightning/impl/specializations/#033[0m#033[32m#033[1mcln_jrpc_blitz.py#033[0m", line #033[33m103#033[0m, in #033[35mlist_invoices#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[35m#033[1mreturn#033[0m #033[35m#033[1mawait#033[0m #033[1msuper#033[0m#033[1m(#033[0m#033[1m)#033[0m#033[35m#033[1m.#033[0m#033[1mlist_invoices#033[0m#033[1m(#033[0m
Jul 20 08:44:34 raspberrypi python[805]:   File "#033[32m./app/lightning/impl/#033[0m#033[32m#033[1mcln_jrpc.py#033[0m", line #033[33m301#033[0m, in #033[35mlist_invoices#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[1minvoices#033[0m#033[35m#033[1m.#033[0m#033[1mappend#033[0m#033[1m(#033[0m#033[1mInvoice#033[0m#033[35m#033[1m.#033[0m#033[1mfrom_cln_json#033[0m#033[1m(#033[0m#033[1mi#033[0m#033[1m)#033[0m#033[1m)#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m│        │      │       │             └ #033[0m#033[36m#033[1m{'label': 'keysend-1639959922.637279029', 'bolt11': 'lnbc1psmlntjpp598l5j8flda48tnfsve4qthn2tzuqykdczzfl5rwkka66kl5ffs3sdz82d...#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m│        │      │       └ #033[0m#033[36m#033[1m<classmethod object at 0x7fb5b4ba00>#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m│        │      └ #033[0m#033[36m#033[1m<class 'app.lightning.models.Invoice'>#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m│        └ #033[0m#033[36m#033[1m<method 'append' of 'list' objects>#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m└ #033[0m#033[36m#033[1m[]#033[0m
Jul 20 08:44:34 raspberrypi python[805]:   File "#033[32m./app/lightning/#033[0m#033[32m#033[1mmodels.py#033[0m", line #033[33m651#033[0m, in #033[35mfrom_cln_json#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[1mamt#033[0m #033[35m#033[1m=#033[0m #033[1mparse_cln_msat#033[0m#033[1m(#033[0m#033[1mi#033[0m#033[1m[#033[0m#033[36m"amount_msat"#033[0m#033[1m]#033[0m#033[1m)#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m      │              └ #033[0m#033[36m#033[1m{'label': 'keysend-1639959922.637279029', 'bolt11': 'lnbc1psmlntjpp598l5j8flda48tnfsve4qthn2tzuqykdczzfl5rwkka66kl5ffs3sdz82d...#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m      └ #033[0m#033[36m#033[1m<function parse_cln_msat at 0x7fb5b9f310>#033[0m
Jul 20 08:44:34 raspberrypi python[805]: #033[31m#033[1mKeyError#033[0m:#033[1m 'amount_msat'#033[0m
Jul 20 08:44:34 raspberrypi python[805]: #033[0m
Jul 20 08:44:34 raspberrypi python[805]: #033[31m2023-07-20 08:44:34#033[0m | #033[31m#033[1m❌#033[0m | #033[31m#033[1mcln_jrpc.py:663#033[0m | #033[31m#033[1mGot an invoice but could not parse the data: 'amount_msat'#033[0m
Jul 20 08:44:34 raspberrypi python[805]: #033[32m2023-07-20 08:44:34#033[0m | #033[1mℹ️#033[0m | #033[1mcln_jrpc.py:944#033[0m | #033[1mRefreshing pay_index#033[0m
Jul 20 08:44:34 raspberrypi python[805]: #033[31m2023-07-20 08:44:34#033[0m | #033[31m#033[1m❌#033[0m | #033[31m#033[1mcln_jrpc_blitz.py:103#033[0m | #033[31m#033[1mAn error has been caught in function 'list_invoices', process 'MainProcess' (805), thread 'MainThread' (548555055168):#033[0m
Jul 20 08:44:34 raspberrypi python[805]: #033[31m#033[1m#033[33m#033[1mTraceback (most recent call last):#033[0m
Jul 20 08:44:34 raspberrypi python[805]:   File "#033[32m./app/lightning/impl/#033[0m#033[32m#033[1mcln_jrpc.py#033[0m", line #033[33m658#033[0m, in #033[35mlisten_invoices#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[1mi#033[0m #033[35m#033[1m=#033[0m #033[1mInvoice#033[0m#033[35m#033[1m.#033[0m#033[1mfrom_cln_json#033[0m#033[1m(#033[0m#033[1mdata#033[0m#033[1m)#033[0m
Jul 20 08:44:34 raspberrypi python[805]:     #033[36m    │       │             └ #033[0m#033[36m#033[1m{'label': 'keysend-1639959922.637279029', 'bolt11': 'lnbc1psmlntjpp598l5j8flda48tnfsve4qthn2tzuqykdczzfl5rwkka66kl5ffs [... cut ...]
Jul 20 08:47:17 raspberrypi python[805]:            │   │      │    └ <function main at 0x7fb731ef70>
Jul 20 08:47:17 raspberrypi python[805]:            │   │      └ <Command main>
Jul 20 08:47:17 raspberrypi python[805]:            │   └ <function Context.invoke at 0x7fb75aa4c0>
Jul 20 08:47:17 raspberrypi python[805]:            └ <click.core.Context object at 0x7fb7eb4040>
Jul 20 08:47:17 raspberrypi python[805]:   File "/home/blitzapi/blitz_api/venv/lib/python3.9/site-packages/click/core.py", line 760, in invoke

rootzoll, 2023-07-28:
looking at logrotate config `/etc/logrotate.d/rsyslog` that is created on `[build_sdcard.sh](https://github.com/raspiblitz/raspiblitz/blob/21d2593b901e1279f15b7ba6d131ef21336cf6e3/build_sdcard.sh#L493)`

```
/var/log/syslog
{
  rotate 7
  daily
  missingok
  notifempty
  delaycompress
  compress
  postrotate
    invoke-rc.d rsyslog rotate > /dev/null
  endscript
}
```

```
/var/log/daemon.log
{
  rotate 4
  size=100M
  missingok
  notifempty
  compress
  delaycompress
  sharedscripts
  postrotate
    invoke-rc.d rsyslog rotate > /dev/null
  endscript
}
```

@openoms what are we missing here. daemon.log has "size=100M"  but is still reported to be bigger. Should it be "size 100M" (without the =)? And should we also add a "size 100M" to the /var/log/syslog?

rootzoll, 2023-07-29:
After looking a bit longer into it I think we can further reduce complexity in the logrotate config to:
```
/var/log/syslog
/var/log/mail.info
/var/log/mail.warn
/var/log/mail.err
/var/log/mail.log
/var/log/daemon.log
/var/log/kern.log
/var/log/auth.log
/var/log/user.log
/var/log/lpr.log
/var/log/cron.log
/var/log/debug
/var/log/messages
{
  rotate 3
  size 100M
  missingok
  compress
  delaycompress
  sharedscripts
  postrotate
    invoke-rc.d rsyslog rotate > /dev/null
  endscript
}
```
see PR https://github.com/raspiblitz/raspiblitz/pull/4038

rootzoll, 2023-07-31:
PR merged now on final testing for next release candidate

rootzoll, 2023-09-14:
added an emergeny delete - but pushing to next milestone for further investigation
@openoms please note here your findings

benjamin-wilson, 2023-09-20:
I have the same issue on a pretty fresh and clean install

geco91, 2024-06-04:
checked a longer running v1.11.0 system ... looks ok. recommend closing.

rootzoll, 2024-06-09:
Closing - please reopen if happens on a RaspiBlitz release >=1.11.1

closed 2024-06-09 by rootzoll