raspiblitz/raspiblitz #4024: Specter Desktop cant be installed on 1.9

surfac3, 2023-07-20:
When attempting to instal Specter Desktop on Raspiblitz 1.9 the install fails on following errors

Building wheels for collected packages: hidapi
  Building wheel for hidapi (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/specter/.env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/setup.py'"'"'; __file__='"'"'/tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-j7skhru3
       cwd: /tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/
  Complete output (9 lines):
  running bdist_wheel
  running build
  running build_ext
  Compiling hid.pyx because it changed.
  [1/1] Cythonizing hid.pyx
  /tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/.eggs/Cython-3.0.0-py3.9-linux-aarch64.egg/Cython/Compiler/Main.py:381: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/hid.pyx
    tree = Parsing.p_module(s, pxd, full_module_name)
  building 'hid' extension
  error: unknown file type '.pxd' (from 'chid.pxd')
  ----------------------------------------
  ERROR: Failed building wheel for hidapi
  Running setup.py clean for hidapi
Failed to build hidapi
Installing collected packages: hidapi, ecdsa, click, base58, zipp, wtforms, urllib3, pyasn1, pyaes, mnemonic, libusb1, idna, flask, charset-normalizer, certifi, bitbox02, babel, aniso8601, stem, requests, python-dotenv, pysocks, pyserial, pyopenssl, psutil, pgpy, importlib-metadata, hwi, flask-wtf, flask-restful, flask-login, flask-httpauth, flask-cors, flask-babel, embit, chardet, cbor, cryptoadvance.specter
    Running setup.py install for hidapi ... error
    ERROR: Command errored out with exit status 1:
     command: /home/specter/.env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/setup.py'"'"'; __file__='"'"'/tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-hzpwsvk2/install-record.txt --single-version-externally-managed --compile --install-headers /home/specter/.env/include/site/python3.9/hidapi
         cwd: /tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/
    Complete output (5 lines):
    running install
    running build

    running build_ext
    building 'hid' extension
    error: unknown file type '.pxd' (from 'chid.pxd')
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/specter/.env/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/setup.py'"'"'; __file__='"'"'/tmp/pip-install-1iqmgjva/hidapi_135e494aec384524917fc9f7557dc7e0/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-hzpwsvk2/install-record.txt --single-version-externally-managed --compile --install-headers /home/specter/.env/include/site/python3.9/hidapi Check the logs for full command output.

I am  attaching a screenshot as well in case the error is poorly readable 
[image]

Gilbog, 2023-07-21:
Same issue. Reflashing v1.9.0 didn't help. Neither did 
```
config.scripts/bonus.specter.sh off
sudo userdel -rf specter
config.scripts/bonus.specter.sh on
```
Same error arises.

openoms, 2023-07-21:
@surfac3 @Gilbog the version 1.13.1 installed ok for me on v1.9 RPi.

Can you try to edit the  file:
```
nano ./config.scripts/bonus.specter.sh 
```
and change the version on top to:
```
pinnedVersion="1.13.1"
```

Uninstall clean first with running:
```
sudo userdel -rf specter
```
then install the updated version:
```
./config.scripts/bonus.specter.sh on
```

Gilbog, 2023-07-21:
Specter v1.13.1 installed without errors for me on RaspiBlitz v1.9.0 as well. 
After installation, Specter normally appears in the RaspiBlitz menu and the onion address is visible there.
Thanks!

surfac3, 2023-07-22:
I am able to see the specter app is installed but i cant log in.

"ERR_CONNECTION_REFUSED" maybe some sort of security browser issue? Tried multiple browsers and Windows and MacOS but still I am not able to access the app.

surfac3, 2023-07-22:
Are you able to observe these from the journalctl logs?

Jul 22 17:34:10 raspberrypi systemd[1]: specter.service: Failed with result 'exit-code'.
Jul 22 17:34:10 raspberrypi systemd[1]: specter.service: Consumed 5.678s CPU time.
Jul 22 17:34:40 raspberrypi systemd[1]: specter.service: Scheduled restart job, restart counter is at 9.
Jul 22 17:34:40 raspberrypi systemd[1]: Stopped specter.
Jul 22 17:34:40 raspberrypi systemd[1]: specter.service: Consumed 5.678s CPU time.
Jul 22 17:34:40 raspberrypi systemd[1]: Started specter.
Jul 22 17:34:42 raspberrypi python3[252830]: [2023-07-22 17:34:42,939] INFO in server: Configuration: cryptoadvance.specter.config.ProductionConfig
Jul 22 17:34:42 raspberrypi python3[252830]: [2023-07-22 17:34:42,940] INFO in server: SPECTER_DATA_FOLDER: /home/specter/.specter
Jul 22 17:34:42 raspberrypi python3[252830]: [2023-07-22 17:34:42,940] INFO in cli_server: CERT:/home/specter/.specter/cert.pem
Jul 22 17:34:42 raspberrypi python3[252830]: [2023-07-22 17:34:42,950] INFO in specter_migrator: Initiated MigDataManager(/home/specter/.specter/migration_data.json events:14 execs:3 )
Jul 22 17:34:42 raspberrypi python3[252830]: [2023-07-22 17:34:42,953] INFO in specter_migrator: Collecting possible migrations ...
Jul 22 17:34:42 raspberrypi python3[252830]: [2023-07-22 17:34:42,954] INFO in specter_migrator: Collecting possible migrations ...
Jul 22 17:34:42 raspberrypi python3[252830]: [2023-07-22 17:34:42,954] INFO in specter_migrator: No Migrations to execute!
Jul 22 17:34:42 raspberrypi python3[252830]: [2023-07-22 17:34:42,955] INFO in server: Initializing LoginManager
Jul 22 17:34:42 raspberrypi python3[252830]: [2023-07-22 17:34:42,955] INFO in server: Initializing Specter with data-folder /home/specter/.specter
Jul 22 17:34:44 raspberrypi python3[252830]: Traceback (most recent call last):
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
Jul 22 17:34:44 raspberrypi python3[252830]:     return _run_code(code, main_globals, None,
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
Jul 22 17:34:44 raspberrypi python3[252830]:     exec(code, run_globals)
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/__main__.py", line 7, in <module>
Jul 22 17:34:44 raspberrypi python3[252830]:     entry_point()
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
Jul 22 17:34:44 raspberrypi python3[252830]:     return self.main(*args, **kwargs)
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 1055, in main
Jul 22 17:34:44 raspberrypi python3[252830]:     rv = self.invoke(ctx)
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
Jul 22 17:34:44 raspberrypi python3[252830]:     return _process_result(sub_ctx.command.invoke(sub_ctx))
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
Jul 22 17:34:44 raspberrypi python3[252830]:     return ctx.invoke(self.callback, **ctx.params)
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 760, in invoke
Jul 22 17:34:44 raspberrypi python3[252830]:     return __callback(*args, **kwargs)
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/cli/cli_server.py", line 115, in server
Jul 22 17:34:44 raspberrypi python3[252830]:     init_app(app, hwibridge=hwibridge)
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/server.py", line 141, in init_app
Jul 22 17:34:44 raspberrypi python3[252830]:     specter = Specter(
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/specter.py", line 95, in __init__
Jul 22 17:34:44 raspberrypi python3[252830]:     self.node_manager = NodeManager(
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/managers/node_manager.py", line 36, in __init__
Jul 22 17:34:44 raspberrypi python3[252830]:     self.update(data_folder)
Jul 22 17:34:44 raspberrypi python3[252830]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/managers/node_manager.py", line 56, in update
Jul 22 17:34:44 raspberrypi python3[252830]:     Node if nodes_files[node_alias]["external_node"] else InternalNode
Jul 22 17:34:44 raspberrypi python3[252830]: KeyError: 'external_node'
Jul 22 17:34:45 raspberrypi systemd[1]: specter.service: Main process exited, code=exited, status=1/FAILURE
Jul 22 17:34:45 raspberrypi systemd[1]: specter.service: Failed with result 'exit-code'.
Jul 22 17:34:45 raspberrypi systemd[1]: specter.service: Consumed 4.414s CPU time.

openoms, 2023-07-25:
@surfac3 you need to add:

```
    "external_node": true,
```
to:
```
sudo nano /home/specter/.specter/nodes/default.json
```
so it looks like this:
```
₿ sudo cat /home/specter/.specter/nodes/default.json
{
    "name": "raspiblitz_mainnet",
    "alias": "default",
    "autodetect": false,
    "datadir": "/mnt/hdd/bitcoin",
    "user": "raspibolt",
    "password": "PASSWORD_B",
    "port": "8332",
    "host": "localhost",
    "protocol": "http",
    "external_node": true,
    "fullpath": "/home/specter/.specter/nodes/default.json",
    "python_class": "cryptoadvance.specter.node.Node"
```

Here is my main config file also for the record:
```
₿ sudo cat /home/specter/.specter/config.json
{
    "auth": {
        "method": "rpcpasswordaspin",
        "password_min_chars": 6,
        "rate_limit": 10,
        "registration_link_timeout": 1
    },
    "explorers": {
        "main": "",
        "test": "",
        "regtest": "",
        "signet": ""
    },
    "explorer_id": {
        "main": "CUSTOM",
        "test": "CUSTOM",
        "regtest": "CUSTOM",
        "signet": "CUSTOM"
    },
    "asset_labels": {
        "liquidv1": {}
    },
    "active_node_alias": "raspiblitz_mainnet",
    "proxy_url": "socks5h://localhost:9050",
    "only_tor": "true",
    "tor_control_port": "9051",
    "tor_status": false,
    "hwi_bridge_url": "/hwi/api/",
    "uid": "676005be4f4a2f6d",
    "unit": "btc",
    "price_check": false,
    "alt_rate": 1,
    "alt_symbol": "BTC",
    "price_provider": "",
    "weight_unit": "oz",
    "validate_merkle_proofs": false,
    "fee_estimator": "bitcoin_core",
    "fee_estimator_custom_url": "",
    "hide_sensitive_info": false,
    "autohide_sensitive_info_timeout_minutes": null,
    "autologout_timeout_hours": 4,
    "testing": {
        "allow_threading_for_testing": false
    },
    "bitcoind": false
```

surfac3, 2023-07-25:
I have changed according to your instructions
there were missing 
`"python_class": "cryptoadvance.specter.node.Node"`
`"datadir": "/mnt/hdd/bitcoin",`
supprisingly ` "external_node": true,` was there

{
    "name": "raspiblitz_mainnet",
    "alias": "default",
    "autodetect": false,
    "datadir": "/mnt/hdd/bitcoin",
    "user": "raspibolt",
    "password": "PASSWORD-B",
    "port": "8332",
    "host": "localhost",
    "protocol": "http",
    "external_node": true,
    "fullpath": "/home/specter/.specter/nodes/default.json"
    "python_class": "cryptoadvance.specter.node.Node"
}

Sadly still doesnt work

openoms, 2023-07-25:
can try running:
`config.scripts/bonus.specter.sh config`

or reinstall including deleting those config files

surfac3, 2023-07-26:
When i ran 
`config.scripts/bonus.specter.sh config`
i reset the config to original values 

so again is missing 
"python_class": "cryptoadvance.specter.node.Node"
"datadir": "/mnt/hdd/bitcoin",

"name": "raspiblitz_mainnet",
    "alias": "default",
    "autodetect": false,
    "datadir": "",
    "user": "raspibolt",
    "password": "PASSWORD-B",
    "port": "8332",
    "host": "localhost",
    "protocol": "http",
    "external_node": true,
    "fullpath": "/home/specter/.specter/nodes/default.json"
    
 `sudo  systemctl status specter` gives out this
● specter.service - specter
     Loaded: loaded (/etc/systemd/system/specter.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Wed 2023-07-26 08:25:13 BST; 20s ago
    Process: 3658039 ExecStart=/home/specter/.env/bin/python3 -m cryptoadvance.specter server --host 0.0.0.0 --cert=/home/specter/.specter/cert.pem --key=/home/specter/.specter/key.pem (code=exited, status=1/FAILU>
   Main PID: 3658039 (code=exited, status=1/FAILURE)
        CPU: 4.772s

surfac3, 2023-07-26:
`sudo journalctl -fu specter -n 50` gives me following log

Jul 26 12:14:19 raspberrypi systemd[1]: specter.service: Main process exited, code=exited, status=1/FAILURE
Jul 26 12:14:19 raspberrypi systemd[1]: specter.service: Failed with result 'exit-code'.
Jul 26 12:14:19 raspberrypi systemd[1]: specter.service: Consumed 4.843s CPU time.
Jul 26 12:14:49 raspberrypi systemd[1]: specter.service: Scheduled restart job, restart counter is at 2072.
Jul 26 12:14:49 raspberrypi systemd[1]: Stopped specter.
Jul 26 12:14:49 raspberrypi systemd[1]: specter.service: Consumed 4.843s CPU time.
Jul 26 12:14:49 raspberrypi systemd[1]: Started specter.
Jul 26 12:14:51 raspberrypi python3[1306702]: [2023-07-26 12:14:51,913] INFO in server: Configuration: cryptoadvance.specter.config.ProductionConfig
Jul 26 12:14:51 raspberrypi python3[1306702]: [2023-07-26 12:14:51,914] INFO in server: SPECTER_DATA_FOLDER: /home/specter/.specter
Jul 26 12:14:51 raspberrypi python3[1306702]: [2023-07-26 12:14:51,914] INFO in cli_server: CERT:/home/specter/.specter/cert.pem
Jul 26 12:14:51 raspberrypi python3[1306702]: [2023-07-26 12:14:51,928] INFO in specter_migrator: Initiated MigDataManager(/home/specter/.specter/migration_data.json events:14 execs:3 )
Jul 26 12:14:51 raspberrypi python3[1306702]: [2023-07-26 12:14:51,932] INFO in specter_migrator: Collecting possible migrations ...
Jul 26 12:14:51 raspberrypi python3[1306702]: [2023-07-26 12:14:51,934] INFO in specter_migrator: Collecting possible migrations ...
Jul 26 12:14:51 raspberrypi python3[1306702]: [2023-07-26 12:14:51,935] INFO in specter_migrator: No Migrations to execute!
Jul 26 12:14:51 raspberrypi python3[1306702]: [2023-07-26 12:14:51,935] INFO in server: Initializing LoginManager
Jul 26 12:14:51 raspberrypi python3[1306702]: [2023-07-26 12:14:51,936] INFO in server: Initializing Specter with data-folder /home/specter/.specter
Jul 26 12:14:54 raspberrypi python3[1306702]: Traceback (most recent call last):
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
Jul 26 12:14:54 raspberrypi python3[1306702]:     return _run_code(code, main_globals, None,
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
Jul 26 12:14:54 raspberrypi python3[1306702]:     exec(code, run_globals)
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/__main__.py", line 7, in <module>
Jul 26 12:14:54 raspberrypi python3[1306702]:     entry_point()
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 1130, in __call__
Jul 26 12:14:54 raspberrypi python3[1306702]:     return self.main(*args, **kwargs)
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 1055, in main
Jul 26 12:14:54 raspberrypi python3[1306702]:     rv = self.invoke(ctx)
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
Jul 26 12:14:54 raspberrypi python3[1306702]:     return _process_result(sub_ctx.command.invoke(sub_ctx))
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
Jul 26 12:14:54 raspberrypi python3[1306702]:     return ctx.invoke(self.callback, **ctx.params)
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/click/core.py", line 760, in invoke
Jul 26 12:14:54 raspberrypi python3[1306702]:     return __callback(*args, **kwargs)
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/cli/cli_server.py", line 115, in server
Jul 26 12:14:54 raspberrypi python3[1306702]:     init_app(app, hwibridge=hwibridge)
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/server.py", line 141, in init_app
Jul 26 12:14:54 raspberrypi python3[1306702]:     specter = Specter(
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/specter.py", line 95, in __init__
Jul 26 12:14:54 raspberrypi python3[1306702]:     self.node_manager = NodeManager(
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/managers/node_manager.py", line 36, in __init__
Jul 26 12:14:54 raspberrypi python3[1306702]:     self.update(data_folder)
Jul 26 12:14:54 raspberrypi python3[1306702]:   File "/home/specter/.env/lib/python3.9/site-packages/cryptoadvance/specter/managers/node_manager.py", line 56, in update
Jul 26 12:14:54 raspberrypi python3[1306702]:     Node if nodes_files[node_alias]["external_node"] else InternalNode
Jul 26 12:14:54 raspberrypi python3[1306702]: KeyError: 'external_node'

closed 2024-08-10 by openoms