raspiblitz/raspiblitz #4025: Raspiblitz-img install on intel nuc 7i7 m2.sata 240GB

DuckG, 2023-07-21:
I have a problem to use the image raspiblitz.

Etcher version: 1.18.11
Operating system and architecture: Intel NUC 7i7BNH   
Image flashed: raspiblitz-fat-v1.9.0-2023-05-22 also raspiblitz-v1.8.0c-2022-11-01.img
What do you think should have happened: Raspiblitz startup
What happened: non-blinking cursor on black screen.

Hardware is good,(ssd is visable) i have tried other images (ie linux rockylinux or windows) 
SSD= 240Gb [kingston](m2.sata 22x80) = SM228053G2
[image]

adevoss, 2023-07-23:
Raspberry Pi has a arm cpu. Your NUC is x86-64 (amd64).

Two possibilities:
- use a raspiblitz amd64 image
- install Debian (11, bullseye!) and run the build_sdcard.sh script

https://github.com/raspiblitz/raspiblitz/blob/v1.9/alternative.platforms/README.md

openoms, 2023-07-25:
As adevoss wrote above you need an amd64 image for PC, the RPi ones won't work.

The most up to date images and guide are in:
https://github.com/raspiblitz/raspiblitz/blob/dev/ci/README.md#images-generated-in-github-actions